tock/tock #973: NRF52dk BLE Examples Not Working

dcellucci, 2018-06-06:
Hi all, 

When I attempt to run the BLE Advertising examples on the NRF52dk (tutorial 03, `ble/ble_advertise`, and `ble_advertise`) I get the following error:
```
ble_start_advertising, error: Underlying system is busy; retry
```
Is this expected right now?

niklasad1, 2018-06-06:
@dcellucci I think you hadn't recompiled `libtock` with changes in the kernel or libtock. 

It works now?!

dcellucci, 2018-06-08:
Yes, though I am still having issues with BLE-uart. Every time I run on the NRF52dk, I get `BLE ERROR: Code = 16385`

bradjc, 2018-06-08:
The ble-uart app I believe only works with a platform that supports Nordic's serialization library (platforms like Hail and imix). I'm not sure what the state of having apps be able to support both serialization and the nrf radio directly is.

niklasad1, 2018-06-08:
@bradjc Yeah, you are right

I think we have everything we need for using the nrf radio directly on `hail` via I2C but I haven´t tried yet!

dcellucci, 2018-06-09:
I finally realized this after messing around with the nrfserialization library and seeing that `NRFMakefile.mk` looks customized for hail/imix. Sorry for the unnecessary issue...

On a side note, would a bluetooth ping tutorial with two NRF52s be appreciated?

bradjc, 2018-06-09:
Oh no worries. That is a work in progress area of Tock that we should better clarify.

And absolutely, documentation and tutorials are always welcome :)