tock/tock #994: RFC: Future of Tock Hardware

bradjc, 2018-06-13:
_Writing down some things from recent discussions._

With new tutorials coming up, the issue of what hardware to use is important, and getting an early start gives us time to make sure things will work. This then prompts the obvious question of what hardware do we want to support generally going forward.

Current Issues
----------------

- Doing tutorials with Hail doesn't seem to be getting us much buy-in, despite that Hail is available for purchase. It might be cost, or that researchers don't see how they can effectively use hail, or something else.
- The SAM4L is a source of numerous bugs and weirdnesses. It is also rather large and expensive. Plus it is difficult to get the hardware correct when making a new board.
- There is no LoRa support on any of our HW, yet many researchers are using LoRa radios.

Thoughts for the Tutorial
-------------------------------

- Want hardware that researchers can seem themselves building off of. This would either be for sensing applications or networking experimentation.
- Would like to do something with the 6lowpan stack.

Ideas
-----

- Use the nRF52840dk board with an arduino sensor shield. This is appealing because the nRF52840 is a very compelling chip that supports BLE and 15.4, the nRF52840dk is readily available, and arduino is popular. However, after looking into this, there seem to be almost no arduino sensor shields. It's a little crazy that that can be true, but it seems the arduino crowd uses jumpers and breadboards for their sensors.

Potential Future HW Platforms
----------------------

- The [ACD52832](https://www.mouser.com/ProductDetail/aconno/ACD52832?qs=sGAEpiMZZMve4%2fbfQkoj%252bAOKB4ochF2XCuH5Oopp1aE%3d) is very interesting. It has some useful peripherals, and they are working on a nRF52840 version.
- The [permamote](https://github.com/lab11/permamote) could be a very useful base for an energy-harevesting / low power Tock board.
- Something with LoRaWAN. Based on my searching, this would have to be a custom board as well.

brghena, 2018-06-14:
Two options on the arduino shield side:

- [Sparkfun Weather Shield](https://www.sparkfun.com/products/13956) which has temperature, humidity, pressure, and light (uses an SI7021 which we already have a driver for)
- [Grove kit](https://www.seeedstudio.com/Grove-Starter-Kit-for-Arduino-p-1855.html) has several sensors and actuators. Grove allows you to plug sensors into any of several available slots, so we'd really have to get our configuration story down to get this working.

dcellucci, 2018-06-14:
Hello! I thought I'd weigh in since I've been doing my own trades for my research. 

The NRF52832 has a bunch of nice, small, integrated [SOC modules](https://www.digikey.com/product-detail/en/laird-wireless-thermal-systems/BL652-SA-01-T-R/BL652-SA-01-CT-ND/6181738) that can be soldered directly on a single-sided board. The NRF52840 has [a few as well](https://www.digikey.com/product-detail/en/fanstel-corp/BT840F/1914-1019-1-ND/8323672). I haven't tried either out yet, but I'd be happy to get a few and report back if there's interest. I've been mostly working with Adafruit's [NRF52 Feather board](https://www.adafruit.com/product/3406) for NRF dev, its a fine breakout and pretty cheap.

My experience with this level of engagement with DIY platforms is that people usually want to buy a breakout for a specific sensor they want, and then make a jumper/breadboard mess to test things out before graduating to a semi-finished PCB. The goal of the software there should be plug n play, most of the effort should be in the wiring and design of the boards, so what that means is alot of predefined sensor libraries (usually the breakout board vendors offer an Arduino library as a download on their product page). I'd be interested in helping with the tutorials for writing your own Tock sensor library, it would be along the lines of specifying the protocol (I2C, SPI) and the registers, right? 

I've found that you can make the breakout boards for the modules using something like EAGLE and a PCB mill that costs [a few hundred dollars](http://store.linksprite.com/diy-cnc-3-axis-engraver-machine-pcb-milling-wood-carving-router-kit-arduino-grbl-assembled-version/), and everything is integrated, since it's pretty easy to end up with spaghetti if your breadboard gets too sophisticated. This is how the [Fab Lab](http://www.fabfoundation.org/index.html) model works, and it's actually pretty effective. Having something like those standalone modules described above is pretty useful, then, since you don't have to worry about soldering tiny components- it just all comes as a box and is soldered onto the PCB you mill with the sensors.

Oh also. I've also worked with the CC265x, and I'm sorry @cpluss but I'm going to have to recommend against it. The documentation is shoddy and it's really easy to brick the boards (my collaborator and I bricked 4 of the 5 that I had last week and had to quit working on development there).

dcellucci, 2018-06-14:
Also also- I got given a bunch of [XBee3s](https://www.digi.com/products/xbee-rf-solutions/2-4-ghz-modules/xbee3-zigbee-3#resources). It's a pretty nice module as well. The only catch is it's running a chip that's unsupported- the [Silicon Labs EFR32MG SoC](https://www.silabs.com/products/wireless/mesh-networking/efr32mg-mighty-gecko-zigbee-thread-soc)

bradjc, 2018-06-14:
> Sparkfun Weather Shield 

But it doesn't come with headers, so you have to buy those which is a another (annoying) step, and if you don't have a soldering iron you are out of luck. To me that makes this board a non-starter.

> Grove kit 

I can see this working, but I wasn't impressed with the selection. And the variability (maybe someone doesn't buy all of the sensors or one is on backorder or something) is just a pain to manage.

> graduating to a semi-finished PCB

I completely agree, and I think we are doing a good job of supporting custom boards (and it will hopefully get much better and easier with the new board.rs interface coming soon).

> I've also worked with the CC265x, and I'm sorry @cpluss but I'm going to have to recommend against it.

Good to know. I can't say I'm surprised, TI MCUs are often super exciting on paper and meh in practice.

dcellucci, 2018-06-14:
Having tried to work with some custom applications using Libtock, it might be helpful to recreate the Arduino interface for Tock (providing the Arduino methods, not the GUI). No offense to anyone working on libtock...

alevy, 2018-06-16:
> it might be helpful to recreate the Arduino interface for Tock 

The libtock interface was designed such that it shouldn't be too bad to recreate an Arduino style interface (a single loop-body procedure with blocking I/o calls) if you want to live strictly in that world.

I'm in favor of having an that as an option (moving libtock out of the main repo will make this more palatable morally, imo), but don't necessarily know what the main delta is between what we have now as blocking calls and what Arduino style interfaces would be.

torfmaster, 2018-06-16:
I'd be interested in having tock running on RTL8710 which is an alternative to the ESP8266.

cpluss, 2018-06-16:
> Oh also. I've also worked with the CC265x, and I'm sorry @cpluss but I'm going to have to recommend against it. The documentation is shoddy and it's really easy to brick the boards (my collaborator and I bricked 4 of the 5 that I had last week and had to quit working on development there).

Agree with @dcellucci . The documentation is poor, and it's super easy to brick devices (we've bricked 5 during our thesis work). 

> I'd be interested in having tock running on RTL8710 which is an alternative to the ESP8266.

@torfmaster , it seems to be running on a Cortex-M3 arch, does it have an MPU?

alevy, 2018-06-16:
Just reground the discussion, I *think* @bradjc is talking about "full platforms", rather than just the choice of  MCUs to work on. i.e. a board with enough sensors or other peripherals to be a useful to development platform for early users.

It sounds like there is desire for certain other chips (which is also good to know).

alevy, 2018-06-16:
> But it doesn't come with headers, so you have to buy those which is a another (annoying) step, and if you don't have a soldering iron you are out of luck. To me that makes this board a non-starter.

Given the options available maybe it shouldn't be a non-starter?

If the headers came soldered on would that be a good choice?

dcellucci, 2018-06-16:
> Just reground the discussion, I think @bradjc is talking about "full platforms", rather than just the choice of MCUs to work on. i.e. a board with enough sensors or other peripherals to be a useful to development platform for early users.

Agreed- what is all y'alls vision on the typical user/consumer of tock hardware? That might also help nail down these specific engineering requirements. If it's "makers" (ugh), 'full platforms' might actually be less effective for these people than something closer to the Arduino/Feather model, and the focus I think is more on the softer parts of the experience- docs and software. However, if the target audience is engineering researchers getting started on distributed sensor networks than maybe it makes sense to do a lit review or interviews to find out what the most common sensors are in these kinds of projects and just include those, and maybe price point and learning curve aren't so bad because it's probably a grad student or undergrad who's going to be using it.

> The libtock interface was designed such that it shouldn't be too bad to recreate an Arduino-style interface (a single loop-body procedure with blocking I/o calls) if you want to live strictly in [...]

I agree I think it's mostly a cosmetic/documentation thing, and not a fundamental architecture shift. I might start cracking on it this weekend, definitely in the next few weeks as things clear up on my end. My vision was for a 1-to-1 mapping between the Arduino examples and libtock, where most simple sketches can just be drag/dropped from the Arduino IDE into Tock. 

>Agree with @dcellucci . The documentation is poor, and it's super easy to brick devices (we've bricked 5 during our thesis work).

I have never run into a company that is more passive aggressive to the people buying their products than TI. It's incredible.

bradjc, 2018-06-17:
> If the headers came soldered on would that be a good choice?

I think it would be an ok choice. It's a little expensive ($40, so $90 with the nrf52840dk) and only provides a pressure sensor over what hail has.

> if the target audience is engineering researchers getting started on distributed sensor networks

I don't think tock has to have a singular focus by any means, but given the industry the core team is in I'd say this is the key focus. My hypothesis on hardware has been that availability is key, but while that is necessary it doesn't seem to be sufficient. A lot of groups are still using the telosb, which implies that code availability, low power, and wireless radios matter. I would like to see tock be a leader on the communication side as well as the programming safety side, which is why I think the nrf52840 is exciting, and would like to see Lora and wifi support.

lemmi, 2018-06-26:
We are in the process of developing an open source board for indoor localization. We evaluated a couple of chips and for us the clear winners are the nrf52 chips. I have to agree with @dcellucci that TI's CC2265x lacks the openness to be worth investing in it. So clearly we'd like to see more support for anything that incorporates any nrf52 variant :)

bradjc, 2018-06-29:
I'm just learning now that the nRF52840 has the cryptocell 310 which is apparently closed source.

[image]

[image]
https://openthread.io/platforms/nrf52840

That might be a dealbreaker for Tock on the nRF52840 :(

bradjc, 2019-10-02:
A great contender is the https://store.arduino.cc/usa/nano-33-ble-sense

The one thing that is needed for that board is someone to track down the [bootloader](https://forum.arduino.cc/index.php?PHPSESSID=2jk96uju63768b9jl3bbdc3ki1&topic=631550.msg4314397#msg4314397) and learn how to build it ourselves and flash it on the board. Then we could add tockloader support and it would be a very usable platform.

40Grit, 2019-10-11:
https://os.mbed.com/platforms/agora-dev/

We are still working on availability and pricing but this board is production ready. PTCRB and Mbed OS Ready by end of October.

Additionaly there are demos in the wild using a sim as a single source of identity for both cellular connectivity and device managment.

It is based around a Fanstel NRF52840 module.

Contact sales@embeddedplanet.com for more info.

bradjc, 2019-10-11:
> https://os.mbed.com/platforms/agora-dev/

Looks really interesting. I think the price point will probably be quite high for getting started with Tock, but for someone interested in exploring the different connectivity options this would allow them to use Tock.

bradjc, 2019-10-17:
As a related note, having a small, separate from the other buttons (i.e. user and reset), button on a Tock board would be great. This could be kept separate from the "normal" buttons in the kernel somehow, and could be used by a sticky debug app. For example, the debug app might print the process state and address locations when the button is pressed, or restart all applications, or something.

bradjc, 2020-01-29:
The [Arduino Nano 33 BLE Sense](https://store.arduino.cc/usa/nano-33-ble-sense) is one of the most promising options we have seen recently. It has sensors, a small form factor, is easily available, and is cheap. Plus it runs an MCU (nRF52840) we already support, which supports BLE and 15.4.

So what's the catch? Well, it doesn't have an FTDI chip. That is technically OK, since the board uses the nRF's USB controller, but it does mean we need a USB stack for the nRF if we are going to write a bootloader in Tock, but there is progress on that front.

The main issue is the bootloader that ships with the board is ([as far as I can tell](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/23)) closed source and does not support reading from the chip. Without read functionality we cannot implement Tockloader support for the bossa bootloader, and without full Tockloader support we cannot offer the seamless user experience we want.

Ideas:
- We could buy the boards ourselves, program them with our own code and bootloader, and then re-sell them. This is probably too much overhead and work.
- There might be a way to run a command and have the bootloader overwrite itself. I'm not sure about that, and we don't have a bootloader currently to replace it with.

rpavlik, 2020-05-08:
FWIW, Adafruit now has a Feather NRF52840 Sense, which uses an open-source bootloader supporting UF2 mass-storage as well as CDC (nrfutil) flashing. Unlike the Nano 33 BLE +-Sense using the Arduino mBED-based BSP, the Adafruit bootloader is combined with the softdevice (by default, anyway - you can update to a bootloader without a softdevice)

bradjc, 2020-05-08:
Any idea when those will be in stock?

rpavlik, 2020-05-08:
No - just an adafruit fan, not an employee. They're in NYC, so currently running a minimal crew right now. Since those are newer products, I imagine when they get around to manufacturing again they'll probably be pretty high on the list, but that's just a guess.

alistair23, 2020-05-08:
They have something similar in stock: https://www.adafruit.com/product/4062

SparkFun also have something similar: https://www.sparkfun.com/products/15025

brghena, 2020-05-08:
@alistair23 While those look like fine dev boards, it's the sensors that make the nano33, Feather Sense, and Clue exciting. Lots of dev boards have just UART, LEDs and buttons, and the microcontroller, which really limits what interesting things you can do with applications on the board. We designed Hail intentionally to include interesting sensors and enable interesting demos.

alistair23, 2020-05-08:
Agreed. I just meant while the other one is out of stock you could do testing with those. I also think having lots of similar boards means that people might be able to get started with Tock with what they already have.

rpavlik, 2020-05-19:
Looks like the Clue is back in stock and shipping.

hudson-ayers, 2020-05-28:
The feather nrf52840 sense is also back in stock it looks like: https://www.adafruit.com/product/4516

bradjc, 2021-01-04:
I think we can close this discussion for now, and revisit it when the nRF52 is no longer current :).

It seems there are now a handful of nrf52-based boards (nano33, feather, microbitv2) with sensors, good distribution, and low price points. The tock bootloader also now works with the nrf52 USB stack, so we can have full `tockloader` support.

closed 2021-01-04 by bradjc