tock/tock #1002: svd2regs: Add support for register clusters

bradjc, 2018-06-19:
The nRF52 SVD file uses `<cluster>` tags, and the svd2regs.py script ignores them. The script needs to be updated to parse the cluster correctly and include those registers.

lemmi, 2018-06-21:
@stefanhoelzl: Are you able to help with this? I'm currently looking at implementing the ADC for nrf52 and quite a number of registers are defined within clusters.
https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_cluster

stefanhoelzl, 2018-06-21:
svd2reg.py uses the cmsis-svd package to parse the SVD file.
The cmsis-svd package does not support clusters but there is a Issue for it
https://github.com/posborne/cmsis-svd/issues/47

As soon as cmsis-svd supports clusters, I can adapt the svd2regs.py to support it too.

stefanhoelzl, 2018-06-21:
forked cmsis-svd and added the cluster support.
you can install my version of cmsis-svd with the following command:
`pip install  -U -e 'git+https://github.com/stefanhoelzl/cmsis-svd.git@cluster-support#egg=cmsis-svd&subdirectory=python'`
and give it a try

lemmi, 2018-06-22:
@stefanhoelzl thnx for taking this on so quickly :+1: 

I had a quick look and I found the following:

At the moment it seams to generate a type for every register in the cluster, so there is a type `EVENTS_CH[0]_LIMITH`, a type `EVENTS_CH[1]_LIMITH`, ...

But I think that cluster represent arrays of the same type. So instead of generating
```rust
...
/// Description cluster[0]:  Last results is equal or above CH[0].LIMIT.HIGH
events_ch_0__limith: ReadWrite<u32, EVENTS_CH[0]_LIMITH::Register>,
/// Description cluster[0]:  Last results is equal or below CH[0].LIMIT.LOW
events_ch_0__limitl: ReadWrite<u32, EVENTS_CH[0]_LIMITL::Register>,
/// Description cluster[0]:  Last results is equal or above CH[0].LIMIT.HIGH
events_ch_1__limith: ReadWrite<u32, EVENTS_CH[1]_LIMITH::Register>,
/// Description cluster[0]:  Last results is equal or below CH[0].LIMIT.LOW
events_ch_1__limitl: ReadWrite<u32, EVENTS_CH[1]_LIMITL::Register>,

...
```
(which I don't think is valid syntax anyway), it should look more like

```rust
...
events_ch: [EVENTS_CH; 8],
...
}

#[repr(C)]
struct EVENTS_CH {
    limith: ReadWrite<u32, EVENT::Register>,
    limitl: ReadWrite<u32, EVENT::Register>,
}
```

I'm testing this with `./svd2regs.py --mcu Nordic nrf52 SAADC`.

bradjc, 2019-06-04:
I'm still trying: https://github.com/posborne/cmsis-svd/issues/73

bradjc, 2019-10-02:
No progress on https://github.com/posborne/cmsis-svd/issues/73

bradjc, 2020-02-21:
Certainly doesn't seem like this one is ever going to happen unless we fork cmsis-svd.

bradjc, 2021-01-04:
Yeah, this isn't happening.

closed 2021-01-04 by bradjc