tock/tock #1008: RFC: ADC HIL

bradjc, 2018-06-20:
I am implementing basic ADC support for the nrf52 and have a couple questions about the interface. One is a specific detail and one is a little more general.

1. The unit for what a sample means is not currently defined. It seems that currently it is a raw value where 0 is 0 V and 2^12-1 is 3.3 V. Should this be changed to something like the sample is in microvolts? The nrf52 can do 14 bit ADC reads, but can't use all of those bits in this interface.

2. What is the purpose of the `initialize()` function? I think this is confusing from both a user and implementor's perspective. Clearly it must be called before anything, but does it need to be called before every call? What is an implementor supposed to put in that function, given that there is no `deinitialize()` function?

    Removing the `initialize()` function I think makes things more clear, and allows each chip implementation to do its own thing. If there is some step that needs to be run initially then it can check for that and do it as needed.

    Also, I believe we were pushing for chips to automatically encourage low power modes, so I don't think we want to add a deinit function that the user has to remember to call.

alevy, 2018-06-20:
I agree with (2) completely. I could be missing something, but I don't believe there is a reason for `initialize` to be part of the ADC HIL.

I also agree with (1) that we should have a well defined, rather than totally arbitrary unit. Is there something we can use that would serve the same purpose as the `Frequency` trait for time? Where the low-level driver doesn't do the conversion itself, and thus there isn't any loss of precision or accuracy until it's actually _necessary_ to convert units, but clients can determine how to convert units in a portable way.

For the system call driver, microvolts certianly makes sense.

alevy, 2018-06-20:
> Is there something we can use that would serve the same purpose as the Frequency trait for time?

E.g. a trait that who's implementations specified the voltage range used:

```rust
trait ADCUnits {
    fn max_reading() -> u32 // the maximum possible reading
    fn reference_voltage() -> u32 // returns voltage in microvolts
    fn base_voltage() -> u32 { // returns voltage in microvolts, but zero by default
        0
    }
}
```
Then, when a client got a reading:

```rust
let adc_reading: u32 = ...;
let microvolts =
    (adc_reading / MyAdc::ADCUnits::max_reading()) *
        (MyAdc::ADCUnits::reference_voltage() - MyAdc::ADCUnits::base_voltage()) +
   MyAdc::ADCUnits::base_voltage();
```

brghena, 2018-06-20:
I really wish I could go back and understand my thinking when I upgraded the ADC HIL. Both of these points seem like obvious good ideas.

On the initialize point (2), my only excuse is that it was already there in the HIL before I made modifications, and I left it there without thinking. If you take a look at the SAM4L chip implementation, you'll see that `config_and_enable` is called at the start of each `sample` function anyways. We should remove `initialize` from the interface altogether, which in practice will necessitate no code additions, only deletions.

On point (1), I agree with that the HIL makes more sense as microvolts. In my opinion, the chip driver should be in charge of determining how to convert the reading into microvolts. It could do it in many different ways depending on what options it implements. The SAM4L driver would probably just hardcode the maximum as 3300000 microvolts, since its current implementation doesn't allow reference voltages to be modified.

A potential issue here, is that this would require upping the sample size to 32 bits (giving a possible range of 0 to 4000 volts in units microvolts). This would require changing the ABI for the ADC, which I believe we've claimed to be stable as of Tock 1.0.

alevy, 2018-06-20:
> A potential issue here, is that this would require upping the sample size to 32 bits (giving a possible range of 0 to 4000 volts in units microvolts). This would require changing the ABI for the ADC, which I believe we've claimed to be stable as of Tock 1.0.

The system call documentation for 1.0 doesn't specify a unit and is already a 32 bit value

bradjc, 2018-06-20:
Using a trait like `ADCUnits` would give us quite a bit of flexibility.

Is there a way to balance the memory vs. flexibility tradeoff? For example, they make 24 bit ADCs, but supporting those by making all ADC samples 32 bits isn't ideal. Maybe for the single sample case we could make it a `u32` without much worry, but for the streaming samples into a buffer case that would be bad if a chip only has an 8 bit ADC. Would rust be ok with us using `u8` buffers and then filling them with u16 or u32 values and specifying what the chip is doing somehow?

brghena, 2018-06-21:
This is a really good point. I forgot all about the streaming samples case. The problem there, is how do you guarantee that userland is allocating the same size buffer as you are? The solution might have to be a syscall from userland that requests the size of a sample on this platform.

bradjc, 2018-06-21:
> The problem there, is how do you guarantee that userland is allocating the same size buffer as you are?

Userland could have a different API where the samples are fixed as 16 bit values, say. Apps probably don't want to handle that variability.

phil-levis, 2018-06-22:
I’d recommend not having units but rather ensuring that values are left-justified. I.e., the bits of the ADC are the most significant bits of the provided value. This is important when you have external voltage references.

Phil

phil-levis, 2018-06-22:
Sorry, I'll expand on my previous comment.

If the ADC HIL ever allows selecting voltage references, then units becomes difficult, especially in the context of external voltage references (because their voltage is unknown unless you sample it). If the HIL is intended to be the lowest-level hardware-independent interface, then it should allow selecting voltage references (eventually).

Instead, I'd suggest that ADC values should be fractions of the voltage reference in the range of 0-MAXINT. This means that for an N-bit ADC simply sets the N most significant bits of the result. An 8-bit ADC sets the top 8 bits and leaves the bottom 24 as 0s, while a 12-bit ADC sets the top 12 bits and leaves the bottom 20 as 0s. There are cases then this is exactly what you want, because you don't care what the reference voltage is. E.g., when I sample a resistive sensor should I have to know what the voltage reference being used is?

If there is a separate interface that allows you to set/configure the voltage reference, then one can provide a higher-level interface that translates these raw ADC readings into specific units. E.g., if there's a 3V reference and you get 128 on an 8-bit ADC, that's 1.5V.

bradjc, 2018-07-09:
> Should there be separate HILs for on-chip vs. off-chip ADCs? That seems not very hardware independent. I think the one place we agreed this was the right decision was GPIO, because forcing every GPIO operation to be asynchronous would be extremely onerous.

> Examples of errors that the callback might want to give you:

> -    a sample was missed (LOVR, 38.6.18 in SAM4L data sheet)
> -    the ADC was turned off between when the sample started and acquisition started
> -    the ADC sample was cancelled before acquisition