tock/tock #1035: RFC: Initialize in HIL interfaces

bradjc, 2018-06-27:
The HIL interfaces in Tock can probably be lumped into two groups: those that have an `initialize()` or `enable()` function and those that don't. As far as I can tell whether a HIL has a function that should be called before other operations is more or less random and based on how the original author conceptualized what an interface to the hardware might look like.

Having functions like `initialize()` or `enable()` (I'll just refer to `initialize()` from now on) adds complexity to users of the HIL and to future implementors of the HIL. Therefore, it probably makes sense to standardize what `initialize()` functions do and when they should be called.

Issues with `initialize()`
-------------------------

Here are some of the issues that come up with `initialize()` like functions:

- It is not clear when they should get called. Are the called in the reset handler when the board boots? Are the called before every use of the other HIL functions?
- It is not clear how `initialize()` interacts with low power operation. Does `initialize()` imply the peripheral is in a high power state? Is `initialize()` only valid if the chip can still sleep after calling it?
- Should there be a `deinitialize()`? When would that function get called?

One Proposal
--------------

Remove all `initialize()` / `enable()` functions from HIL interfaces. Any initialization that needs to be done must be handled by the peripheral driver whenever a HIL function is called. The peripheral driver is then free to keep track of its enabled/initialized state, and is responsible for managing its power state itself.

This makes HIL interfaces easy to use as the caller is not responsible for deciding when and where to call `initialize()`.

This makes HIL interfaces easier to implement as implementors don't need to think about what operation is "safe" (i.e. doesn't break the Tock guidelines on what an `initialize()` function can do) to put in the `initialize()` function.

Any configuration values must be passed to each HIL function that needs them. This should make it easier to add a virtualization layer on top of the HIL.

phil-levis, 2018-07-02:
I think this makes sense. Initialize is greatly needed, but maybe it can occur at a lower level. I.e., the actual chip peripheral implementations need initialization, but this can be part of their Component, new() functions, etc. Most software implementations need some sort of initialization, but new() can take care of this. 

But let's walk through a specific example: SPI. SPI can operate in many different modes (speed, slave/master, pin polarities, clock polarities). Where/when are these set? If they are not in the HIL, they must be set in some kind of chip-specific initialize, or a series of calls to the control interface. Otherwise, the values might arbitrary. It's less bug prone to have a single call with everything required than multiple calls which a caller might forget. It's also less bug prone for the required configuration call to have a standard name.

Requiring this be handled by the HIL driver is messy. For example, would it mean that the SPI parameters are passed to new(), stored in RAM, then before every operation the SPI checks if it is the first operation, if so, uses the values in RAM to configure the hardware? This wastes RAM for a one-time operation -- that RAM is unused and wasted for the lifetime of the device.

alevy, 2018-07-02:
At least in the SPI case, this is a platform (i.e. board) specific set of configurations: whether a particular HW SPI should be in slave/master mode and the pin polarities (and whether software has the choice to set those in the first place) simply depends on which MCU is used and how the board hooks up the MCU to peripherals.

To me, this implies that it's perfectly fine (even preferable) to initialization functions that are hardware specific (i.e. not in a HIL at all, the particular MCU's low-level driver just exposes a regular old method).

Are there examples of controllers where this isn't the case? Where the "one-time-initialization" of the hardware isn't board+mcu specific?

phil-levis, 2018-07-02:
I think this boils down to whether we want to have have a flat or hierarchical initialization. I.e., if a capsule requires initialization, is it assured that the peripherals it relies on are initialized? So this is pushed to main() to initialize everything in the right order (flat). Alternatively, if the capsule itself initializes the peripheral, then you can be sure what it's state is in. If a capsule is initializing the peripheral, then this has to be in a HIL (hierarchical).

E.g., suppose a sensor driver capsule sits on top of a UART. It needs to configure the UART to be the right speed and data layout. But suppose that the hardware needs to be initialized before calls to set its parameters will stick (e.g., the UART is enabled). Should it be that the sensor capsule relies on boot having called initialize() properly before it is initialized()? Should boot know that it has to call initialize?

ppannuto, 2018-07-03:
Perhaps this could be helped by disentangling runtime configuration from peripheral initialization.

Specifically, I imagine initialization as the "power-on" type activities: e.g. attaching a clock so the peripheral can be used at all. I would imagine such things to be the domain of the board (flat) and done before the capsule.

Configuration could happen multiple times, e.g. a SPI Master with two peripherals requiring different polarities. From a SPI device HIL perspective, I would imagine that a configuration remains set until it is explicitly changed. For the case of a single capsule sitting atop a single device, the capsule can configure the device when it starts and then not worry about re-configuring it for each subsequent call. With multiple users of a device, the virtualization layer would be responsible for storing the configuration requested by each capsule, tracking which configuration is 'active', and updating the configuration of the underlying device based on which capsule is calling, thus providing the same 'configure-once' interface to capsules.

I don't think `initialize` should be part of the HIL. I think `configure` should be part of the HIL. I think the distinction is whether it's sensible to call the method more than once during the lifetime of the device.

phil-levis, 2018-07-04:
If a function is hardware-specific, then it can only be (directly) called by hardware-specific code. So the implication is that, if we ever want to have generic mechanisms for managing the boot sequence (e.g., automatically order dependencies), we'll have to add a layer of indirection.

So I think there's a slightly finer point here. On one hand, I think that there will need to be, at some point, methods in the HIL for initialization. However, they should almost certainly be in separate traits than the standard datapath. Given where our implementations are now, I'd generally agree with the point that none of the datapath HILs should have `initialize` in them. However, I'd also argue that we will want a HIL for initialization. Right now, this is `Component` (#1012).

alevy, 2018-07-04:
OK, given that I think there is rough consensus that removing `initialize` from the existing (data-path) HILs, I think it's time to move forward to proposed code changes.

E.g. the only thing I'm nervous about is moving configuration to each HIL method. My guess is that in practice this is not actually going to be bad, but it sounds scary, so worth seeing what the changes would look like and going from there.

hudson-ayers, 2020-06-12:
I realize this issue is very old, but I want to chime in to say that I think `initialize()` in HIL interfaces makes a lot more sense now that we have generic components.

bradjc, 2020-06-15:
I can see that components allow for one author to add the call to initialize and it would be less likely to be forgotten, but I'm not sure what else components help with.

hudson-ayers, 2020-06-15:
I just think that the ideal version of components is that instantiating a component returns you a ready-to-use capsule with no additional steps needed. If initialize() is not in the HIL, generic components cannot do this initialization, so its an extra step that has to be taken by each board.

ppannuto, 2021-05-25:
I think Phil basically captured the essence of this issue:

> ... I think that there will need to be, at some point, methods in the HIL for initialization. However, they should almost certainly be in separate traits than the standard datapath. Given where our implementations are now, I'd generally agree with the point that none of the datapath HILs should have initialize in them. However, I'd also argue that we will want a hardware-independent API for initialization. Right now, this is Component (#1012).

Now that components are reasonably established, perhaps it is time to imagine what initialization HILs should look like.

bradjc, 2021-05-26:
I think `configure(arg1, arg2, ...)` is important, but I'm still not sure how `hil.initialize()` (no arguments) is useful. Why should a board initialize a peripheral (in components or otherwise) if it never ends up being used? The implementation of the HIL can internally track the state of the hardware and do any initialization that may be necessary. By prohibiting including `initialize()` in the HIL, the driver author is responsible for making sure the peripheral is in a sound state, and not pushing that on the users.

hudson-ayers, 2021-05-26:
The only downside there is adding state checking to every operation taken by every driver, to confirm the device is initialized. I realize that adds up to not-very-much overhead though, and also agree it is more user friendly.

phil-levis, 2021-05-27:
> I think `configure(arg1, arg2, ...)` is important, but I'm still not sure how `hil.initialize()` (no arguments) is useful. Why should a board initialize a peripheral (in components or otherwise) if [...]

I can think of two reasons:
  - Fail fast. If there's something wrong and the peripheral can't be initialized, you want to discover this at boot, not at first use.
  - Predictable operation. Lazy initialization means that any call, if it's the first call, might take significantly longer than subsequent ones.