tock/tock #1069: Design of RNG HIL (work in progress)

phil-levis, 2018-07-03:
The RNG HIL currently is:

```pub trait RNG {
    fn get(&self);
}

pub trait Client {
    fn randomness_available(&self, randomness: &mut Iterator<Item = u32>) -> Continue;
}```

This has four major issues:

1. Setting a client is part of an implementation, rather than the HIL. Therefore it's not possible to build a client transparently of what implementation it sits on top of. This is important when we move to a generalized virtualizer that sits on top of the HIL. Setting the client for callbacks should be part of the RNG HIL.

2. Setting a client as part of the HIL requires adding a lifetime parameter, such that the trait can be implemented for lifetimes generically (rather than only for static).

3. Acquiring random numbers cannot fail. This implies a lot of things about the underlying implementation. For example, if for some reason a callback will never be issued there is no way to signal this to the caller. Similarly, if acquisition failed, there is no way to signal this to the caller in the callback (any error cases in the call must also be expressible in the callback).

4. There is no guarantee/statement about what kind of randomness this produces. There are generally two cases for random numbers: security-sensitive and security-insensitive. Security-sensitive random numbers are used for cryptography or other cases where there is an adversary. They require entropy guarantees. Security-insensitive random numbers are used for things like random backoffs in networking, or retries, or other probabilistic behavior. Random numbers with entropy guarantees are typically more expensive to acquire than those without. So we don't want to charge everyone who is security-insensitive the costs of security-sensitive random number generation.

alevy, 2018-07-04:
At a high level I think the proposed are much better. I may have some nits as i read more carefully, but my high-level vote is strongly in favor.

niklasad1, 2018-07-06:
I would also change want to `&mut Iterator<Item = u32>` is possible to move `u32` from the HIL and up to the implementation to decide that?

I guess that would require moving the some/entire `randomness_available` to each implementation instead?

However, I would be fine changing  `&mut Iterator<Item = u32>` to  `&mut Iterator<Item = u8>` which is a better assumption IMO because I guess all RNGs would at least provide one byte of randomness at the time! For chips that generate more then 1 byte at the time just mask non-requested RNG bytes!

phil-levis, 2018-07-11:
Given that we're only really considering 32-bit architectures, are there real concerns about an 8-bit randomness source? E.g., an 8-bit TRNG or an 8-bit LCG? I mostly worry that since 32-bits is the most efficient arithmetic representation, having it be a `<Item = u8>` would involve regenerating 32-bit values from numerous 8-bit values. It also seems extremely wasteful to generate 32-bit random values then mask away 24 of the bits.

I could be convinced that there should be a version of the trait for `<Item = u8>`, and a standard implementation that elevates a `u8` implementation into a `u32` one. Similarly, a standard implementation that does what you suggest, and mask a `u32` into a `u8`. I don't think it would be worth generalizing to arbitrary types, though.

niklasad1, 2018-07-18:
> Given that we're only really considering 32-bit architectures, are there real concerns about an 8-bit randomness source?

```nrf5x``` generates one-byte randomness at the time and the current implementation chains four generations and then yields the iterator which is not that nice. Mainly, because It is non-blocking and calls the same function four times in worst case we can have 4 stack frames for this (assuming non-inlined function)

> I mostly worry that since 32-bits is the most efficient arithmetic representation, having it be a <Item = u8> would involve regenerating 32-bit values from numerous 8-bit values. It also seems extremely wasteful to generate 32-bit random values then mask away 24 of the bits.

Yes, agree didn't think about this!

> I could be convinced that there should be a version of the trait for <Item = u8>, and a standard implementation that elevates a u8 implementation into a u32 one. Similarly, a standard implementation [...]

Yepp, that sounds good to me!

alevy, 2018-08-09:
> Given that we're only really considering 32-bit architectures, are there real concerns about an 8-bit randomness source?

Yes. As @niklasad1 pointed out, the nrf5x outputs 8-bits of randomness, and the cc26x2/cc13x2 output 64 bits of randomness. Randomness sources aren't really related the architecture's word-size.

phil-levis, 2018-08-21:
In the call yesterday, we mostly agreed that Rng should not have an associated type (i.e., keep Rng32 and Rng8). But looking at @niklasad1 's concerns with the nRF TRNG, I'm going to go a bit further: we should only have Rng, which is 32 bits.

The reason for this is pretty simple: you rarely want to generate 8-bit random numbers, and when you do, you can just generate a 32-bit and modulo it. 

@niklasad1 's comment about the nRF51 TRNG is well taken, though -- we want a trait that can abstract a wide range of hardware TRNGs. I'd argue that this should *not* be the Rng trait; instead, it should be the Entropy trait. Recall, Rng can represent a simple arithmetic random number generator (not cryptographically strong, or secure). Entropy, in contrast, is used to represent a source of true entropy (e.g., a TRNG, or AES in CTR mode with a TRNG-generated key). We need to decouple entropy (and hardware true RNGs) from standard random numbers. I realize this might seem like I'm splitting hairs at this point, but entropy and randomness is so critical to both cryptography and distributed systems that I'd argue if we want to be a secure OS we need to get this right.

So I think the answer to this question is to cut Rng8 and make Rng32 into Rng. Then, have two entropy traits, one of which generates bytes of entropy (just a byte stream), and once which generates 32-bit entropy values out of a byte stream. This second trait is just provided as a convenience for easily generating word-sized values rather than lots of code having to parse the byte stream. There will also be a converter from Entropy32 to Rng (but *not* vice versa). So the traits will look like this:

```
pub trait RNG<'a> {
    fn get(&self) -> ReturnCode;
    fn cancel(&self) -> ReturnCode;
    fn set_client(&self, &'a Client);
}

pub trait Client {
    fn randomness_available(&self, 
                             randomness: &mut Iterator<Item = u32>, 
                             error: ReturnCode) -> Continue;
}
```
and
```
pub trait Entropy<'a> {
    fn get(&self) -> ReturnCode;
    fn cancel(&self) -> ReturnCode;
    fn set_client(&self, &'a Client);
}

pub trait Client {
    fn randomness_available(&self, 
                             randomness: &mut Iterator<Item = u8>, 
                             error: ReturnCode) -> Continue;
}

pub trait Entropy32<'a> {
    fn get(&self) -> ReturnCode;
    fn cancel(&self) -> ReturnCode;
    fn set_client(&self, &'a Client);
}

pub trait Client32 {
    fn randomness_available(&self, 
                             randomness: &mut Iterator<Item = u32>, 
                             error: ReturnCode) -> Continue;
}
```

A hardware TRNG is expected to implement the Entropy trait. This can be converted into an Entropy32 with the converter from the `entropy` module, which can itself, if wanted, be converted into a Random32 with a converter from the `random` module. I'll also implement a struct in the `random` module that uses a TRNG for its initial seed but is then a simple linear congruence generator (or other statistically strong RNG).

ppannuto, 2020-01-30:
Closed by #1145.

closed 2020-01-30 by ppannuto