tock/tock #1072: Tracking & RFC: UART HIL Refinements

ppannuto, 2018-07-05:
n.b. The goal of this issue to to serve as a central location for discussion surrounding the overall direction UART HIL updates. As there are a few orthogonal HIL issues, the actual implementation of all the issues addressed here will be broken into discrete PRs, one for each major change.

----

**Tracking**

- [ ] Remove `initialize` from UART HIL.
   - [ ] RFC in #1035 
   - [ ] PR in #1072
- [ ] Add generic lifetime to UART HIL
   - [ ] RFC in _pending_
   - [ ] PR in _pending_
- [ ] Add `ReturnCode`s to appropriate HIL methods
   - [ ] RFC in #1052 
   - [ ] PR in #1049

bradjc, 2018-07-05:
Idea: go all the way and add a `UartDevice` trait (like I2C and SPI) that has the `UartParameters` baked in.

Pros:
- More consistent.
- Can be used to separate code that needs a capability to access hardware and code that doesn't.
- Makes it easier to determine when to call `configure()` (only when the actual user of the bus changes does any code need to even consider calling `configure()` again.

Cons:
- Why put in a bunch of code and complication for an interface that is very unlikely to not need all this complexity, given how UART busses are commonly used (between exactly two devices with fixed settings)?
- How would a receiver handle getting an unpredictable ordering of UART messages at different baud rates, parity, stop bits, and flow control settings (as could happen with something like `virtual_uart.rs`)?

What might be _more_ useful, and Tock-esque, is to have a `UartDevice` trait that:
- removes the `.configure()` function
- removes the `.abort_receive()` function

Then any normal user of the UART bus has access, it could be easily virtualized, and it doesn't have to worry about the bus settings and instead uses whatever was configured by something else (likely a board's main.rs). It also matches the use case we have so far: console and debug.rs sharing a UART channel.

Anything more advanced simply has to use the `Uart` trait, and then knows it has exclusive access to the uart bus and can change the settings as it pleases.

phil-levis, 2018-07-06:
The UART trait should be split into three different traits:
  - Control
  - Receive
  - Transmit

You have to separate Receive and Transmit because correct virtualization has very different behavior: you don’t want to allocate virtualized receive buffers when all you want is to transmit. If you separate out Control then the fact that virtualized interfaces can’t access control is explicit (the virtualizing struct does not expose Control).

Generally speaking, you always want to separate control from data. For asynchronous packet interfaces, you want to separate transmit from receive.

Phil

phil-levis, 2018-07-09:
https://groups.google.com/forum/#!topic/tock-dev/3NtJfzuz-ag

closed 2019-06-04 by bradjc