tock/tock #1074: RFC: Generic over static lifetimes in HIL interfaces

ppannuto, 2018-07-05:
Early in Tock we used `'static` lifetimes fairly liberally. As interfaces matured, we shifted towards generic lifetimes (`'a`) for the flexibility it (could eventually) affords components. Unfortunately, many of the HILs currently specific static rather than generic lifetimes.

As noted in (2) of #1069, one example of a negative consequence is an artificial restriction of where and how clients can be set.

I believe that the use of static lifetimes is largely a historical artifact, however I wanted to open this as an RFC in case this is something folks feel we need to discuss further before adapting.

## Current Status

From a highly scientific and imprecise study, we have more HILs with static lifetimes than generic, but I think that can be attributed to history. Loosely, the newer HILs seem to prefer generic lifetimes:

```bash
# Static
$ git grep static | grep -v '//!' | cut -d':' -f1 | sort | uniq -c
   9 adc.rs
   5 ble_advertising.rs
   4 flash.rs
  11 i2c.rs
   4 led.rs
   1 ninedof.rs
   5 nonvolatile_storage.rs
  12 radio.rs
   4 sensors.rs
  14 spi.rs
   3 symmetric_encryption.rs
   6 uart.rs
# Generic
$ git grep "'a" | grep -v '//!' | cut -d':' -f1 | sort | uniq -c
   2 flash.rs
  10 symmetric_encryption.rs
```

bradjc, 2018-07-06:
I thought we had a general notion in Tock that changes have to be used to be added. Is there a use for the generic lifetimes?

ppannuto, 2018-07-06:
What started me down this path was trying to move the `set_client` call for the UART from the top-level board file into the `initialize` method of the console capsule. We currently can't do that because the UART HIL requires static lifetimes but the console capsule has a generic lifetime.

I suspect there are other uses that we aren't currently doing because we can't. Let me flip the question, is there a reason that the HILs should be enforcing static lifetimes rather than permitting generic ones?

bradjc, 2018-07-06:
Well my concern is it adds a ton more rust syntax and is more work, and then it should probably get us some benefit. Sounds like the `set_client()` example is pretty compelling.

phil-levis, 2018-07-09:
They should all have generic lifetimes -- the set_client example is exactly why they transitioned. All kinds of things like hardware-independent virtualization (or true HILs) are effectively impossible without set_client().

bradjc, 2019-10-02:
If you are wondering what the count looks like today:

```
git grep static | grep -v '//!' | cut -d':' -f1 | sort | uniq -c
   9 adc.rs
   5 ble_advertising.rs
   4 flash.rs
   8 gpio.rs
  11 i2c.rs
  12 radio.rs
   4 sensors.rs
  14 spi.rs
   3 symmetric_encryption.rs
   8 uart.rs
```

lschuermann, 2023-06-14:
Wohoo!

closed 2023-06-14 by lschuermann