tock/tock #1079: nRF52-DK broken on `master`

niklasad1, 2018-07-06:
Hey,

`nRF52-DK` seems to be broken on the latest master (f695769e34b2b67073c7f02518b166c537111359)

I can see the following print: 
```bash
Initialization complete. Entering main loop
NRF52-DK HW INFO: Variant: AAB0, Part: N52832, Package: QF, Ram: K64, Flash: K512
```

But then `LED3` and `LED4` lights up even without any `app` loaded!

Anyone that can produce this?

alevy, 2018-07-07:
I was able to reproduce this. I haven't no idea what's going on, though if I had to guess, it's to do with the common nrf52-base board making an incorrect assumption about PIN assignments (or something like that---e.g. pulling those pins low because it's thinks they are buttons or something like that).

It doesn't impact other functionality, though. I'm able to run applications just fine and the work other than those two LEDs not being functional and just staying on.

niklasad1, 2018-07-08:
Found it, the `SPI` driver assigns pin 19-20 which are LED3 and LED4 on `nRF52-DK`!

niklasad1, 2018-07-08:
But the `UART` doesn't work for me!

PR #1097 (2018-07-09): Assign proper gpio pins to nRF52-DK
### Pull Request Overview

In `nrf52dk_base` we had assumed some pins are the same but they were
not. Because for `nRF52DK` pin 19 and pin 20 were assigned to both the
LEDS and the SPI drivers!

However, now `setup_board` takes very many parameters which is not
efficient i.e, can't go in the registers but since this is during
the initialization I guess it is ok!

Closes #1079 

### Testing Strategy

This pull request was tested on `nRF52-DK`

### TODO or Help Wanted

- [x] test on nRF52840DK

### Documentation Updated

~- [ ] Updated the relevant files in `/ [... cut]
closed 2018-07-09 by alevy