tock/tock #1116: Over 50% of attempts to flash Imix fail

hudson-ayers, 2018-07-16:
Over 50% of the time when I attempt to reflash the Imix kernel, or even when I attempt to flash new apps, the procedure fails, giving the following error:
...
NOTE: Padding binary with 184 0xFFs.
Error: No response after issuing command
Error: Error when flashing page
Traceback (most recent call last):
  File "/usr/local/bin/tockloader", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/tockloader/main.py", line 479, in main
    args.func(args)
  File "/usr/local/lib/python3.5/dist-packages/tockloader/main.py", line 198, in command_flash
    tock_loader.flash_binary(binary, args.address)
  File "/usr/local/lib/python3.5/dist-packages/tockloader/tockloader.py", line 58, in flash_binary
    self.channel.flash_binary(address, binary)
  File "/usr/local/lib/python3.5/dist-packages/tockloader/bootloader_serial.py", line 463, in flash_binary
    if ret[1] == self.RESPONSE_BADADDR:
IndexError: index out of range

This error often only occurs after 15-30 seconds, which can be quite annoying given that it sometimes happens as much as 4 times in a row before the command eventually works.

The command I use to flash the kernel is simply `make program` from the boards/imix directory. This happens across multiple boards and cables (tested on 4 imixes and 2 usb cables). 

Mostly posting this because I am curious if anyone else runs into a similar problem.

hudson-ayers, 2018-07-16:
It is 0.7.2

brghena, 2018-07-16:
Well, I'd recommend updating to start. Tockloader is on version `1.2.0` now, and that could just solve your problem.

hudson-ayers, 2018-07-16:
Well I feel stupid; that seems to have fixed it. Thanks for the advice, I'll close this.

closed 2018-07-16 by hudson-ayers