tock/tock #1118: RFC: Reorganize Capsules Directory

bradjc, 2018-07-16:
Currently we have a single directory which contains all capsules. However, capsules can be used in a variety of ways, as illustrated by the [README](https://github.com/tock/tock/blob/master/capsules/README.md):

```
- Sensor and other IC Drivers
- Libraries
- MCU Peripherals for Userspace
- Helpful Userspace Capsules
- Virtualized Sensor Capsules for Userspace
- Virtualized Hardware Resources
- Utility Capsules
```

It's probably a good thing we are seeing a variety of different types of capsules since all capsule code cannot use unsafe.

As the list of capsules gets longer, however, it gets more and more difficult to manage the flat list for at least three reasons:

1. Finding groups of capsules (for example to find the ones that provide a `Driver` interface) is challenging. Finding groups would help with finding capsules which are not virtualized, for example.
2. Finding capsules to copy from if you are implementing a new capsule is daunting.
3. The flat list doesn't provide any information about how that capsule is likely to be used, so finding that can mean inspecting each file individually.

Based on this, I think we should have subfolders for capsules.

Open Questions
----------------

The main question, if you agree with adding folders, is: which folders should be added?

Current Solutions
------------------

I know of two things we are currently doing-ish that help with this:

1. The README groups capsules into reasonable categories.
2. Some capsules have the file name appended with `_driver` if they provide a `Driver` interface, or `virtual_` if they provide a virtualized hardware peripheral interface.

Initial Proposal
---------------

While the readme groups mostly based on what the device the capsule interfaces with does, I think a more useful folder-level grouping would be based on characteristics of the code. The characteristic that stands out the most to me is whether the capsule implements `Driver` or not. So, I think a start would be to create a `drivers` folder and put all of the capsules that provide a syscall interface in that folder.

In some cases capsules do multiple things, and in any organization scheme those capsules may fall into multiple buckets, so likely some case-by-case discretion would be needed.

ppannuto, 2018-07-17:
I think part of this should probably be a pass to re-evaluate our use of the word "capsule", for example the first sentence of that readme says all capsules are drivers:
> Capsules are drivers that live in the kernel and are written in Rust.

As we keep refining chips, it's looking like we can possibly remove the use of `unsafe`, which is another loose metric we've used to define a "capsule" before. That opens the question to why is a `sam4l` a "chip", but a `fxos8700` is not a "chip" (is the definition one self-contained file or not; is it "is-MCU"?)

Where this is going is a question of whether a `capsules` folder should exist at all, or should we have a top-level `drivers` folder, a top-level `ABIs` (for userspace-facing stuff?) folder, etc? Or is that probably too many top-level folders?

Basically, what exactly defines a "capsule", and can we use that to help refine what are appropriate sub-categories?

bradjc, 2018-07-18:
> Basically, what exactly defines a "capsule", and can we use that to help refine what are appropriate sub-categories?

One view of what a capsules is is it is any kernel rust code that does not fit in the other folders.

Since capsules are shared among all platforms, I'm not sure it makes sense to create more crates for them.

alevy, 2018-07-18:
Capsules are an isolation mechanism + runtime model that almost the entire kernel is composed of. Some of the capsules (those in `chips/`) are "specially privileged" in that they can use Rust's `unsafe`. 

The `capsules` directory is currently a misnomer that includes everything that doesn't belong anywhere else. We called it that originally just because :/

A better name would be "drivers" or "contrib"

@ppannuto conceptually, I think each module in capsule should be a crate, and it should (logically) live outside of the main Tock repo. Please don't yetll at me, I'm not suggesting actually having a separate repo for each one, just that logically they don't really belong together.

Now, it probably makes sense to group, e.g., all the sensor drivers into a crate, all the 6lowpan and 15.4 networking drivers into a crate, etc.

Finally we could keep all the mainline Tock cross-platform-drivers in a single crate and group them into modules that are thematic. That would be the most lightweight thing to do.

Re: workspaces, my understanding is that workspaces is just a way for multiple "projects" to share a build directory, and therefore compilation artifacts. So, if three boards used the cortex-m4, kernel, and (now) capsules crates, all with the same compilation options, those crates would each only be compiled once instead of once per board.

bradjc, 2018-07-27:
Is having folders within the capusle crate a useful proxy, or does making any change without going to a multiple crate structure not worth it? It's not clear to me what value having a whole host of separate crates affords users (and I know that having to add dependencies and update cargo.lock to use another capsule is a cost). Also, just adding folders makes it possible to add organization without requiring any changes outside of the capsule crate.

phil-levis, 2018-07-28:
I think this is exactly the kind of iterative change/churn that makes keeping up with Tock difficult and which I'd like to avoid.

bradjc, 2018-07-30:
> I think this is exactly the kind of iterative change/churn that makes keeping up with Tock difficult and which I'd like to avoid.

I want to note that a useful change can be made without any changes outside of the capsules crate.

closed 2018-12-06 by bradjc