tock/tock #1135: Tracking: Support RISC-V and HiFive1 Board

bradjc, 2018-07-26:
This RISC-V architecture continues to gain momentum, and having Tock support risc-v boards seems quite feasible. With the [embedded version still in development](https://en.wikipedia.org/wiki/RISC-V#ISA_Base_and_Extensions), it seems like Tock could be a leading embedded OS for risc-v platforms. Also, supporting multiple architectures would demonstrate the Tock architecture flexibility.

- [ ] Add support for risc-v (at least riscv32imac) in nightly rust. (https://github.com/rust-lang-nursery/embedded-wg/issues/36)
- [ ] Simplify binutils for riscv32. Ideally the LLVM tools that ship with rust will work. But if not we need some way to distribute riscv32-ld and riscv32-objcopy that doesn't rely on users building from source.
- [ ] Create a HiFive1 board in `/boards`.
- [ ] Implement `SyscallInterface` for risc-v. This would enable context switches to userland.
- [ ] Figure out the status of risc-v MPUs and ideally implement support for it/them.
- [ ] Add support to Tockloader for loading code on risc-v boards. It seems like they use openocd, but unfortunately it seems to be a custom fork.

bradjc, 2018-08-10:
The development for this is happening in the https://github.com/tock/tock/tree/riscv branch.

So far:
- [x] riscvimac crate
- [x] basic e310x chip
- [x] basic hifive1 board
- [x] UART TX working
- [x] GPIO set working
- [ ] GPIO interrupts (untested)
- [ ] RTC implementation
- [ ] alarm hil implementation

bradjc, 2018-08-17:
I'm having trouble figuring out how to implement the context switch for Tock. Now the HiFive E310 chip only has M-mode, which means we won't have the application isolation we want, but it will serve as a proof of concept.

I think there are three primary operations to make context switching happen:

### Switching to userland the first time

After the kernel is setup and the processes are loaded, the kernel must switch to the first process. I believe this can be done by setting the `mepc` register and calling the `MRET` instruction:

```asm
// Set mepc to the starting PC of the app.
// Here it is assuming the app entry point is at address 0x20c00000.
lui a0, %hi(0x20c00000)
addi a0, a0, %lo(0x20c00000)
csrw 0x341, a0

// Now go to what is in mepc.
mret
```

The `mret` instruction changes the privilege level before it executes code at the new location. This doesn't matter on a platform with only M mode, but it sets up an easier transition to a chip with user mode.

### Returning from the app to the kernel.

To call a syscall the app has to return control to the kernel. I believe the app can do this with the `ECALL` instruction.

### Handling the syscall and returning to the app

When the app calls the `ECALL` instruction that causes the trap handler to execute (which is configured by setting the `mtvec` register) with `mcause` set to "Environment call from M-mode". The trap handler can save the register state and handle whatever else it needs to, but it also needs to return execution to the instruction after the jump to the userland app. That is, the `switch_to()` function needs to be able to return. This is what I am unsure of how to do.

Using the `MRET` instruction at this point would, if I understand correctly, switch back to the app.

anmolsahoo25, 2018-08-27:
> ECALL and EBREAK cause the receiving privilege mode’s epc register to be set to the address of
> the ECALL or EBREAK instruction itself, not the address of the following instruction.

So based on this, we would have to set mepc to mepc + 4 and then mret which will then return to the userland app, if that is what is required. Setting mepc to the next instruction address is necessary.

DieracDelta, 2018-10-12:
Is this still in development on the riscv branch? If so, has anyone tried to test this on the riscv board? Did you manage to get openocd working well enough to actually load tock without userland/context switching onto the board?

closed 2020-07-15 by bradjc