tock/tock #1143: Tracking issue: Redesign (actually just design) kernel crate internals

alevy, 2018-08-03:
The discussion in #1113 is making quite clear that when I wrote the various parts of the kernel crate it was without putting much thought into how to logically structure it, but to just get something working.

i don't have specific things to track yet, but we should sit down and properly design it, since the mess I created is starting to hamper us.

bradjc, 2018-08-16:
Might consider this commit: 8b392f17f7ac8366dbd8e524946ee0592b2ad928. It changes what must be `mut` in the kernel.

bradjc, 2021-09-28:
I think this issue has run its course. For 2.0 major re-writes have happened and outstanding issues with grants and alignment have their own issues.

closed 2021-09-28 by bradjc