tock/tock #1147: Package name can potentially access all flash and ram

itf, 2018-08-07:
I haven't written code that exploits the bug yet, so it is just a theoretical bug.

## Summary
If you are using a TBF version 1, there is no check on the length of the package name, so it could be as large as all the flash memory available.  

Since the package name is public, every capsule could have unrestricted read access to all memory, using only safe code.

## Exploiting the bug

There are two ways of exploiting the bug:
One could create a User process using a TbfHeaderV1 with an arbitrary large package name length, and later on a Capsule could use its package name to read arbitrary memory. 

Or, a capsule could call the safe function "get_package_name" with a raw pointer that points to a fake TBF version 1 header and use it to read the data anywhere by using the package name offset and length.

## Code: 
Inside process.rs lines 267
``` rust
/// Name of the app. Public so that IPC can use it.
pub package_name: &'static str,
```
Now lines 590, and 704 inside the function create:
``` rust
let package_name = tbf_header.get_package_name(app_flash_address);
[...]
process.package_name = package_name;
```

And at last, inside the function ``` tbf_header.get_package_name```

``` rust
crate fn get_package_name(&self, flash_start_addr: *const u8) -> &'static str {
        match *self {
            TbfHeader::TbfHeaderV1(hd) => unsafe {
                let package_name_byte_array = slice::from_raw_parts(
                    flash_start_addr.offset(hd.pkg_name_offset as isize),
                    hd.pkg_name_size as usize,
                );
                let mut app_name_str = "";
                let _ = str::from_utf8(package_name_byte_array).map(|name_str| {
                    app_name_str = name_str;
                });
                app_name_str
            },
```

There are no checks for either pkg_name_offset nor pkg_name_size. 

## Proposed solutions

First the ```get_package_name``` should probably be made unsafe, and then either some package name size check should be placed (either in the get package name function, or in the parse_and_validate_tbf_header function) or the support for V1 header should be dropped.

bradjc, 2018-08-07:
We almost certainly should remove support for TBF header V1 as I can't imagine it is still used anywhere at this point.

As for the `get_package_name()` function, unfortunately we are going the wrong way with respect to that. #1113 changes how the process-specific code is handled, and actually makes way _more_ functions public through the `ProcessType` trait (although code would still need a reference to something that actually implements that trait to call the functions). What we ideally need is some way to make the `ProcessType` trait public, while making all of the functions of the trait only accessible within the crate.

alevy, 2018-08-08:
@itf excellent work and thank you!

This is not _currently_ exploitable by arbitrary capsules, since `get_package_name` has `crate` scope, and (untrusted) capsules are not housed in the kernel crate.

Nonetheless, all of your critiques are still valid as misusing this function inside the kernel crate seems likely, and it seems risky to _rely_ on it not being more visible (seems logical to expose this particular function at some point in the future in a careless attempt to make the API more useful).

We should _definitely_ remove header V1... it's beyond deprecated. We should also do a pass on marking functions inside the kernel crate unsafe unless they expose a safe API.

commit d67c9d5e (2018-08-17): Merge #1150

1150: kernel: remove TBF header v1 r=bradjc a=bradjc

V1 of the TBF header has been deprecated for a while now, and there is very little chance that any boards still have apps with the old tbf header version.

This also removes the PIC options field that is not supported by the kernel, not standardized, and not something we want the kernel to have to support in the future. It is also, as far as we know, unused.

~~Blocked because #1113 _needs_ to be merged.~~

Fixes #1147.

### Testing Strategy

This pull request was tested by compiling.

### TODO or Help Wanted

n/a

### Document [... cut]
PR #1150 (2018-08-17): kernel: remove TBF header v1
V1 of the TBF header has been deprecated for a while now, and there is very little chance that any boards still have apps with the old tbf header version.

This also removes the PIC options field that is not supported by the kernel, not standardized, and not something we want the kernel to have to support in the future. It is also, as far as we know, unused.

~~Blocked because #1113 _needs_ to be merged.~~

Fixes #1147.

### Testing Strategy

This pull request was tested by compiling.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relev [... cut]
closed 2018-08-17 by bors[bot]