tock/tock #1149: Race condition in context switching

ppannuto, 2018-08-09:
Both our old and new (#1113) context switching code do separate volatile reads and writes to clear the various flags set in fault handlers. Technically, this is a tiny window for a race condition. These should be using some atomic intrinsic (probably CAS).

It's a highly unlikely scenario and not worth blocking #1113 on, but wanted to note it to fix eventually.

alevy, 2018-08-09:
@ppannuto could explain a bit more? Something would have to be able to interrupt the flow between reading and writing those registers. What would that be?

ppannuto, 2018-08-16:
That's what I get for racing a review and plane landing. I hadn't completely thought through the issue, meant more as a save-for-later.

I was thinking of:
https://github.com/tock/tock/blob/master/arch/cortex-m/src/syscall.rs#L197

However I suppose it isn't possible for any of those three interrupts (except perhaps systick?) to fire, and as systick is the lowest priority it doesn't really matter.

closed 2018-08-16 by ppannuto