tock/tock #1151: Help Wanted: Shared components across multiple boards and chips

bradjc, 2018-08-13:
I started to look into how components could be shared by multiple boards. This would help board designers and capsule authors by making it easier to [correctly] include a new feature for a platform, and reduce the amount of duplicated code to include a capsule.

I ran into a issue where my attempt to make the types generic does not sit well with the Rust compiler. I tried starting with the LED component, and I was trying to use it for imix, hail, and nrf52:

https://github.com/tock/tock/blob/ff5456e2816698f2324e17e8bf33805b16bbde24/boards/components/src/led.rs#L23-L44

The idea is that the `Output` type is not fixed, and is instead a template value. When trying what I thought was my best attempt, I get this error:

```
   Compiling components v0.1.0 (file:///Users/bradjc/git/tock/boards/components)
error[E0401]: can't use type parameters from outer function
  --> /Users/bradjc/git/tock/boards/components/src/led.rs:40:31
   |
35 | impl<T: 'static + gpio::Pin + gpio::PinCtl> Component for LedComponent<T> {
   |      - type variable from outer function
...
38 |     unsafe fn finalize(&mut self) -> Self::Output {
   |               -------- try adding a local type parameter in this method instead
39 |         static_init!(
40 |             led::LED<'static, T>,
   |                               ^ use of type variable from outer function

error: aborting due to previous error

For more information about this error, try `rustc --explain E0401`.
error: Could not compile `components`.
warning: build failed, waiting for other jobs to finish...
error: build failed
make: *** [target/thumbv7em-none-eabi/release/imix] Error 101
```

I had no luck in trying to fix that error, or try a different approach. The full branch is here: https://github.com/tock/tock/compare/shared-components

Anyone have an idea of what structure might work with Rust? Or if I am just doing something silly?

bradjc, 2019-06-04:
Having this is still a dream of mine. Creating a new board would be so easy.

niklasad1, 2019-06-07:
Never seen this error before looks like a rustc bug to me, for example [this works](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=9b00411ae82bc7bbb6e392ddadfec1de)

bradjc, 2019-06-12:
Hmm...but I guess the problem is [this doesn't](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=42991b9294d7eba38c1dac62cd743274)

bradjc, 2019-06-12:
OK yes so the issue is that creating a `static` variable cannot be based on a type parameter. With a little finagling I have come up with this:

https://github.com/tock/tock/blob/635e7130d97f781582dc43a65fd62208ed88b699/boards/components/src/led.rs#L29-L82

...which is actually kind of reasonable. I'm not sure how to support different numbers of LEDs, but ignoring that for a minute this is what the main.rs code looks like:

https://github.com/tock/tock/blob/635e7130d97f781582dc43a65fd62208ed88b699/boards/imix/src/main.rs#L365-L368

...which is pretty close to what I was picturing when starting this. That annoying "helper" line is there only to statically allocate space for the pins array and the led capsule state. I _think_ that could be removed if the entire component was made in to a macro...but that might be too much.

Anyone have any comments? I'd like to try this on something more complicated (and find a way around the array length issue...) because while not ideal, it would still make setting up a new board pretty slick and hopefully much more foolproof than without components, while saving a lot of code duplication.

bradjc, 2019-10-02:
This has actually seen some progress in #1338 and should be something to merge after 1.4.

alevy, 2019-10-24:
Can we mark this closed?

bradjc, 2019-10-25:
No, we still don't have a way to do gpio, button, led, or any other variable length components.

hudson-ayers, 2020-01-08:
Can this be closed now? Or are there any other components that still need to be made generic? ADC maybe?

bradjc, 2020-01-08:
Yes! This PR is just to be able to have shared components, we will eventually actually use shared components.

closed 2020-01-08 by bradjc