tock/tock #1154: Tracking: Userland UDP Interface Milestone

hudson-ayers, 2018-08-13:
This issue dictates what tasks must be complete for the Userland UDP Interface milestone to be considered completed. Please add to this as needed.

- [ ] Clean up configuration so that all networking stack configuration happens either in main.rs or via userspace configuration calls. This includes creating some configuration file from which the src MAC, src PAN, RF channel, and src IP addresses are pulled. This also includes creating an interface so that the IP addresses of local interfaces can be accessed by userland apps via the udp driver.
- [ ] Fix driver check to only work if underlying hardware supports the networking stack
- [ ] Enforce port bindings to prevent one application from listening to messages intended for another. Add additional return methods to existing system calls to reflect failures due to unavailable ports
- [ ] Reach agreement that the interface between userland and the kernel is finalized
- [ ] Have someone other than Hudson use the finalized interface (hopefully while working on the Sensys tutorial)
- [ ] Test behavior when multiple userland apps use the driver simultaneously

bradjc, 2018-10-19:
> Reach agreement that the interface between userland and the kernel is finalized

With plans of doing a release in the next couple weeks, it seems unlikely (or at least I think it's unlikely? We still haven't stabilized a GPIO interface...) that we will come to a consensus on this, and we haven't gotten too many hours of use if you will on the interface either. Does it make sense to move this to a new milestone? I think userland UDP is more than enough of a milestone for the release, without having to say the syscall interface is set.

bradjc, 2018-10-31:
I think the only thing left in this issue is testing, which we need to do for the release anyway.

closed 2018-10-31 by bradjc