tock/tock #1160: Process faults

cmcavity, 2018-08-22:
When I comment out lines 246, 247 and 251 of sched.rs (i.e. never enable the mpu), and flash the rng test, I get normal output from the app, which is expected.

https://github.com/tock/tock/blob/d67c9d5e70ace179134bbf918525f85384985d33/kernel/src/sched.rs#L241-L251

However, when I now additionally comment out the body of setup_mpu in process.rs, a function which is not called anywhere else, the rng app faults with [this](https://drive.google.com/file/d/1v1Cq1_CzI_5DjZfXRBpUqSmxIx4ZdfJf/view?usp=sharing) error message.

The expected behaviour was normal output from the app. This test was performed on a Hail.

I did a git bisect, and found that "Merge #1115" is the first commit that introduces this issue. Also, @hudson-ayers has mentioned that he gets the same panic message when running certain test apps after he merged commits from master past "Merge #1113" into his IP Userland branch.

hudson-ayers, 2018-08-22:
For the record, the error message I get is basically identical to the one that Conor gets.

hudson-ayers, 2018-08-22:
Also, I get the error on Imix, while Conor/Danilo got the error on Hail

alevy, 2018-08-23:
@hudson-ayers is this also only happening for you when you change `sched.rs` or `process.rs`?

hudson-ayers, 2018-08-23:
@alevy nope, it happens when I simply resolve merge conflicts between the IP-Userland and master, and then try to flash apps

hudson-ayers, 2018-08-23:
@alevy In fact, I just tried to flash the ip_sense application onto the master branch (without any of my changes in the IP_userland branch at all) and that app also faults with the same error. So I am confident this is not due to any code changes of my own, given that ip_sense worked fine on master prior to #1113 being merged

hudson-ayers, 2018-08-23:
Below is a screenshot of the error message I get when flashing the IP_sense app onto the master branch of Tock.

[image]

alevy, 2018-08-24:
A few take aways from a debugging session yesterday:

First, I can replicate both the specific error @cmcavity pointed out on Hail by changing the kernel crate as specified, as well as the error @hudson-ayers encoutered on imix without any modifications, by simply running the `ip_sense` or `hello_loop` apps by themselves.

Second, I was able to "eliminate" the error on imix by removing either some drivers or specific calls to drivers. Minimally, commenting out `rf233.start()` in imix's `main.rs` gets rid of the error (of course it also breaks the 6lowpan stack). This, as well as other anecdotal evidence, points to some sort of alignment issue, where the important part isn't what code is being executed, but rather the layout/size/etc of the code or RAM.

Third, in the Hail case @cmcavity points out, I managed to track down the error to a corrupted entry in the GOT table, that seems corrupted by userspace. Note that in the error Connor posted it's an "Imprecise Data Bus Error," meaning the fault happened sometime after the faulting instruction. However, by turning off write-caching on the Cortex-M4, I could turn it into a Precise error and see the faulting instruction and address.

Fourth, in the imix case, the error doesn't make any sense. The faulting address is 0x04, but that address doesn't appear anywhere in the registers, and the faulting instruction is a `bx lr` returning from the `memop` procedure, with no loads or stores anywhere around it.

Onto stepping through with JTAG...

dcellucci, 2018-09-21:
If it counts, I also get the same error when I try to load `ble-uart` onto an NRF52840 

[image]

phil-levis, 2018-09-22:
Amit, can you post the binaries, ELFs, and your fault error when you can identify the instruction?

alevy, 2018-10-04:
This is the error I'm getting when I run the launchxl board from master with the `hello_loop` and `rng` apps installed. Note below that the error message claims `hello_loop` has faulted, but it's `rng` whos registers don't make sense (e.g. `pc` address in memory, and lacking a Thumb bit).

The "Imprecise Data Bus Error" flag is indicating that actually the fault may have happened in code that hadn't expired prior to the executing current instruction, so really we shouldn't trust the error message. Working on going down that rabbit hole.

```
[RNG]                                                                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                                          
Kernel panic at /home/alevy/hack/tock/kernel/src/process.rs:417:   
        Kernel version release-1.2-2018-06-549-g7634ef08                                                                                                                                                                                                                       
                                                           
---| Fault Status |---                                     
Data Access Violation:              true                   
Imprecise Data Bus Error:           true                   
Forced Hard Fault:                  true   
Faulting Memory Address:            0x00000004           
Fault Status Register (CFSR):       0x00000482
Hard Fault Status Register (HFSR):  0x40000000           
                                    
---| App Status |---                                       
App: hello_loop   -   [Fault]                              
 Events Queued: 0   Syscall Count: 6   Dropped Callback Count: 0
 Restart Count: 0                                          
 Last Syscall: None                                        

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20005000═╪══════════════════════════════════════════╝
             │ ▼ Grant         504 |    504
  0x20004E08 ┼───────────────────────────────────────────         
             │ Unused                      
  0x20003083 ┼───────────────────────────────────────────
             │ ▲ Heap       4294964803 |   5064 EXCEEDED!     S
  0x20003A40 ┼─────────────────────────────────────────── R      
             │ Data            576 |    576               A
  0x20003800 ┼─────────────────────────────────────────── M
             │ ▼ Stack         200 |   2048
  0x20003738 ┼───────────────────────────────────────────
             │ Unused                                    
  0x20003000 ┴───────────────────────────────────────────
             .....                                  
             │ App Flash      8144                        L                                                                                                                                                                                                                      
  0x00030030 ┼─────────────────────────────────────────── A                                                                                                                                                                                                                               
             │ Protected        48                        S                                                                                                                                                                                                                               
  0x00030000 ┴─────────────────────────────────────────── H                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                                          
  R0 : 0x00000014    R6 : 0x00000004                                                                                                                                                                                                                                                      
  R1 : 0x00000000    R7 : 0x2000386C                                                                                                                                                                                                                                                      
  R2 : 0x000001B4    R8 : 0x00000000                                                                                                                                                                                                                                                      
[... cut ...]
  R2 : 0x00000000    R8 : 0x20006158
  R3 : 0x0000000F    R10: 0x00000000
  R4 : 0x00000000    R11: 0x00000000
  R5 : 0x00000000    R12: 0x20005A74
  R9 : 0x20005800 (Static Base Register)
  SP : 0x20005490 (Process Stack Pointer)
  LR : 0x00032437
  PC : 0x20006140
 YPC : 0x00032418

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 1 1
 EPSR: ICI.IT 0x0e
       ThumbBit false !!ERROR - Cortex M Thumb only!
To debug, run `make debug RAM_START=0x20005000 FLASH_INIT=0x32051`
in the app's folder and open the .lst file.
```

phil-levis, 2018-10-11:
One not-very-helpful observation: every case I've run into where a process has faulted in this way, its debug output has the heap size wrong, and a huge number. In some cases it's not evenly divisible by the word size.

phil-levis, 2018-10-11:
@dcellucci 's example looks different. It seems like allow() was passed a problematic address; the faulting address is a small offset from the allow() address, which is not within the process address space.

phil-levis, 2018-10-11:
I compared the working and faulting versions of the @hudson-ayers kernel; the code layout is totally different. In the working kernel, the first functions are do_process and other core kernel functionality. In the faulting kernel, the first functions are for the MAC layer.

One thing that's frustrating is #1115 only changes the fault output.  So in theory it shouldn't be *causing* faults. I'm going to have to wait until Friday (can get my JTAG from the office) to make further progress.

phil-levis, 2018-10-11:
I'm worried that the process outputs don't make sense in many different ways (faulting address on a bx instruction, address is nowhere; corrupted pc registers), depending on the instance of the error, and this means there are bugs in the fault output code.

closed 2018-10-13 by bors[bot]

phil-levis, 2018-10-13:
WOOOHOOOO!

commit 6dfb1afa (2018-09-01): Merge #1164

1164: Fix undefined behavior in context switch that caused faults r=ppannuto a=alevy

### Pull Request Overview

Closes #1160

Under some circumstances, the process GOT or stored registers seem to get corrupted (details for how to reproduce in #1160). It's been a very difficult bug to isolate since it doesn't appear (or doesn't appear in the same way) when debugging. It also is highly tied to the particular kernel configuration, which implies it probably has something to do with how kernel flash/memory is laid out, i.e. it might be an alignment issue with some structure.

The two  [... cut]
commit a5d96e01 (2018-10-13): Merge #1185

1185: Fix assembly ordering in context switching code r=alevy a=alevy

### Pull Request Overview

Closes #1160 

This pull request fixes a bug in the context switching code that appears under certain circumstances where the compiler reorders volatile writes before inline assembly (which was _not_ marked volatile). The fix is to mark them volatile as well as move the non-assembly code in hardfault_handler to a separate function so we can make `hardfault_handler` `naked` like the other ISRs.

This can lead to defer execution of system calls or handling of faults which _seems_ to int [... cut]
PR #1185 (2018-10-13): Fix assembly ordering in context switching code
### Pull Request Overview

Closes #1160 

This pull request fixes a bug in the context switching code that appears under certain circumstances where the compiler reorders volatile writes before inline assembly (which was _not_ marked volatile). The fix is to mark them volatile as well as move the non-assembly code in hardfault_handler to a separate function so we can make `hardfault_handler` `naked` like the other ISRs.

This can lead to defer execution of system calls or handling of faults which _seems_ to interact in such a way that sometimes a [... cut]