tock/tock #1167: Re-design of UART HIL

phil-levis, 2018-09-21:
This is intended to supersede #1072 . 

Following the agreed-on approach of "only touch a HIL if you think it'll never need to be touched again", that is, trying to keep HILs stable, here's a proposed redesign of the UART HIL. Here are the proposed traits:

```
// Was initialize — changed name to be consistent with Pat’s distinction between
// initialization and configuration
pub trait Configure {
    fn configure(&self, params: UARTParams) -> ReturnCode;
}

pub trait Transmit {
    fn set_client(&self, client: &'static TxClient) -: ReturnCode;
    fn transmit(&self, tx_data: &'static mut [u8], tx_len: usize) -> ReturnCode;
    fn abort_transmit(&self) -> ReturnCode;
}

pub trait Receive {
    fn set_client(&self, client: &'static Client) -> ReturnCode;
    fn receive(&self, rx_buffer: &'static mut [u8], rx_len: usize) -> ReturnCode;
    fn abort_receive(&self) -> ReturnCode;
}

pub trait TransmitClient {
    fn transmit_complete(&self, tx_buffer: &'static mut [u8], rval: ReturnCode);
}

pub trait ReceiveClient {
    // ReturnCode is for generic control/system errors; Error is for UART-specific
    // information, such as parity errors, etc.
    fn receive_complete(&self, rx_buffer: &'static mut [u8], rx_len: usize, rval: ReturnCode, error: Error);
}
```

One useful point is that the fact that reception requires passing a buffer with `Receive::receive`. As a result, there is a chance of a race condition if a new packet arrives between actual reception and `receive` being called again in response to`receive_complete` being called. This implies that there is some sort of buffering underneath (e.g., within chip memory, or a kernel buffer).

phil-levis, 2018-10-03:
Here is a link to the current revision of the UART trait:

https://github.com/tock/tock/blob/uart-redesign/kernel/src/hil/uart.rs

One pending change is a note that calls to cancel() MUST NOT return SUCCESS if the current outstanding call is a buffer call. Otherwise, there will be no callback that returns the buffer to the caller. So SUCCESS can only be returned for word operations.

bradjc, 2019-06-04:
#1211

closed 2019-06-04 by bradjc