tock/tock #1176: Hard Fault Handler Pushes to Stack

hudson-ayers, 2018-10-01:
Currently, overflowing the kernel stack results in the hard fault handler being called, as the MPU ensures only valid memory regions can be accessed. However, the hard fault handler immediately pushes two registers onto the stack, resulting in yet another hard fault. This manifests in a complete lockup of the device, with no panic or debug info printed to the console, and no blinking LEDs to indicate a panic. 

Stepping through such a fault in gdb only shows that the board has halted within `hard_fault_handler()` in lib.rs, along with the message "clearing lockup after double fault". Further, backtrace does not work after this has occurred, as gdb detects that the stack has been corrupted. As a result, the only definitive clue that a stack overflow has occurred is that printing the registers reveals that the stack pointer is below 0x20000000. Stack overflows should lead to clear debug messages.

bradjc, 2019-10-02:
I looked at this, and couldn't see any easy way to _not_ have the handler push to the stack.

ppannuto, 2019-10-02:
We've talked about this before and I think we have a solution to it; the
basic premise is to reserve a small region of memory as kernel "emergency
stack" that's only ever used for a lightweight routine that HF handler can
use. Requires rewriting the beginning of the handler in assembly and
immediately setting stack pointer to the reserve region (or using no stack)

commit 93f799aa (2020-07-20): Merge #1965 #2002

1965: hil: gpio: remove 'static r=ppannuto a=bradjc

### Pull Request Overview

Part of #1074.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.

2002: cortex-m: Detect stack overflow r=ppannuto a=bradjc

### Pull Request Overview

This pull request adds logic to the kernel hardfault handler to detect a stack overflow condition, and resets the stack before using it. With IPC on hail I'm able to trigger this case and verify  [... cut]
PR #2002 (2020-07-20): cortex-m: Detect stack overflow
### Pull Request Overview

This pull request adds logic to the kernel hardfault handler to detect a stack overflow condition, and resets the stack before using it. With IPC on hail I'm able to trigger this case and verify I get the panic handler.

This also moves the userland state saving down below since I'm not sure why that code was run in both the kernel and userspace fault case.

Fixes #1176.

### Testing Strategy

Verifying that IPC on hail causes this panic handler to be displayed.

### TODO or Help Wanted

1. Someone to look over the assembly.

2. I can  [... cut]
closed 2020-07-20 by bors[bot]