tock/tock #1199: How to debug hail by using Jlink under Openocd.

hujiaodigua, 2018-10-29:
I have a problem with how to debug hail. At the lower left of the top layer, I can see the SAM4L_SWDCLK, SAM4L_DIO and GND test points. But,  I don't know what is the six test point and three holes mean? Test points or a kind of interface?
Please help me.

hujiaodigua, 2018-10-29:
And, is there any documentation on debugging about Openocd and GDB?

bradjc, 2018-10-29:
That connector is for the 6 pin tag connect adapter: http://www.tag-connect.com/catalog/6. Those pins allow for a [j-link jtag adapter](https://www.segger.com/products/debug-probes/j-link/) to connect to the SAM4L.

There is a quick note about debugging in the readme: https://github.com/tock/tock/tree/master/boards/hail#debugging-the-kernel. Once GDB is setup it should work like any GDB environment.

closed 2018-10-29 by alevy