tock/tock #1201: Release 1.3

bradjc, 2018-11-01:
Testing

- Hail
  - [ ] Basic apps (blink, printf_long, c_hello, etc).
  - [ ] hail test app
  - [ ] IPC
  - [ ] witenergy
  - [ ] process console
  - [ ] adc test app
  - [ ] accel_leds
  - [ ] ble-uart
  - [ ] rot IPC
  - [ ] tutorial 5 ipc
  - [ ] sensors app
  - [ ] console timeout
  - [ ] tests/mpu*

phil-levis, 2018-11-02:
One critical issue for me is the MPU in cortex-m4. I can lead moving it into cortex-m, if it is generic across the cortex-ms? Putting a copy in cortex-m3 and cortex-m0 is also OK, but clearly less desirable.

bradjc, 2018-11-13:
Here is a buggy app that displays the behavior (well I think it does, it at least seems to panic): https://github.com/tock/libtock-c/tree/buggy-app/examples/tests/malloc_test01

bradjc, 2018-11-15:
Ok we tracked down the issue with the IPC apps (luckily it was contained in userspace, so it doesn't actually change anything in the kernel).

bradjc, 2018-11-20:
Hail
===

Testing for Hail on tag `release-1.3-rc1`.

- [ ] `examples/sensors`
  - sense_to_console: Application that samples a sensor and outputs values to console
  - In development: https://github.com/tock/libtock-c/pull/26
- [ ] `examples/c_hello` and `examples/tests/printf_long`
  - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
- [ ] `examples/blink`
  - blink: blinks LEDs
- [ ] `examples/rot_client` and `examples/rot_service`
  - rot_ipc: tests IPC with a simple service
- [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
- [ ] `examples/lua-hello`
- [ ] `examples/tests/console_timeout`
- [ ] `examples/tests/malloc_test01`
- [ ] `examples/tests/stack_size_test01`
- [ ] `examples/tests/stack_size_test02`
- [ ] `examples/tests/mpu_stack_growth`
- [ ] `examples/tests/mpu_walk_region`
- [ ] `examples/tests/multi_alarm_test`
- [ ] `examples/tests/adc`
- [ ] `examples/tests/adc_continuous`
- [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`

- Miscellaneous
  - Test the apps compiled at 1.0
  - [ ] process console works as expected

phil-levis, 2018-11-20:
_Do we want to upgrade the console capsule, given:_

Let's not block on this. There are plenty of system call interfaces that don't virtualize. This seems like a good milestone for 1.4: virtualize every system call and HIL.

alevy, 2018-11-27:
NRF52-DK
=========

Testing for NRF52 development-kit on tag `release-1.3-rc2`.

- Userspace
  - [ ] `examples/sensors`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
 
- NRF52dk
  - [ ] `examples/ble_advertising`
  - [ ] `examples/ble_passive_scanning`

- Miscellaneous
  - Test the apps compiled at 1.0
  - [ ] process console works as expected

alevy, 2018-11-27:
LaunchXL CC26x2
===============

Testing for Launchxl development-kit on tag `release-1.3-rc2`.

- Userspace
  - [ ] `examples/sensors`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
        - _worked_, but I was unable to program using `tockloader install`, instead I had to manually combine the binaries and flash them as a unit. Otherwise installing one after the other seemed to erase both apps. Likely a bug in `tockloader`.
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
 
- Launchxl

- Miscellaneous
  - [ ] Test the apps compiled at 1.0

bradjc, 2018-11-27:
Wait, is anyone testing the cortex-m0? That seems like an important thing to do...

bradjc, 2018-11-29:
@hudson-ayers @phil-levis Can someone test the imix app on imix? On mine the button doesn't seem to control the LED, and a very quick look at the code implies that it should.

hudson-ayers, 2018-11-29:
@bradjc I just ran it on my Imix - the button does control the LED just fine, but it doesn’t print sensor values to the console like I expect.

alevy, 2018-11-29:
@hudson-ayers that is _likely_ because it is blocking on the non-existent ninedof sensor, which is not on the new hardware revision, but is none-the-less included as a driver in the `imix` board configuration. So the app queries to see if the driver exists (it does), and then tries to read from the sensor (which blocks indefinitely).

hudson-ayers, 2018-11-29:
@bradjc if I comment out sample_sensors() and send_ieee802154_packet() and instead put a printf() statement in the while loop, the print statement doesn’t print immediately...nothing prints for about 30 seconds, then a backlog of about 30 prints gets printed to the console all at once (even though there should be a delay of 1000ms between each print).

bradjc, 2018-11-29:
Ok with a little debugging it seems that the imix app works enough so that the button controls the LED if have the imix app installed and the unplug and replug the board. It does not work after pressing the reset button, or after running `tockloader listen` (which effectively pushes the reset button).

hudson-ayers, 2018-11-29:
Okay, my issue with the printf() was forgetting a \n ... my bad. Now, my app works if I comment out sample_sensors(). AND it works on the first install, and works after running tockloader listen.

hudson-ayers, 2018-11-29:
I tried a second board (a previously untested one from the new batch), however, and saw exactly what you describe, so it does sound like this may be somewhat hardware related..both boards I tried were from the new batch as well.

alevy, 2018-11-29:
NRF51-DK
=========

Testing for NRF51 development-kit on tag `release-1.3-rc3` (+ minor modifications to be pushed).

- Userspace
  - [x] `examples/sensors`
  - [x] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [x] `examples/c_hello` and `examples/tests/printf_long`
  - [x] `examples/blink`
  - [x] `examples/rot_client` and `examples/rot_service`
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
        - _worked_, but I was unable to program using `tockloader install`, instead I had to manually combine the binaries and flash them as a unit. Otherwise installing one after the other seemed to erase both apps. Likely a bug in `tockloader`.
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
  - [x] `examples/tests/adc_continuous`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/accel_leds`
  - [x] `examples/ble-uart`
 
- NRF51dk
  - [x] `examples/ble_advertising`
  - [x] `examples/ble_passive_scanning`

- Miscellaneous
  - [x] Test the apps compiled at 1.0

bradjc, 2018-11-29:
> Now, my app works if I comment out sample_sensors().

Ok @hudson-ayers it seems like this is a problem with hw and the app, and not the kernel so it won't block 1.3. I do think you should update the imix app so that it works on this hw rev, but that is not pressing.

bradjc, 2018-11-29:
Sign-offs on 1.3
---------------

Checking the box next to your username indicates that you are satisfied with the latest Tock release candidate and believe it should be tagged as the 1.3 release. When all of these are checked 1.3 will be done!

- [ ] @alevy 
- [ ] @bradjc
- [ ] @brghena 
- [ ] @hchiang 
- [ ] @hudson-ayers  
- [ ] @niklasad1 
- [ ] @ppannuto 
- [ ] @phil-levis

hudson-ayers, 2018-11-29:
@bradjc I agree. All of the individual components of the app (BLE, 15.4, sensors, button, LED) work fine independently, so I think the problem is with the app. I'll rewrite the Imix app when I get the chance, but that can be post 1.3.

phil-levis, 2018-11-30:
I'm going to finish my tests tonight.

ppannuto, 2018-11-30:
> @ppannuto <https://github.com/ppannuto>
>
> Currently, when downloaded as a zip file, this line:
>
>
> https://github.com/tock/tock/blob/efeaa77c70ac2609b5ae331b816b5d8f6ef89444/boards/Makefile.common#L53
>
> Causes fatal: not a git repository (or any of the parent directories):
> .git to be printed when a board is built, and then sets
> TOCK_KERNEL_VERSION to something other than "1.3" (or similar). How do we
> fix that?

$ git describe --always 2> /dev/null || echo notgit
notgit

bradjc, 2018-11-30:
> $ git describe --always 2> /dev/null || echo notgit

That is half of the issue, but presumably a user would expect that after downloading version 1.3,

https://github.com/tock/tock/blob/efeaa77c70ac2609b5ae331b816b5d8f6ef89444/arch/cortex-m4/src/lib.rs#L242

wouldn't show `notgit`.

ppannuto, 2018-11-30:
I mean, we could do something like "echo v1.3-sourcedist", at a certain
point there's not so much magic we are capable of doing

brghena, 2018-11-30:
I approve. Tried running all the make commands from root of Tock repo. Loaded a kernel on Hail. But a bunch of random apps on Hail (`hail`, `hello_loop`, `writeable_flash_regions`, and `crash_dummy`) and everything still works.

bradjc, 2018-11-30:
https://github.com/tock/tock/releases/tag/release-1.3

Woo! Thanks all!!!

closed 2018-11-30 by bradjc