tock/tock #1225: Ensure Errors on Boot Propogate to main.rs

bradjc, 2018-11-29:
Errors like a process not fitting in RAM should make it to a board's `main.rs` so the user can be informed.

phil-levis, 2018-11-29:
I'd change this slightly: make it so that main.rs decides how to handle errors.

For example, if you actually want to propagate process loading errors to main.rs, you will need to be able to pass a list (or array, or slice) of (error, string) tuples back, so that main.rs can correctly assess each process loading error. The string is the name of the process; this seems like the easiest way to associate errors with apps. The same is true for capsules: unless we want to incorporate error handling into the core loop, the capsules need to locally report errors.

However, it can be that main.rs decides how error reports are handled. E.g., there is a default DebugWriter that outputs only kernel errors, but you can configure it to output capsule errors as well. This configuration could even be at runtime (e.g., via the process console). The one slightly tricky part is that you want to be able to statically specify that certain errors will not print, such that their logic/strings are optimized out of the binary. So you want to be able to statically specify which errors are possible, but be able to dynamically control which of the possible errors are actually output.

Also, popping up a level, while I think it's a laudable goal that the kernel never panics, I don't think that's realistic. There will be cases when state is corrupted in a way that it's unclear how to possibly go forward. If we say that panics can only occur on hard faults from the kernel,  we're going to push a lot of complexity into implementations.

I'll use the RF233 stack as an example, because this came up when I first implemented it. Here's a small snippet of code:
```
fn read_write_done(
        &self,
        mut _write: &'static mut [u8],
        mut read: Option<&'static mut [u8]>,
        _len: usize,
    ) {
        self.spi_busy.set(false);
        let rbuf = read.take().unwrap();
```

The important line is the `unwrap()`, which can of course panic. The key thing is that the RF233 always calls read_write with the optional read buffer. So if the read buffer is not returned, then the SPI implementation has lost it. If the SPI implementation has lost it, then the RF233 can be totally broken and can likely no longer function. E.g., it has lost the read buffer it stores the result of register reads into. If we wanted to handle this case in such as way that the RF233 fails gracefully and doesn't propagate across the system, then it's going to 

  1) need guards everywhere to check if it's in a fail state (to return errors on subsequent calls)
  2) complete/unwind any outstanding operations by it or on it (so being perpetually blocked doesn't block other parts of the system)

Doing both of these is going to make the code a lot more complex and larger. So the implementation takes the position that if the SPI implementation does something this bad, all bets are off and you should just fail fast.

phil-levis, 2018-12-04:
I don't think this should be a release-blocker.

ppannuto, 2020-04-21:
Closed by bf5610f861

closed 2020-04-21 by ppannuto