tock/tock #1233: Re-design of GPIO HIL

bradjc, 2018-12-03:
Please edit to include any concerns. Making issue for 1.4 tracking.

ppannuto, 2018-12-03:
- Low-power-listening capable for interrupts

This is what burned / is burning us on Signpost for GPIO :/

phil-levis, 2019-01-04:
@ppannuto can you explain? Is this an API problem or an implementation problem?

phil-levis, 2019-01-04:
Here's a first cut at a slightly updated GPIO HIL. Feedback welcome. Key points are merging PinCtl and Pin and removing identifier from interrupts (since it forces storage in the implementation and can be layered as a very simple wrapper). I was trying to find cases when we want to separate PinCtl and Pin and it seemed like most implementations just combined them. If we want to separate out control and data, then I'd suggest more than just pull down/pull up; we should also consider read/write configuration, etc.

This API leaves function-specific configuration out of the HIL, since it's very chip-specific, but does expose if a pin is in such a state.

```
pub enum InputMode {
    PullUp,
    PullDown,
    PullNone,
}
pub enum InterruptMode {
    RisingEdge,
    FallingEdge,
    EitherEdge,
}
pub enum PinState {
    Disabled,
    Input,
    Output,
    InputOutput,
    Function, // Chip-specific, requires chip-specific API for more detail
    Unknown,
}

/// Interface for synchronous GPIO pins.
pub trait Pin {

    fn input_mode(&self) -> InputMode;
    fn set_input_mode(&self, _: InputMode);

    /// The current configuration state of the pin.
    fn state(&self) -> PinState;

    /// Configure the GPIO pin as an output pin.
    fn make_output(&self) -> PinState;

    /// Configure the GPIO pin as an input pin.
    fn make_input(&self) -> PinState;

    fn is_input(&self) -> bool;

    fn is_output(&self) -> bool;

    /// Disable the GPIO pin and put it into its lowest power
    /// mode.
    fn disable(&self);

    /// Set the GPIO pin high. If the pin is not an output or
    /// input/output, this call is ignored.
    fn set(&self);

    /// Set the GPIO pin low. If the pin is not an output or
    /// input/output, this call is ignored.
    fn clear(&self);

    /// Toggle the GPIO pin. If the pin is not an output or
    /// input/output, this call is ignored. Return the new value
    /// of the pin.
    fn toggle(&self) -> bool;

    /// Get the current state of an input GPIO pin. For an output
    /// pin, return the output; for an input pin, return the input;
    /// for disabled or function pins the value is undefined.
    fn read(&self) -> bool;
}

pub trait Interrupt {
    /// Set the client for interrupt events.
    fn set_client(&self, client: &'static Client);
    
    /// Enable an interrupt on the GPIO pin. This does not
    /// configure the pin except to enable an interrupt: it
    /// should be separately configured as an input, etc.
    fn enable_interrupt(&self, mode: InterruptMode);

    /// Disable the interrupt for the GPIO pin.
    fn disable_interrupt(&self);
}

pub trait Client {
    fn fired(&self);
}
```

phil-levis, 2019-01-17:
We discussed low power interrupts on the 1/16 call. I made the point that adding this to the traits might be a big leap; there are lots of peripherals and subsystems which have implications to low power operation, and so if we want to include this then we should do so in a comprehensive and coherent way. We agreed that this is something which should be indicated at compile-time, rather than run-time. I.e., you want this information to know if you're designing your code in a way that precludes low power operation, not just so that you can at runtime adapt your behavior to reduce energy consumption.

The tentative conclusion is that this signaling power information might be best accomplished by changing how chips present their GPIO pins. In particular, for the SAM4L, rather than presenting the pins by ports, the chip could present two arrays of GPIO pins, LowPowerPins and NoLowPowerPins, or something similar. That way, when you access  a particular pin or pass it in to a structure, you know whether you are impacting low power operation.

It's worth noting that in the case of Signpost I don't think this would have helped, unless you would have chosen your hardware pins based on the operating system abstractions?

ppannuto, 2019-01-17:
I think this is a sensible conclusion. I agree that it doesn't really fit in the GPIO HIL. That sounds like a reasonable possible answer for how to handle this in the kernel, though I think it's part of a bigger problem of energy auditing and debugging that would be nice to tackle somehow -- it's quite surprising that adding a button callback to an app increases power draw by an order of magnitude on hail! -- but that's orthogonal to the current effort and is something with can discuss another time.

ppannuto, 2019-01-17:
Looks good to me overall, one thought:

Should `    fn configuration(&self) -> Configuration;` possibly take an `Option<Configuration>` parameter -- this would allow code to opaquely capture and restore prior configuration?

phil-levis, 2019-07-19:
That's an interesting idea -- what's the use case?

ppannuto, 2019-08-16:
Closed by #1297.

closed 2019-08-16 by ppannuto