tock/tock #1257: Kernel fault: reading from userspace stack pointer without validating it first

jrvanwhy, 2019-01-08:
A faulting userspace application can make the kernel read from an application-provided address.

Reproduction steps:
  1. If the application wants to trigger a read at address X, it sets its stack pointer to X + 32
  1. The application faults (e.g. by reading from an invalid address)
  1. The hardware decrements the application's stack pointer by 32 bytes.
  1. The kernel's fault handling logic passes the application's stack pointer to cortex-m::SysCall::process_detail_fmt, which reads from its address: https://github.com/tock/tock/blob/4b00dfda4f064f26e11c3d877be77d6602e9be26/arch/cortex-m/src/syscall.rs#L387
  1. If X is not a kernel-readable address, the kernel faults.

The kernel should probably confirm that the stack frame it is reading is within the application's memory space before reading it.

bradjc, 2019-06-13:
I assume the internal operation that places the register content on the stack is subject to MPU rules? it has to be. If we can guarantee that if an app does put its stack pointer somewhere that is invalid (as enforced by the MPU), and that when the core tries to push registers there that `switch_to_process()` detects that as an app fault, then adding a check in process.rs after calling `switch_to_process()` to verify that the returned stack pointer is in the app's valid memory space will prevent the kernel from reading from a non-app memory address.

bradjc, 2021-02-01:
I believe this is resolved with #2383

hudson-ayers, 2021-02-17:
I agree that #2383 resolved this on master, closing

closed 2021-02-17 by hudson-ayers