tock/tock #1274: [RFC] Create a read-only-allow

torfmaster, 2019-02-11:
This is possibly a duplicate of https://github.com/tock/tock/issues/418 but however, it is possibly more concrete:

# Use cases
By default .rodata stays in flash and currently we create an object on the heap to allow the capsules to access it. It has the following advantages:
 - we don't have to take care, in which memory segment a variable resides, from the standpoint of of the Rust compiler any reference to a `&[u8]` is okay.
 - we don't have to copy data from flash to memory.
Currently, we get an access violation when allowing flash memory. As `str`s currently riside in .rodata this happens quite easily.

# Implementation
On the kernel side I would propose either two types: An AppSlice and an AppSliceMut or some abstraction covering both. For simplicity and for backwards compatibility I would propose two implement another syscall, as the use cases for allow and allow_mut inside the same driver would be rather disjoint.

I'd try to implement something in this direction if there is a consensus that this is a good idea.

Woyten, 2019-02-11:
In my opinion, this is a very important soundness feature. Especially for libtock-rs!

alevy, 2019-02-11:
Yes! This has come up in a few discussions over the years, and recently has posed a concrete problem for @phil-levis-google while implementing U2F, because it adds undo memory pressure to copy data into RAM.

ppannuto, 2019-02-12:
Another +1 from me, I've wanted this for years, from when I first wrote
this comment:
https://github.com/tock/libtock-c/blob/e26203e251c0b0834f5bd670973b09027e19e13f/libtock/console.c#L77
(and
the few other places the same thing happens in libtock-c)

bradjc, 2020-01-22:
Just making a note that a PR for this is still of interest.

hudson-ayers, 2021-02-12:
This has been implemented in the tock 2.0 branch (`tock-2.0-dev`)

closed 2021-02-12 by hudson-ayers