tock/tock #1288: Define a syscall interface for applications to make a RAM segment executable.

jrvanwhy, 2019-02-27:
We intend to change the default permissions of application RAM to be non-executable ([issue 1287](https://github.com/tock/tock/issues/1287)). We may need a mechanism for applications to mark memory as executable in case they need to do dynamic code generation.

Open questions:

1. Do we want to allow applications to mark their own memory executable?
2. If yes, what should the API look like (e.g. a memop() command?)

rajivr, 2019-02-28:
> 1. Do we want to allow applications to mark their own memory executable?

From slack conversation, it seems like the use-case for `rwx` memory is to support JIT based applications @alevy said its not used much, which I can understand.

If JIT based applications doesn't have users, we could mark app memory as `rw-`, till a compelling use-case shows up. This would be a simple patch.

> 2. If yes, what should the API look like (e.g. a memop() command?)

If we decide to support both `rwx` and `rw-` memory, then my preference would be to modify TBF header to support it.

jrvanwhy, 2019-02-28:
> If we decide to support both `rwx` and `rw-` memory, then my preference would be to modify TBF header to add an option to indicate that memory type and change `Process::create`.

My preference is the opposite: I'd rather see it down at runtime with a syscall. A disadvantage of controlling execute permission using a TBF header is it cannot be updated at runtime, so the executable permission could not be disabled when unnecessary.

Also, would you want to whitelist a specific sub-range of the application's memory (in which case the decision of where the writable memory would be needs to be made at build time) or all memory (which would give less protection)? It's probably easier for applications to decide what memory they want to write at runtime than at compile time.

rajivr, 2019-03-01:
> My preference is the opposite: I'd rather see it down at runtime with a syscall. A disadvantage of controlling execute permission using a TBF header is it cannot be updated at runtime, so the [...]

That's correct. I would prefer to wait for a use-case to emerge where we need to go from `rw-` to `rwx` and back.

On a more general note - My preference would be to _limit_ the number of syscalls to bare minimum and if there is a way to address the use-case without introducing new syscalls. Once syscalls are introduced and users start depending on it, it is very hard to take it back, in case we got the design wrong. Also syscall interface is where bugs tend to happen, as the kernel evolves.

One of the many things that I like about Tock is its driver interface. It is succinct and seems to be very well thought through.

> Also, would you want to whitelist a specific sub-range of the application's memory (in which case the decision of where the writable memory would be needs to be made at build time) or all memory [...]

Yes. But this would complicate the MPU implementation and traits. I took a quick look at ARMv7-M and it seems like XN would need to be set on region basis, so we won't be able to have sub-regions with different XN. While it might be possible to do this with newer MPUs, the question would be how can we generalize it for older MPUs.

bradjc, 2023-06-16:
Is this still a goal? I haven't heard this come up in a long time; I assume this is not something we want to pursue.

jrvanwhy, 2023-06-16:
I don't think it's something we want to pursue either.

closed 2023-06-19 by bradjc