tock/tock #1289: [RFC] Real-time scheduling

thymbahutymba, 2019-03-01:
Hi guys, I recently found this project and I'm interested on it or better I wish to contribute somehow for implement real-time scheduling.
I try to explain my self better: I'm new in the rust world and I'm fascinated by it. I'm also studying the real time systems in the embedded ecosystem. I used for some project an OSEK/VDX hard real time operating system in C. Hence I asked to my self "I like both rust and embedded platform so why don't start some project with this purpose?" and it's why I found this project and I'm intrested in it.

Seems that tock doesn't support real time scheduling with multiple task. The idea that came to my mind is something like that:

```
...

#[task(20)]
fn my_first_task() {
    // Body of the task
}

#[task(50)]
fn my_second_task() {
    // Body of the task
}

...
```

Have an application with two tasks which respectively have a period of 20 and 50 milliseconds scheduled by Rate Monotonic with preemption, this just for start.

I wish to have some feedback from you and I want not reinvent the wheel since there is already embedded operating system.

alevy, 2019-03-01:
Hi @thymbahutymba , and welcome!!

> Seems that tock doesn't support real time scheduling with multiple task. The idea that came to my mind is something like that:

Tock currently supports multitask scheduling in _processes_, not in the kernel. The process scheduler, however, is pretty bad (it's not even really round robin, because it doesn't keep it's place in the queue whenever there is hardware interrupt). This would be a great place to start---implementing a better processes scheduler.

Having separate tasks inside the kernel would require having separate stacks for each task---basically implementing language level threads. Since Tock takes a different approach than typical embedded OSs---of explicitly separating kernel drivers from application code---it's not been totally clear in which cases multitasking _inside_ the kernel would actually be necessary.

Perhaps you could provide some motivating examples based on your experience building these kinds of systems that would help figure out what such a mechanism should look like, or whether it's something that should be accomplished in processes or in the kernel.

thymbahutymba, 2019-03-01:
Implementing a better "task" scheduler should be a good starting point. 

> Having separate tasks inside the kernel would require having separate stacks for each task

If we assume the basic idea of Rate Monotonic with fixed number of task started at time t = 0 (Earliest Due Date) there is no issue with the stack. 
Suppose that your application have three tasks (1, 2, 3) with the following periods 20, 50 and 100ms, the first task will have the highest priority, the second the mid priority and the last the lowest priority. Suppose also that at time t != 0 we have the second task running and a new activation of task one arrives, the task two can be preempet without any problem because once the task one is started it can not be preempet by a lower priority task, when it finishes the stack will be the same that the task two left and it can start from the point in which it has been stopped.

If we assume other stuff like fixed priority servers (sporadic server, deferrable server and polling server) there may be problems but we worry about it when the time comes.

> Perhaps you could provide some motivating examples based on your experience building these kinds of systems that would help figure out what such a mechanism should look like, or whether it's something that should be accomplished in processes or in the kernel.

There are a lot of examples that can be made. Imagine something like a thermostat: your embedded system read value from different kind of sensor and start heating you house if the temperature is below some minimum value. In this system there is also a humidity sensor that when the value is too high it starts some others things that bring the humidity to desidered value. So in your system you could have one task for reading from sensor, one actuator task, one task for printing some informations into an LCD display, one other task for taking input from some buttons that allows to set the desidered humidity, temperature and so on.

You can find a lot of real time applications that need for periodic task to read sensor and activate the actuator.

I hope that this simple example explain to you what are my point of view and what is in my mind!

alevy, 2019-03-02:
So, the thermostat use case you've given is currently supported in Tock, without preemptable kernel tasks, in two ways:

## In the kernel

At a high level, you've described periodic, independent tasks, but there aren't really real time requirements for any of them. If you sample the temperature at time 142:12:00:00:00 (minutes:seconds:ms:us) or 142:12:00:04:34 (i.e. within a few milliseconds), it doesn't really matter (ambient temperature or humidity doesn't change anywhere near that quickly).

You can implement this entirely in the kernel with capsules:

- All I/O is asynchronous so capsules don't block each other from making progress
- Capsules can access virtual alarms to be invoked periodically

## In processes

In practice, you'd implement much of this functionality in processes.

Capsules in the kernel take care of low-level details of particular sensors---e.g. formatting an I2C packet and sending it (asynchronously) a command to the temperature sensor to request a reading. The kernel exposes relatively high-level APIs to processes to access sensors---`read_temperature`, `read_humidity`, `button_subscribe`.

Those processes are preemptively scheduled, so even if they are doing very expensive computations (I'm not sure what they would be in this case), each process cannot prevent the others from receiving data from the sensors they've subscribed to, or otherwise making progress.

Of course, having a better _process_ scheduler, would still be useful in this case.

thymbahutymba, 2019-03-03:
I think that you misunderstand what I mean. The thermostat  example was just an example, of course there are a lot of solution that doesn't require a real-time scheduler. Imagine a car, airplane, or train in these vehicles there MUST be a real-time scheduler that performs some measurements every Xms.

I post the definition of real time operating system taken from Wikipedia:
> A real-time operating system (RTOS) is any operating system (OS) intended to serve real-time applications that process data as it comes in, typically without buffer delays. Processing time requirements (including any OS delay) are measured in tenths of seconds or shorter increments of time. A real time system is a time bound system which has well defined fixed time constraints. Processing must be done within the defined constraints or the system will fail. They either are event driven or time sharing. [...]

So what is in my mind is to extend the scheduler for allowing the real time scheduling. In these day I thought at what I would want in the final program running on top of tock.

```
...
#[feature(rate-monotonic)]
// #[feature(deadline-monotonic)] or also EDF for the future or nothing for the normal scheduler

#[task(20)]
fn my_first_task() {
    // Body of my task that finishes hopefully before 20 ms before the next activation
}

```

jrvanwhy, 2019-04-12:
An RTOS must be able to do at least one of the following:

  1. Prioritize the real-time task(s) over other tasks, preempting slower tasks to allow the real-time task to make its deadline.
  2. Make guarantees on the maximum execution time of any task (so that the real-time task can make its deadline without preemption).

More generally (combining the above two points), _an RTOS must have a guarantee on the combined maximum execution times of all tasks the real-time task(s) cannot preempt_.

Tock is currently designed so that the kernel can preempt userspace applications, but currently has no concept of prioritization between userspace applications or _within_ the kernel. I think you could easily service particular interrupts before others (not preemptively, however (*)) and particular applications over other applications. However, because the kernel is not preemptible, the shortest deadline Tock can support is no faster than the slowest individual operation within the kernel. So far, the kernel has not been written with real-time guarantees in mind, so it's unclear what the slowest operation is or how long it can take. Making the kernel preemptible would require large changes to the kernel and would introduce costs (e.g. thread synchronization) that hurt other cases of Tock.

(*) You could implement your real-time task entirely in interrupts to preempt the kernel, but in doing so you are leaving the scope of Tock's design. At that point, I argue that you are making a real-time hypervisor below Tock rather than making Tock realtime. For example, you would not be able to call into existing Tock code because that code is no reentrant.

bradjc, 2020-03-18:
Closing this with a reference to #1441. With a scheduler interface this might be something that could be explored.

closed 2020-03-18 by bradjc