tock/tock #1298: unnecessary unsafe tag

songlh, 2019-03-25:
I think the unsafe tag used for StaticRef::new() is unnecessary. Should we 

//kernel/src/common/static_ref.rs
impl<T> StaticRef<T> {
    /// Create a new `StaticRef` from a raw pointer
    ///
    /// ## Safety
    ///
    /// Callers must pass in a reference to statically allocated memory which
    /// does not overlap with other values.
    pub const unsafe fn new(ptr: *const T) -> StaticRef<T> {
        StaticRef { ptr: ptr }
    }
}

niklasad1, 2019-03-25:
Hey @songlh 

You are right this is possible to do in without unsafe Rust but I think it is marked unsafe to indicate it can violate memory safety because we can't validate that the `pointer` is valid which may be `dereferenced` later

Thus, I think it still should be unsafe 

/cc @alevy thougts?

alevy, 2019-03-25:
@niklasad1 is right. It's marked unsafe not because of what's happening _inside_ the `new` function, but because the rest of the `StaticRef` implementation (specifically `Deref`), which is _not_ marked `unsafe`, is only safe if the pointer passed satisfies the requirements in the function documentation.

So it's critical for safety to guard the _creation_ of `StaticRef` values.

alevy, 2019-03-25:
(@songlh Obviously feel free to reopen if you think this is not a satisfactory answer, I just closed for book-keeping purposes)

closed 2019-03-25 by alevy