tock/tock #1302: Low-level debug driver and/or exit syscall.

jrvanwhy, 2019-04-12:
While working on the Rust entry point and relocation issues, I wrote a driver that allows me to print debug information using only the `command` system call: https://github.com/google/tock-on-titan/blob/34cbcea1cf84e1e1b0b0575db7281f64beda387d/golf2/src/debug_syscall.rs. During the development process I found this system call invaluable.

It would be nice to move this debugging functionality upstream (into Tock rather than Tock-on-Titan) so that other developers have access to it.

In addition to the debug print functionality, there is currently a need for it in libtock-rs. libtock-rs' entry point can currently detect an error condition that makes printing diagnostics very difficult (incorrect link location). Currently, it crashes and spins on `yield` when this occurs: https://github.com/tock/libtock-rs/blob/3765065b0c6f072e7dc488eb82e8de35482fb484/src/entry_point.rs#L53. The debug driver would allow it to print out an error message indicating why the app is misbehaving.

It's not currently clear what functionality the debug driver should have. Here's a suggestion:

1. Exit with an error code -- terminates the process.
2. Print error code
3. Print two integers

We would want to divide the error code space in two: one set of error codes that is defined and documented in this repository and another set that can be used by specific boards or projects (similar to driver numbers).

jrvanwhy, 2019-04-12:
I'll leave this open for some time (a week, probably) so that people can provide suggestions. After that I'll provide a more concrete design suggestion.

rajivr, 2019-04-14:
Currently in `memop`, we have op type 10 and 11 for debugging purposes. If it makes sense, instead of introducing this as a capsule, maybe we can introduce `debugop` family of syscalls (`svc 5`) where we could include this functionality and move `memop` type 10, 11 into `debugop`.

alevy, 2019-04-15:
I think a capsule is the right place for this. I don't think this is something we'd want to _force_ platforms to include (e.g. a production platform should not have to rely on a serial device being available or on), and in any case it probably makes sense for the semantics of the error codes to be tunable by the platform.

The one exception, IMO, might be `exit`, which maybe deserves to be centered more in something like `memop`, or perhaps as part of the system call number that currently only handles `yield`.

jrvanwhy, 2019-04-15:
I agree that the debugging functionality should be optional, and therefore belongs in a capsule. `exit` feels different, and perhaps is a completely different concern.

`libtock-rs` currently has a reason to depend on `exit`, and I don't like the idea of `libtock-rs`'s entry point having a dependency on a particular capsule. Instead of providing `exit` from the kernel, we could provide the debug print call and make userspace the following:

1. Call the debug driver
1. Ignore any errors the debug driver may return
1. Loop on `yield`

Then if the debug capsule is not present it would behave like a `yield` loop, and if the debug capsule is present it would display the error code and `yield`.

jrvanwhy, 2019-04-26:
# Debug Driver RFC

Here's the promised concrete proposal. I'll leave this open for comments before I implement it.

## Design

The debug driver will be implemented as an optional Tock capsule (that any board may choose to include). It will be part of the "Base" driver set, using the next available driver number (currently `0x00008`). It will only implement the `command` system call.

### `Command` Behavior

The debug driver will provide 4 commands:

0. No-op for syscall detection.
1. Print a predefined status code
2. Print 1 number
3. Print 2 numbers

All printing commands (1-3) will print an application identifier (name or number) with the message, so that users can distinguish messages from multiple applications.

#### Print a predefined status code

We will document (in the Tock doc/ folder) a set of "predefined status codes", such as misplaced image (.text placed incorrectly with static linking) or panic (libtock-rs panic handler). This will print out a diagnostic error message indicating that a predefined status code (passed by the application in argument 1) was used.

To save flash space, the capsule will only print the numeric status code, not what it means. Users will need to look the status code up in the documentation to determine its meaning.

#### Print 1 number

This will print a single application-provided number, passed in argument 1. It will print the number in hexadecimal so that pointers can be resolved easily. The message attached to the numbers will differ from the predefined status code message so that users can distinguish application-specific messages from predefined messages.

#### Print 2 numbers

This is the same as "print 1 number" but will also print out argument 2. A use case for this is an application that has multiple debug calls in it -- the first argument can indicate which debug call triggered and the second can be the value it wants to print.

### Command Buffer

Applications can send multiple debug messages in sequence. Because there is not a mechanism for applications to wait until the capsule is ready, the capsule must queue the debug messages and print them when available. If the queue for a given application fills up the capsule should remember the location and number of dropped messages and print a dropped message indication in the correct location in the log.

### Design Notes

1. I'm not 100% sure this belongs in the "Base" set, but the only other set that it may match is "Kernel".
1. I wanted to add `exit`'s functionality to `yield` but `yield` currently accepts *any* arguments so that would be a breaking change. As I do not feel that `exit` is significant enough to allocate a new system call driver number I opted to use the userspace implementation.
1. I'm not sure how useful command #0 is -- the use cases I can imagine are "fire and forget" and would not care about its return code. Would anything care whether or not this driver is available?
1. Instead of having a custom command buffer, we could use `debug!` and its builtin buffer. This would remove the need for the grant region and greatly simplify the implementation. However, it is a much less efficient use of space (the console messages will probably be longer than 3 words). Assuming this driver will last, I decided to opt for the more capable design.
1. If you're debugging relocation issues in the entry point then the "subscribe, yield loop, continue" sequence is very difficult to put together, hence the lack of any wait mechanism in the design.

bradjc, 2019-06-04:
This seems like a useful capsule, and as a capsule it should be pretty easy to merge.

closed 2019-12-10 by bradjc