tock/tock #1306: Buggy behavior in cortexm NVIC

phil-levis, 2019-04-25:
https://github.com/tock/tock/blob/85d066d678002a5069fb770170bdeb2d8b1c08e4/arch/cortex-m/src/nvic.rs#L59

I ran into this bug on H1B. I'd argue the expected behavior is that if you disable an interrupt (NVIC number), then there won't be subsequent triggers of the interrupt handler. However, that's not what happens. Instead, whether the handler is trigger is independent of whether the interrupt is disabled or enabled, since this code traverses the pending bits (a disabled interrupt is still pending, it just doesn't actually trigger the ISR).

However, if you disable *all* interrupt, then all interrupts are disabled. This is because if *any* interrupt fires, then it will trigger *all* pending interrupts, whether they are disabled or not.

One simple fix is to check both whether the interrupt is pending and enabled. This will add cycles, though.

alevy, 2019-04-25:
I doubt the cycles would be significant. It should be just indexing into like two or three extra instructions (just indexing into the same word of the IER and taking a bitwise intersection).

Seems worth adding. It's kind of not "expected" behavior (which is why we didn't expect it :) ), but totally could happen and probably should be handled.

phil-levis, 2019-04-25:
I lost a few hours trying to figure out why a disabled interrupt continued to be delivered, but if I disabled all interrupts then none were delivered...

phil-levis, 2019-04-25:
_I doubt the cycles would be significant._ 

The number of instructions is probably small, but these are interrupt handlers; they need to be fast for energy and performance reasons. This is one of the places we want to be most careful about bloat.

bradjc, 2019-06-04:
Documenting from a recent discussion:

The approach of checking if an interrupt is pending and enabled won't work in Tock because Tock's interrupt handling approach is to disable interrupts in the interrupt controller immediately after they fire (so they don't trigger the ISR again). Later when the kernel is ready to service the interrupts it checks if the interrupt pending bit is set, and if the kernel also checked the enable bit it would always be false.

phil-levis, 2019-06-06:
It sounds like we need to keep track of this in software, then.

closed 2019-12-10 by bradjc