tock/tock #1327: Release 1.4

bradjc, 2019-06-28:
We are targeting a new release of Tock for mid-August 2019. The following tracking issues contain the milestones that will be included in the release:

- [ ] #1230 (Docs: Add README.md to /libraries/enum_primitive)
- [ ] #1215 (Tracking: Unified Course release-blocker tracking)
- [x] #1167 (UART HIL re-design)
- [ ] #1233 (GPIO HIL re-design)
- [ ] #1234 (Time HIL re-design)

Additionally, we would like to merge or close a majority of the currently open pull requests to have them included the in the release.

## Testing

To test the release we use the following applications and run them on all applicable boards and ensure they execute successfully. This list should be copied for each platform and will be tested once a release candidate is ready.

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.3, and work for 1.4.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`

- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected

## Sign off

Once the release candidate has been tested there will be check boxes for each maintainer to sign off on release 1.4.

bradjc, 2019-08-19:
> 1) Push updates to Time HIL to Time PR (end of weekend)
done
> 2) Update all time implementations to new HIL (mid next week)
> 3) Test (mid next week)
> 4) Merge Time HIL PR (end of next week)
> 5) Bump Rust version (next weekend?)
> 6) Test
> 7) Release (August 27?)

bradjc, 2019-09-24:
## imix

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.3, and work for 1.4.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 
- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected

ppannuto, 2019-09-24:
## hail

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.3, and work for 1.4.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 
- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected

alevy, 2019-09-25:
Assuming things go smoothly for imix and Hail in the next couple days, I'll take on the NRF52 on Thursday

ppannuto, 2019-09-25:
@alevy, looks like you're on the hook for doing some work tomorrow :) 👍

bradjc, 2019-09-26:
imix testing is pretty much done, and things look like they are in pretty good shape. I created a Release Candidate 2 tag (although very little has changed). I think we are in a great position to start testing all other boards.

bradjc, 2019-09-26:
## ACD52832

  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send

bradjc, 2019-09-27:
## arty-e21 (on 100t FPGA)

- [ ] Kernel boots and prints the "init complete" message.
- [ ] `examples/blink`

alevy, 2019-09-27:
## nrf52dk

- All Boards
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.3, and work for 1.4.
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
  - [x] `examples/tests/adc_continuous`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [x] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [x] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 
- imix
  - [x] tests/rng

- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [x] process console works as expected

alevy, 2019-09-27:
For `console_recv_short` and `console_recv_long` are we expecting both of them to work concurrently, or each independently?

alevy, 2019-09-27:
the NRF52dk has ieee802154 enabled since it's part of the nrf52_base configuration, but there is no ieee802154 radio in the hardware, so all 6lowpan tests sort of work until eventually there is a debug buffer overrun

ppannuto, 2019-09-27:
I tested the console recv apps independently

brghena, 2019-09-27:
## nrf52840DK

  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.4, and might work for 1.5.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send

alevy, 2019-09-27:
* Hail - @ppannuto 

  * imix - @bradjc 

  * NRF52DK - @alevy 

  * NRF52840DK - @brghena 

  * LaunchXL - @alevy 

  * ADC52832 - @bradjc / @alevy (@bradjc has a lot on his plate, and I can easily take this board over)

  * Arty - @bradjc

  * STM boards -- ??

rajivr, 2019-09-28:
> * STM boards -- ??

I can test on STM32F446RE if I can be given pointers on what should be the relevant tests that we need to pass to keep STM32 port current. STM32 only has support for UART, Timer and GPIO.

I don't use Tock on STM32 on a daily basis so it will take me a day or two to get my old STM32 development environment back up and running. (Our actual use of Tock is on an out-of-tree chip and board. We also use `libtock-rs` and not `libtock-c`, so I'll need to refresh my memory on `libtock-c`).

bradjc, 2019-10-01:
@rajivr Our basic approach is to copy all of the tests from the first post in this thread into a new post, and then run each of them and (manually) verify the output looks ok. Some tests require running multiple apps, and not all tests may be expected to work on a particular platform.

bradjc, 2019-10-02:
After my PRs are merged I'm happy with this release.

alevy, 2019-10-03:
## Launchxl

- All Boards
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.3, and work for 1.4.
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
  - [x] `examples/tests/adc_continuous`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [x] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [x] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
 
- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [x] process console works as expected

bradjc, 2019-10-04:
New RC to use for testing: https://github.com/tock/tock/releases/tag/release-1.4-rc3

alevy, 2019-10-04:
Retested the NRF52dk with RC3, and everything works with #1390 (which removes the ieee802154 driver for the nrf52dk)

rajivr, 2019-10-05:
## STM32 Nucleo-F446RE

Tested using Release 1.4-RC3

- [ ] `examples/sensors`
  - N/A

- [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - N/A

- [x] `examples/c_hello` and `examples/tests/printf_long`
  - To be run in parallel
  - Tested individually - Works!
  - In parallel does not work. Multiple C apps has not been tested on NucleoF446-RE.

- [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - We expect this fail for 1.4, and might work for 1.5.
	 
- [x] `examples/blink`
  - blink: blinks LEDs
	 
- [ ] `examples/rot_client` and `examples/rot_service`
  - N/A
  - Multiple apps has not been tested

- [x] `examples/blink` and `examples/c_hello` and `examples/button`

- [x] `examples/lua-hello`

- [x] `examples/tests/console_timeout`

- [x] `examples/tests/malloc_test01`

- [x] `examples/tests/stack_size_test01`

- [x] `examples/tests/stack_size_test02`

- [x] `examples/tests/mpu_stack_growth`

- [x] `examples/tests/mpu_walk_region`

- [x] `examples/tests/multi_alarm_test`
  - There is only one LED on board, so this test is somewhat redundant.

- [ ] `examples/test/adc`
  - N/A

- [ ] `examples/tests/adc_continous`
  - N/A

- [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - N/A
  - Multiple apps have not been tested

- [ ] `examples/ip_sense`
  - N/A

- [ ] `examples/tests/udp/udp_send`
  - N/A

- [ ] `examples/tests/udp_receive`
  - N/A

alevy, 2019-10-08:
I think we are only waiting on confirmation that the newest RC works on Hail & Imix

bradjc, 2019-10-08:
Ok I tried the hail and imix apps on the respective boards on pr 1390 and they work as expected.

ppannuto, 2019-10-08:
Also tested a handful of the most complex apps on hail -- 👍 from me to call this ready to 🚢

alevy, 2019-10-08:
Ship it!

closed 2019-10-08 by bradjc