tock/tock #1336: Process console interferes with interactive processes

ppannuto, 2019-07-12:
I didn't look into this in any depth yet, but when trying to run the [number guessing game example](https://github.com/tock/libtock-c/tree/master/examples/tests/number_guess_game), the process console seems to unexpectedly steal input, and also not accept commands correctly:

```
$ tockloader listen
No device name specified. Using default "tock"
Using "/dev/cu.usbserial-c098e513000e - Hail IoT Module - TockOS"

Listening for serial output.
Initialization complete. Entering main loop
Welcome to the number guessing game!

OK I have selected a random number between 0 and 65535
Enter your guesses, and I will let you know when you are correct!
Enter a number: 11
Valid commands are: help status list stop start fault

Nope. You are too low
Enter a number: help
Welcome to the process console.
Valid commands are: help status list stop start

Nope. You are too low
Enter a number: start

Nope. You are too low
Enter a number: 1100
Valid commands are: help status list stop start fault

Nope. You are too low
Enter a number:
```

bradjc, 2019-07-12:
It shouldn't steal input, but the input stream should go to both the app and the process console.

alevy, 2019-07-12:
Ugh... I (increasingly) think we should have a protocol on top of UART that lets tockloader and tock distinguish which process input and output is related to. I.e., I should be able to tell only send UART input to a particular process and only see output from a particular process

ppannuto, 2019-07-12:
So, other slightly odd current behavior, I tried to type the numbers `1` and `10`, not `11` and `1100` -- it seems something (process console?) echos back some characters? Again, not dug into deeply, but there's some definitely weirdness going on here. I'm inclined to agree with Amit that the better path forward may be developing input streams, rather that stressing about this particular bug [which would likely be fixed along the way to implementing those]

bradjc, 2019-07-12:
Well the guessing game only echos the numbers. So both things echo them.

bradjc, 2019-07-15:
From my point of view, things seem to work exactly as intended. Is there something specific we want to work on?

ppannuto, 2019-07-15:
The two things that caught my attention:

1) I didn't expect `Valid commands are...` to print every time when I'm just running an app...
2) I don't expect number-by-number echoing coming back at me (though it sound like that's something that the app is doing, so isn't an issue with the console?)

More generally, it's confusing to see output coming from two sources not differentiated in some way -- though a "debug channel" is much more of a wishlist thing than a bug.

bradjc, 2019-07-15:
Well but the process console is also running, assuming it is enabled in `main.rs`.

We could add a command to the process console which deactivates it, or we could implement some sort of channels, but I don't have an idea of what that would look like or how to make that user friendly, or we could add something to the beginning of process console lines, or we could do nothing.

bradjc, 2020-01-27:
Goal: allow Tockloader and users to communicate with different consoles on a board over a shared UART channel.

- [ ] Implement HDLC as the link protocol to enable differentiating messages, in particular source and destination addresses.
- [ ] Create a trait that consoles in the kernel use to access the shared UART.
    Right now I think three capsules in Tock would need to be updated to use it: `Console`, `ProcessConsole` and `kernel/debug`. I imagine others would be useful in the future as well. Ideas: kernel inspection capsule, and networking management.
- [ ] Design a protocol for allowing Tockloader to inspect the supported consoles so that users can select the one they want to interact with.
- [ ] Implement the console mux capsule that would address and dispatch messages.
    Likely this would have some special case for `Console` as it supports userspace, and applications should be directly accessible.
- [ ] Update tockloader to use the consoles.

There is a start here: https://github.com/tock/tock/tree/console-mux and other discussion in #1346.