tock/tock #1350: Tracking: Permissions for Syscall Drivers

bradjc, 2019-08-02:
This serves as a pointer to #1300 and the discussion started there. As someone picks this back up we should absolutely re-open the PR.

ia0, 2020-05-13:
I tried to use the new `filter_syscall` in `Platform` to enforce applications to only modify parts of the flash for which they have permissions. However I encountered a problem and I'm not sure if this scenario is expected to be covered by this mechanism.

Here is how an application writes to the flash:
- They allow the slice they want to write from, i.e. the data going to be written.
- They command to write to a given address in flash, i.e. the allowed slice is written at that address.

To implement the policy, one needs to know the length of the slice from the first syscall and the address in flash from the second syscall. I don't see how to do that without a state:
- either remembering the length from the first syscall in platform but this looks wrong
- or by accessing the driver state to read the length which also looks wrong

A possible workaround for this scenario would be to ask the application to give the length of the slice again in the second syscall. Then the syscall filter would check that the destination slice is writable by the application and the driver would check that both length match. This might be fine (although hacky) for such simple scenario, but if the permission to be checked spans multiple consecutive syscalls, the issue might be more problematic.

Are such scenario (where the permission depends on multiple syscalls) expected to be covered (now or at a later date)?

Thanks!

bradjc, 2020-05-13:
Hmm, this is an interesting point. One thought is this is perhaps not a good fit for syscall filtering, as it is too syscall-specific. Instead, the logic should be in the `Driver` implementation.

In this particular case, it might be better to require that the second syscall includes both an address and a length. This permits a process to allow an arbitrary length buffer once, and then issue different length writes from that same shared buffer. Otherwise, it has to "re allow" the same buffer, with just a different length. That matches how many APIs work in tock.

Lastly, a more complex syscall filtering process that looks at traces of syscalls could be interesting. I haven't seen any discussion of that so far, but it could be something to experiment with.

ia0, 2020-05-13:
Thanks for the quick response!

Indeed, currently the logic is in the `Driver` implementation but since the driver is a chip-specific driver it makes the chip be aware of the app which I'm trying to avoid (see [this comment](https://github.com/tock/tock/pull/1467#issuecomment-555058102)).

I think I'll go with the command syscall also taking the length. As you say, this could even be a feature if the application decides to optimize the allow syscalls.

hudson-ayers, 2020-06-19:
Should this be closed?

closed 2021-01-14 by bradjc