tock/tock #1372: [RFC] Providing a "no_debug" building mode

gendx, 2019-08-27:
This follows some discussion in libtock-rs, about `derive(Debug)` being expensive https://github.com/tock/libtock-rs/issues/84

In general, it could be useful to have a `no_debug` building mode for Tock where:
- derived and manual implementations of Debug are removed for all structs,
- modules related to debug (kernel/debug.rs, UART capsule, etc) are not compiled,
- the `debug!` macro does nothing.

This would reduce the code size of the kernel (potentially making more flash available for apps to use), make the code faster for production, etc.

In terms of implementation, I would define a `no_debug` cargo feature, and forward it from boards to their dependencies (capsules, the kernel etc).

What do you think?

gendx, 2019-08-28:
I definitely agree that too many features can create a mess of combinations, some of which are useless, some of which are not tested and don't compile or worse are buggy, etc. So features should indeed be limited to widely-useful, easy-to-understand and well-tested patterns, that cannot easily be obtained otherwise, and provided good documentation is given.

As you mentioned, if one wants to include specific capsules into their build, the way to go is definitely to select a list of capsules in the board's main file.

However, regarding disabling debugging, it seems hard to do without telling the kernel module to e.g. use some "empty" version of `kernel/src/debug.rs`. Looking into it, there isn't much use of the `debug!` macro in the kernel, it's mostly in individual capsules/boards. There are however a few `#[derive(Debug)]` statements in the kernel (and in capsules), that may or may not end up optimized if unused.

I'm not sure how much it makes sense to put options in Tock's custom component system vs in Cargo's features - in any case a user would have to understand either of those. And there is already some precedent of Cargo features with `#[cfg(feature = "nrf52")]` and `#[cfg(feature = "stm32f429zi")]`.

Regarding testing, I see that there is already a Travis-CI integration, so custom options should definitely be tested there (at least for compiling).

alevy, 2019-08-29:
I think cargo features are specifically a good match for something like whether to include `debug!`.

I wonder if it makes sense, though, to use the already existing `debug_assertions` feature, which is enabled when compiling in non-release mode.

gendx, 2020-01-22:
An quick update on this. When https://github.com/tock/tock/pull/1443 is merged, I think we can experiment with the `const` configuration framework to toggle debugging statements more easily.