tock/tock #1373: [RFC] Providing a "large_debug_buffer" building mode

gendx, 2019-08-27:
In the opposite direction of https://github.com/tock/tock/issues/1372, I've found it useful to increase the size of the debug buffer in the kernel (`INTERNAL_BUF` in https://github.com/tock/tock/blob/master/kernel/src/debug.rs#L227), to avoid a kernel panic when this buffer is full, especially if it takes to long for the debug sink (e.g. UART) to flush it. When printing text, 1024 bytes can be full quite quickly.

However, increasing the debug buffer size too much requires to adjust the `APP_MEMORY` size in `<board>/src/main.rs` so that everything fits in memory ; otherwise the linker cannot compile Tock.

So using a cargo feature which takes care of adjusting the sizes of both buffers makes it simpler to toggle. The feature would be in each board, and forwarded to the kernel.

Any thoughts?

alevy, 2019-08-29:
I need to play around a bit with this, but actually it might be simpler to use "all remaining space" for APP_MEMORY instead of hardcoding a value and letting the linker just verify it fits

gendx, 2019-08-30:
Definitely, if you find a way to do that, it would make things simpler, and any size-affecting change in the kernel (e.g. adding new capsules) would automatically be picked up by `APP_MEMORY`.

In that case, changing the debug buffer size would simply amount to changing the size of the debug buffer, so a cargo feature likely won't be needed for that.

gendx, 2020-01-22:
Given the migration of these buffers to the `DebugWriterComponent` (https://github.com/tock/tock/pull/1547) - which makes it easier to change the buffer sizes for a specific board, and the fact that Cargo features would be overkill for this, I'm closing this issue.

closed 2020-01-22 by gendx