tock/tock #1378: tests/console_recv_long is .. confusing?

ppannuto, 2019-09-25:
No quite clear what this is aiming to test, the readme says 64 bytes, but the receive itself asks for 61 bytes? The point of the somewhat complimentary printf_long was a test artifact of the SAM4L implementation, that does DMA in 64 byte chunks, so printing something greater than 64 bytes exercised the internal fragmentation mechanisms -- not clear that's happening with 61 here?

bradjc, 2019-09-25:
I think I remember where that came from. 64 is the size of the buffer in the capsule driver, and anything larger returns an error. And I think the app was reading for 64 bytes before, but being a nice power of two that matches the buffer size makes it too easy of a test. So it changed to 61 to be potentially a little more tricky.

bradjc, 2019-10-02:
I updated the readme, should be less confusing now.

closed 2019-10-02 by bradjc