tock/tock #1400: [RFC] New macro to generate register structs more safely and easily

gendx, 2019-10-11:
# Rationale

One pain point when defining structs of registers for chips is that one needs to manually check that each field ends up at the address defined on the data sheet, and compute the necessary padding between some fields.

This is a quite tedious and error-prone task, which can lead to:
- spending extra time to debug and double-check that all the padding is correct,
- or worse, to incorrect structures making it to the code, if a misaligned field wasn't caught by any test.

A quick `git grep _reserved` shows many padding fields throughout `chips/`:
- Some use arrays of `u8`, some use arrays of `u32`.
- A few use `ReadOnly<u8/u32>` to prevent writing inside the padding. But reserved fields shouldn't even be readable as this could cause a fault too, it should be `NoAccess<u8/u32>` or something like that instead.
- Some have comments explaining the length. But how many bugs have made it among these hundreds of padding fields?

# Proposal

Similarly to the `register_bitfields` macro to manipulate bits within registers, there could be a `register_structs` that:
- Takes as input the address of each field (so it's easier to compare to a data sheet).
- Automatically computes the size of each padding.
- Checks that the addresses are correct, given the size of each field.

## API

I wrote a simple proof-of-concept with `macro_rules`, and the following API should work. If one takes `chips/nrf52/src/adc.rs`, the macro invocation would look like the following:

```rust
register_structs! {
    AdcEventChRegisters [
        (0x000 => limith: ReadWrite<u32, EVENT::Register>)
        (0x004 => limitl: ReadWrite<u32, EVENT::Register>)
        (0x008 => END)
    ],
    AdcChRegisters [
        (0x000 => pselp: ReadWrite<u32, PSEL::Register>)
        (0x004 => pseln: ReadWrite<u32, PSEL::Register>)
        (0x008 => config: ReadWrite<u32, CONFIG::Register>)
        (0x00C => limit: ReadWrite<u32, LIMIT::Register>)
        (0x010 => END)
    ],
    AdcRegisters [
        /// Start the ADC and prepare the result buffer in RAM
        (0x000 => tasks_start: WriteOnly<u32, TASK::Register>)
        /// Take one ADC sample, if scan is enabled all channels are sampled
        (0x004 => tasks_sample: WriteOnly<u32, TASK::Register>)
        /// Stop the ADC and terminate any on-going conversion
        (0x008 => tasks_stop: WriteOnly<u32, TASK::Register>)
        /// Starts offset auto-calibration
        (0x00C => tasks_calibrateoffset: WriteOnly<u32, TASK::Register>)
        (0x010 => _reserved0)
        /// The ADC has started
        (0x100 => events_started: ReadWrite<u32, EVENT::Register>)
        ...
        /// Last result is equal or above `CH[X].LIMIT`
        (0x118 => events_ch: [AdcEventChRegisters; 8])
        (0x158 => _reserved1)
        /// Enable or disable interrupt
        (0x300 => inten: ReadWrite<u32, INTEN::Register>)
        ...
        /// Number of 16 bit samples recorded to RAM
        (0x634 => result_amount: ReadWrite<u32, RESULT_AMOUNT::Register>)
        (0x638 => END)
    ]
}
```

As you can see:
- Each field looks like `(ADDR => field_name: field_type)`. This supports arbitrary complex types (ReadWrite/WriteOnly, arrays, custom types).
- Comments are allowed between fields.
- Each padding looks like `(ADDR => padding_name)`. No need to specify the type nor the length. This saves time computing & debugging the lengths. And a consistent type can be used under the hood.
- The end is marked with `(ADDR => END)`.

This is a proof-of-concept, please let me know your suggestions to improve the syntax (provided that isn't too hard to implement with Rust macros).

## Testing

Besides generating the underlying struct, tests of the form `assert_eq!(core::mem::size_of::<field_type>(), ADDR_NEXT - ADDR, "Field {} has not the expected size", field_name)` can be automatically generated. Once the CI supports running tests on chips (e.g. with https://github.com/tock/tock/pull/1393), this will give feedback to developers adding new registers. Padding lengths won't be wrong anymore, but this can catch errors like forgetting a field.

# Known limitations

## Syntax

- Due to the difficulty of generating identifiers with `macro_rules` (https://stackoverflow.com/questions/27415011/can-a-rust-macro-create-new-identifiers), one still needs to spell out a name for each padding field, instead of using a generic name such as `PADDING`.

- Calling a reserved field `END` would be ambiguous. If that's an issue, the end could be marked as some non-ambiguous symbol like `@END`. Or the padding could be more explicit e.g. `PADDING(_reserved0)`.

## Tests

- Rust doesn't have proper support for `static_assert` (https://internals.rust-lang.org/t/idea-static-assertions/7753). There is a [static_assertions](https://crates.io/crates/static_assertions) crate, but it cannot provide error messages.

I suggest to use a `#[cfg(test)]` gated `#[test]` function, that contains assertions with proper messages (to tell on which field the address is wrong). However, this requires CI running those tests (see https://github.com/tock/tock/pull/1393).

- I've seen some examples where the register type contains a pointer directly (e.g. https://github.com/tock/tock/blob/master/chips/nrf52/src/adc.rs#L58 is `result_ptr: VolatileCell<*const u16>`). As a corollary of the previous point, if the size_of test is done on a desktop that has a different pointer size than the chip (typically 64 vs. 32 bits), then the size test fails on the desktop.

Such pointers don't seem too widespread in the codebase. One solution would be to use a `u32` in the struct and convert that to/from a pointer in the driver code. Another solution would be to define a wrapper type like `ConstPtr32Cell<u16>`, which would contain a `*const u16` on architectures where `#[cfg(target_pointer_width = "32")]` and contain a dummy `u32` with `unimplemented` methods on other architectures. This can generalize to chips that use any pointer size ; within a chip, we know the pointer size anyway.

rajivr, 2019-10-11:
> One pain point when defining structs of registers for chips is that one needs to manually check that each field ends up at the address defined on the data sheet, and compute the necessary padding between some fields.

Have you tried [`svd2regs.py`](https://github.com/tock/tock/blob/6519a10275fc9b673cd78f612701af282e937d29/tools/svd2regs.py)? When porting to a new chip, I typically use `svd2regs.py` to generate the register map `struct` and `register_bitfields!` macro. Most of the heavy lifting is handled by `svd2regs.py`. I then only need to worry about the registers that relate to the HIL and cross check those with the data sheet (and occasionally using GDB) in case something doesn't work.

Using `svd2regs.py` has been very helpful to me when porting Tock to multiple chips. Of course some vendors provide partially broken SVD files, in which case minor tweak to `svd2regs.py` was necessary.

bradjc, 2019-10-11:
Shameless plug of https://github.com/posborne/cmsis-svd/issues/73 to get the SVD library updated.

jmichelp, 2019-10-14:
Although this is a corner case, the issue still remains when the SVD file simply doesn't exist. This is currently the case for the CryptoCell in the nRF52840 chip.

ARM recently opened the code and specification of the [CryptoCell-312](https://github.com/ARM-software/cryptocell-312-runtime) and we're actively porting that in Tock with a few tweaks required for the CryptoCell-310 (the one used by Nordic in their chip). Unfortunately, ARM only released an HTML file for the registers and that's when such macro could be handy to avoid miscalculation on the padding areas.

gendx, 2019-10-14:
Thanks @rajivr for pointing out `svd2regs.py`! This indeed makes things much easier to write.

I see two main caveats though.
- This assumes up-to-date SVD files provided by the manufacturer. This is not always the case, as pointed out by @jmichelp.
- `svd2regs.py` makes things easy to write, but they are still un-intuitive to read. In particular, if one want to review a commit or audit the current code base of Tock, they would basically need to find the svd files themselves (I haven't seen any comment of the form `Generated by svd2regs.py --svd <file>`), and re-run svd2regs. On the contrary, with the macros proposed here, it would be much easier to compare each struct to the data sheet (thanks to the explicit offsets, each struct looks like a data sheet table). **Making code as easy to audit as possible sounds like a relevant property** for a "secure embedded OS" such as Tock.

The good news is that `svd2regs.py` could easily be updated to generate a macro of the form suggested in this RFC. This would provide an easy-to-write **and** easy-to-review workflow when an SVD is available. When one has to resort to "manual mode", these macros additionally make it easier to write.

bradjc, 2019-10-14:
I'm not usually a huge fan of macros, but this seems like it would be an overall good addition. Getting the padding lengths right is always a pain.

closed 2019-11-01 by bradjc

gendx, 2019-11-01:
The PR is still not merged. I think that this issue should close automatically upon merge (due to the "Fix #1400" message).

commit b7dbfbc1 (2019-10-31): Merge #1410

1410: Add macro to generate register structs, and apply it to nrf52/clock (Fix #1400) r=ppannuto a=gendx

### Pull Request Overview

This pull request adds helper macros to generate register structs, as described in https://github.com/tock/tock/issues/1400. As an example, `chips/nrf52/src/clock.rs` is migrated to use these macros.

The `register_structs` macro does two things:
- Generate the register structs, in particular computing the length of padding fields.
- Create unit tests, checking that each field indeed takes the size between its offset and the next field's offset. Thes [... cut]
commit 32b0bb39 (2019-11-01): Merge #1410

1410: Add macro to generate register structs, and apply it to nrf52/clock (Fix #1400) r=alevy a=gendx

### Pull Request Overview

This pull request adds helper macros to generate register structs, as described in https://github.com/tock/tock/issues/1400. As an example, `chips/nrf52/src/clock.rs` is migrated to use these macros.

The `register_structs` macro does two things:
- Generate the register structs, in particular computing the length of padding fields.
- Create unit tests, checking that each field indeed takes the size between its offset and the next field's offset. These t [... cut]