tock/tock #1402: Getting started documentation suffers from some bitrot

fhars, 2019-10-12:
Here are some observations about trying to get up to speed with an NRF52840-DK.

The "Getting Started"-guide works up until `make flash` if you jump ahead and read the following "Optional Requirements" section (it might be useful to say that the NRF52*-DK boards come with a built in JLink, but then I'd expect people who own one to know that). 

Then it tells you to say `tockloader install blink` to download and install the `blink`app. But that does not work:

```
$ tockloader install blink
Error opening and reading "blink"
No device name specified. Using default "tock"
No serial port with device name "tock" found
Found 2 serial port(s).

[0]     /dev/ttyS0 - ttyS0
[1]     /dev/ttyACM0 - J-Link - CDC

Which option? [0] 1

Installing apps on the board...
No TABs to install

```

(After some later tests: It looks like you have to specify the `--board` parameter before the download logic kicks in. This should be mentioned, and it would be nice if the `boards/README.md` linked to from the getting started guide somewhere around here would link to the specific READMEs.)

Downloading doesn't work, but there are examples in the two `libtock-*` repos mentioned earlier in the guide, and `libtock-rs` even contains a handy `run_example.sh` script. It might be even better if it (like the makefile in `libtock-c`) tried to install `elf2tab`, for those who have missed the dependency in the readme.
I you manage to find that, you can install a blink app on the board, but it doesn't work. Nothing blinks.

You can try the same thing in the `libtock-c`repository. A `make` in the `examples` directory will install `elf2tab` (if you haven't done before) and compile all the examples. The readme also mentions that one might need tockloader (which one would have installed at the beginning of following the getting started guide), but doesn't mention what to do with it. Finding the correct incantation requires some light reverse engineering of the `run_example.sh` script in `libtock-rs`. Then you can type `tockloader install --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52   --app-address 0x20000  blink/build/blink.tab` and it will happily install the app on the board. But, alas, it still won't blink. It only starts blinking after you also install `ble_passive_scanning`. But `ble_passive_scanning` doesn't start.

It (i.e., blink) seems to work reliably even if installed alone if you leave out the `--app-address` parameter copied from the script in the other repo.

As far as I can see, there seems to be no almost no mention of how to build apps (and especially tab-files) for a different target that the default cortex m4, even though the code is build by default for m0 and m3, too. As there currently seems to be no supported board that can run apps that is not an m4, that is not a real problem now, but might be in the future.

And the "tutorials" link on https://www.tockos.org/documentation/getting-started is a 404.

**Bonus content**

Something is really confusing here if you just start playing with the example apps, and I am not sure where the bug is (tock, tockloader or one or more of the apps (here from `libtock-c/examples`), this might merit a separate issue in the appropriate repo): Sometimes you can install two apps in parallel and they both work:

```
$ tockloader install --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52  blink/build/blink.tab 
Installing apps on the board...
Using known arch and jtag-device for known board nrf52dk
Finished in 2.542 seconds

$ tockloader install --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52  ble_passive_scanning/build/ble_passive_scanning.tab 
Installing apps on the board...
Using known arch and jtag-device for known board nrf52dk
Finished in 4.588 seconds

$ tockloader uninstall --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52 
Preparing to uninstall apps...
Using known arch and jtag-device for known board nrf52dk
There are multiple apps currently on the board:

[0]     ** Delete all
[1]     ble_passive_scanning
[2]     blink

Select app to uninstall [0] 0
Uninstall complete.
No apps on board.
Finished in 15.187 seconds
```

and sometimes you can't, and both vanish into the ether:

```
$ tockloader install --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52  blink/build/blink.tab 
Installing apps on the board...
Using known arch and jtag-device for known board nrf52dk
Finished in 2.549 seconds

$ tockloader install --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52  c_hello/build/c_hello.tab 
Installing apps on the board...
Using known arch and jtag-device for known board nrf52dk
Finished in 3.059 seconds

$ tockloader uninstall --jlink --arch cortex-m4 --board nrf52dk --jtag-device nrf52 
Preparing to uninstall apps...
Using known arch and jtag-device for known board nrf52dk
No apps are installed on the board
```

`blink` and `c_hello` always annihilate one another.

fhars, 2019-10-14:
I have found out why the blink app from libtock-rs didn't blink: tock/libtock-rs#98

bradjc, 2019-10-14:
> blink and c_hello always annihilate one another.

Yes, using JLinkExe to load a binary when that binary is less than a page in size (4096 bytes on the nrf52840) results in the page being erased. This behavior is gone in tockloader on master (apps on the nRF52 are forced to be at least 4096 bytes). A new release of tockloader is probably due, but it would be nice to get some more testing time on the dev version of tockloader since a few things have changed.

closed 2019-12-10 by bradjc