tock/tock #1408: nRF52840-dk board definition doesn't map P0.31

fhars, 2019-10-15:
Unlike the NRF52-DK board definition, the nRF52840-DK does not map the P0.31 pin (the sixth on header P2, Arduino A5). It would be easy to just add it to the array, but that would move the numbers of then pins on header P6 in the API up by one and so break any code (and potentially hardware) that uses the P6 pins as GPIOs.
The question is of course if any such code exists, as the pins exposed on header P6 have other default uses on the board (NFC, UART, and 37.768kHz crystal).

fhars, 2019-10-15:
The current GPIO configuration for the nRF52840 is actually a bit useless, four of the pins mapped on the P6 header are already in use for the UART, and the other four require soldering and cutting traces to use them as GPIOs. I'd completely redo that by by exposing the GPIOs corresponting to the Arduino pins D0 to D15 on headers P2 and P4 as GPIOs 0 to 15. I an not quite sure what to do with the pins on header P2 (Arduino A0 to A5) and P0.02. Those are the nRF52840 ADC inputs. One could of course expose them as additional GPIOs, but there seems to be no logic in place to prevent the code from trying to use them as both an analog in and a GPIO in that case.

See https://github.com/fhars/tock/compare/master...fhars:nrf52840-io for my current work in progress.

bradjc, 2019-10-17:
I think you revamping the nRF pin mappings should be no problem. I don't think any code is relying on them.

But it's great to have another set of eyes on them!

fhars, 2019-10-17:
From the recent mailing list discussions it looks like @jmichelp is working on the nRK52840, too, so there may even be a third pair of eyes. I'd like to wait for his approach, as my code is a bit ugly.

bradjc, 2020-01-22:
The board still does not map P0.31, but that is because it is labeled as an analog pin on the board. I'm closing this in favor of #1500, as then we could add the ADC capsule and pass those pins to userspace as ADC pins.

closed 2020-01-22 by bradjc