tock/tock #1409: error: component 'clippy' ... is unavailable for download for channel nightly-2019-09-19

gendx, 2019-10-15:
The following netlify build fails: https://app.netlify.com/sites/docs-tockosorg/deploys/5da5dbbc2acd8c0008206740

I'm not sure whether this is the failure reason, but the netlify log contains the following line:

```
error: component 'clippy' for target 'x86_64-unknown-linux-gnu' is unavailable for download for channel nightly-2019-09-19
```

This appeared in https://github.com/tock/tock/pull/1394, after adding a commit fixing a typo (so likely unrelated to this specific pull request). The build before this typo fix was fine.

bradjc, 2019-10-15:
rustup just updated, and it must be using the new version which installs clippy by default. https://github.com/tock/tock/pull/1399 updates the Rust version and (should) address this (@ppannuto?).

fhars, 2019-10-15:
If you fix it here, you should also fix it in libtock-rs.

bradjc, 2019-10-21:
Now it is #1417

closed 2019-10-31 by bradjc