tock/tock #1416: nrf5x: GPIOPin::disable_input() can destroy the processor

fhars, 2019-10-16:
GPIOPin::disable_input() is defined as:
```rust
    fn disable_input(&self) -> hil::gpio::Configuration {
        self.make_output()
    }
```

This will result in the processor actively driving the pin to some level determined by the contents of OUT register. Depending on what is connected to the pin and how much current it can source or sink, this this may burn out the processor's GPIO driver circuit.

If it does anything at all, it should disconnect the input buffer instead (safe default state is probably all pins configured as input with disconnected input buffers).

fhars, 2019-10-16:
The redeeming quality is that it is dead code, this compiles just fine: 
https://github.com/fhars/tock/commit/8caf423ae7a8a938b6511c83a2d3b77608de186a

alevy, 2019-10-21:
What is the means of doing this on the nrf5x?

fhars, 2019-10-26:
The correct way may be to just switch it to PullNone (what `deactivate_to_low_power` currently does), or to disconnect the input buffer, what `deactivate_to_low_power` should actually do. (Looking at it again, it may be that `deactivate_to_low_power` is currently a no-op on pins configured as output.)

The correct answer depends a bit on what the semantics of the function is intended to be. I am not sure that having `disable_input` and `disable_output` methods is the best abstraction. (I cite the fact that nothing in Tock actually calls these methods as supporting evidence)  Not all MCUs have both a disabled input and a disabled output state (do any have?). Usually, GPIOs can be either inputs or outputs with different modes, and there is an recommended low power state, which is usually some kind of input state, the details of which depend on the chip in question. On the nRF it is input with disabled input buffers, on some STMs it is IIRC analog in.

It may be best to rethink the GPIO API and how to expose it to applications (see also tock/libtock-rs#100 for something else that is subotimal). Since most of the supported boards are devel boards with lots of user accessible GPIOs, there seems to be a temptation to make the GPIO API for application very flexible. But I am not sure that this is in line with the architecture of Tock where hardware access should be mediated by the kernel as much as possible. One could argue that once you connect a button or a LED to one of the GPIOs of your board or add some shield, you have actually modified the board, and you should generate a new kernel matching that board that adds that pin to the Led or Button drivers.  This would arguably obviate the need for a generic GPIO driver (maybe keep a GPIO_IN and GPIO_OUT driver). What one would need instead would be a way to more easily generate kernels for custom boards along the lines of the recently merged #1338.

alevy, 2019-10-31:
@phil-levis / @phil-levis-google can you help clarify the semantics of `disable_input`/`disable_output`/`deactivate_to_low_power`?

phil-levis, 2019-10-31:
`disable_input` means that the pin will no longer (from a software perspective) behave as an input. Depending on what hardware does, this can mean disabling it as input, doing nothing, setting it floating, etc. But calls to `Input::read()` shouldn't let you continue to read the input. Otherwise you could get into code making assumptions of the API, i.e., for chip X disable does nothing, so your code continues to call `read()` and it still works. But then this breaks on another chip.

Same is true for `disable_output()`.

The exact details of what the hardware implementation does is up to the hardware.

In contrast, `deactivate_to_low_power()`, as its comment says
```
    /// Put a pin into its lowest power state, with no guarantees on
    /// if it is enabled or not. Implementations are free to use any
    /// state (e.g. input, output, disable, etc.) the hardware pin
    /// supports to ensure the pin is as low power as possible.
    /// Re-enabling the pin requires reconfiguring it (i.e. the state
    /// of its enabled configuration is not stored).
```
Chips differ on how you put a pin into a low power state. Rather than start making (sometimes incorrect) assumptions, such as "input but floating" or "output with value 0", the API makes the operation explicit.

phil-levis, 2019-10-31:
_Usually, GPIOs can be either inputs or outputs_

Except, of course, for chips when they can be both simultaneously. Which is why we have the current abstraction.

_and there is a recommended low power state, which is usually some kind of input state, the details of which depend on the chip in question. On the nRF it is input with disabled input buffers, on some STMs it is IIRC analog in._

Yes, this is why putting the pin into its low power state is a separate operation.

phil-levis, 2019-10-31:
```
    fn disable_input(&self) -> hil::gpio::Configuration {
        self.make_output()
    }
```

is, as @fhars points out, completely wrong.

commit acd2e828 (2019-12-16): Merge #1504

1504: nrf52: gpio: make disable_input a no-op r=alevy a=bradjc

### Pull Request Overview

This pull request removes the potentially problematic implementation of `disable_input` and copies the CC26xx chip and just does nothing.

Fixes #1416

### Testing Strategy

I didn't do anything.

### TODO or Help Wanted

Perhaps if there is a better implementation someone can add it here?

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.
PR #1504 (2019-12-16): nrf52: gpio: make disable_input a no-op
### Pull Request Overview

This pull request removes the potentially problematic implementation of `disable_input` and copies the CC26xx chip and just does nothing.

Fixes #1416

### Testing Strategy

I didn't do anything.

### TODO or Help Wanted

Perhaps if there is a better implementation someone can add it here?

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make formatall`.
closed 2019-12-16 by bors[bot]