tock/tock #1419: GDB cant load Tock elf

cody82, 2019-10-18:
I compiled Tock on Windows and want to load the elf file with GDB. But I get this error:

> BFD: C:\Users\cody\Documents\repo\tock\boards\nucleo_f429zi\target\thumbv7em-none-eabi\release\nucleo_f429zi symbol number 4 references nonexistent SHT_SYMTAB_SHNDX section
> BFD: C:\Users\cody\Documents\repo\tock\boards\nucleo_f429zi\target\thumbv7em-none-eabi\release\nucleo_f429zi symbol number 4 references nonexistent SHT_SYMTAB_SHNDX section
> Reading symbols from target\thumbv7em-none-eabi\release\nucleo_f429zi...
> BFD: C:\Users\cody\Documents\repo\tock\boards\nucleo_f429zi\target\thumbv7em-none-eabi\release\nucleo_f429zi symbol number 4 references nonexistent SHT_SYMTAB_SHNDX section
> Can't read symbols from C:\Users\cody\Documents\repo\tock\boards\nucleo_f429zi\target\thumbv7em-none-eabi\release\nucleo_f429zi: bad value

I am using GDB from gcc-arm-none-eabi-2019-q3-update. It loads other files without problems.
What could be wrong here?

rajivr, 2019-10-18:
I don't use windows, but on Linux (using `gcc-arm-none-eabi-8-2019-q3-update`), I have run into issues with GDB and `lto = true` (which is enabled when Tock kernel is compiled in [release](https://github.com/tock/tock/blob/master/boards/nucleo_f429zi/Cargo.toml#L8-L18) mode).

Try setting `lto = false` or do `make V=1 debug`. That could help.

cody82, 2019-10-18:
I get the same error without lto.

bradjc, 2019-10-20:
I have no idea what is going on, but some thoughts:

- We only use LLVM tools, which are not by any means perfect copies of the GNU versions.
- Perhaps we are missing something in the linker file?

cody82, 2019-10-21:
Looks like this only happens if I use cargo to build the binary. When I use make it works. What is the difference here? I used cargo directly on Windows because make did not work.

cody82, 2019-10-22:
I found the problem:
I did not set RUSTFLAGS properly which results in an invalid elf file.
On Windows you have to do the following:
`set RUSTFLAGS=-C link-arg=-Tlayout.ld -C linker=rust-lld -C linker-flavor=ld.lld -C relocation-model=dynamic-no-pic -C link-arg=-zmax-page-size=512`
`cargo build --target=thumbv7em-none-eabi --release`

closed 2019-10-22 by cody82