tock/tock #1424: libtock-rs apps fail to run on HiFive1

alistair23, 2019-10-23:
The problem I see is when libtock-rs apps call the memop::set_brk() syscall with a value of 0x80003800 (the default app_heap_break passed in via Tock) the app doesn't work.

If it the app doesn't execute the syscall or uses 0x80003400 as the address instead (the stacktop calculated in assembly) then it works.

With the not working setup, `process::alloc()` gets called where:

self.app_break.get = 0x80003c00
new_break = 0x80003bb4

which then returns None and results in the application failing to run (I assume it panics).

An initial attempt to fix this was sent as a PR: https://github.com/tock/tock/pull/1423

I have GDB connected to the app, so let me know what extra information would help.

To reproduce this with libtock-rs you will need this PR applied: https://github.com/tock/libtock-rs/pull/97

DieracDelta, 2019-10-31:
What command(s) are you using to attach gdb? I remember this working at one point, but currently when I use the head of the riscv openocd fork and the head of the riscv gcc toolchain, I get a segfault upon attaching to openocd with gdb.

alistair23, 2019-10-31:
I am connecting it to QEM, so I'm just using
```
target extended-remote :1234 
```

DieracDelta, 2019-10-31:
What version of qemu/gdb are you using? Using qemu's head (https://git.qemu.org/git/qemu.git) I had to run the following commands first not to get errors (using head of riscv-gcc-toolchain compiled for a 32bit imac riscv target). Running without anything gives: "bfd requires xlen 8, but target has xlen 4." Running with a gdbinit file looking like:

``` set riscv use-compressed-breakpoint off
set architecture riscv:rv32
```
I'm not really sure why this is necessary/maybe there's a way to get around it? I'm invoking qemu with 
```
$QEMU_BINARY -s -S -kernel $PATH_TO_TOCK_ELF --machine sifive_e```

alistair23, 2019-10-31:
That is a GDB bug, you need to point to the elf when you start GDB to avoid the error. I thought it was fixed in 8.1, but it doesn't appear to be and I haven't looked into it yet.

alevy, 2019-11-22:
OK, I've made some progress on this.

The core issue is that there simply isn't enough memory to run the app as-is:

By default, elf2tab requests 4116 bytes of RAM for the hello app.

Tock's default `()` implementation for the MPU (which the HiFive board currently uses in lieu of a complete PMP implementation) uses that number, plus the initial spaced required 784 bytes for the grant region (pointers, callbacks, the process struct itself...) gets us to 4864 bytes allocated to the process.

When all is said and done, the app itself consumes 4096 after allocating the heap, which leaves 4116 - 4096 = 20 bytes to allocate grants.

This fails on the first grant request for the console's grant struct, which is 96 bytes.

Increasing app memory by a bit does the trick, but strangely, the whole thing is very sensitive. In particular, asking for _too_ much app memory causes a _different_ fault (which I have not yet identified).

The investigation continues.

bradjc, 2020-01-22:
Did this ever get resolved?

alistair23, 2020-02-04:
I still see issues. It looks like the major one is the lack of PMP on RISC-V. I'll start working on that.

alistair23, 2020-02-24:
This is mostly fixed now, there is still the issue of apps not fitting, but that is being tracked here: https://github.com/tock/libtock-rs/issues/156

closed 2020-02-24 by alistair23