tock/tock #1447: Soundness of static_init macro?

gendx, 2019-11-04:
The current static_init implementation relies on `static mut` memory and heavily unsafe transmuting of memory to initialize it.

https://github.com/tock/tock/blob/0be31eac710967b6376a637be339e7e883498b24/kernel/src/common/utils.rs#L30-L33

From a soundness point-of-view, this seems problematic for several reasons.

First, `static mut` seems inherently hard to use correctly:
- https://users.rust-lang.org/t/why-is-static-mut-bad/32888 
- https://github.com/rust-lang/rust/issues/53639
- https://github.com/rust-lang-nursery/lazy-static.rs/issues/117

Second, Tock seems to be aliasing mutable memory, which is against Rust's basic memory-safety rules. For example, unless I missed something, the debugger's `OUTPUT_BUF` and `INTERNAL_BUF` are aliased mutably multiple times.

https://github.com/tock/tock/blob/8c4baa27c29967eff577c472b2897b55cef17188/kernel/src/debug.rs#L226-L227

First, in the console component.

https://github.com/tock/tock/blob/8c4baa27c29967eff577c472b2897b55cef17188/boards/components/src/console.rs#L71-L78

Second, in the process_console component.

https://github.com/tock/tock/blob/8c4baa27c29967eff577c472b2897b55cef17188/boards/components/src/process_console.rs#L66-L73

Third, in each board's setup.

https://github.com/tock/tock/blob/8c4baa27c29967eff577c472b2897b55cef17188/boards/nordic/nrf52dk_base/src/lib.rs#L252-L259

In this case, the worst that could happen is likely debug messages being interleaved, repeated or corrupted, which is probably not terrible, but this seems to point out an inherent issue with how unsafe the `static_init` API really is, and how easy it is to get it wrong. There could be other cases where aliased mutable memory is more problematic.

Has the [lazy_static](https://crates.io/crates/lazy_static) crate been considered to initialize static memory? It seems to provide the functionality that Tock needs, but I would be much more confident with it, because it is widely used (16M+ downloads) and has undergone much more scrutiny than Tock's home-made unsafe `static_init` mechanism. `lazy_static` supports `no_std` and only depends on [spin](https://crates.io/crates/spin) to provide a locking mechanism (as far as I understand, this is used to avoid any data races when initializing the static memory).

niklasad1, 2019-11-04:
Please see https://github.com/tock/tock/blob/master/doc/Soundness.md

> Has the lazy_static crate been considered to initialize static memory?

Tock has policy against external dependencies. 

//cc @alevy

gendx, 2019-11-04:
Quoting https://github.com/tock/tock/blob/master/doc/Soundness.md

> The thing that would make the use of `static_init!` unsafe is if it was used to create aliases to mutable references.

Aliasing of mutable references seems to be exactly the problem here. The `static_init` API seems too broad, in that it allows to pull in mutable references without any checks.

The soundness explanation of it doesn't seem like a convincing proof that it's actually safe (and indeed, all boards' initialization code is in broad `unsafe` blocks).

> Tock has policy against external dependencies.

I agree that in general external dependencies can cause issues when used in an uncontrolled way, especially when they recursively pull many other dependencies. However, `lazy_static` only pulls two dependencies in its current form. And re-implementing the wheel for such an unsound-prone component doesn't seem necessarily better than adding a well-reviewed dependency (it's actually in the [rust-lang-nursery](https://github.com/rust-lang-nursery) organization, and may one day be included in the standard library).

If the concern is about not controlling external code, the relevant code could be copy-pasted from `lazy_static` into the Tock repository (although Tock would have to manually check for potential bug fixes). `lazy_static` alone is only about [300 lines of code](https://github.com/rust-lang-nursery/lazy-static.rs/tree/master/src) (including comments) and the licenses seem compatible with that.

phil-levis, 2019-11-19:
You'll notice that all three are creating the same struct, an instance of kernel::debug::DebugWriter. It turns out that these three are all mutually exclusive: you should not include a Console and a ProcessConsole in your boot sequence. However, I think your larger point is valid -- this is just a programming idiom and someone can make this mistake. It should, instead, be type checked or redesigned in a way that this doesn't happen.

However, for problem 1, the use of transmute, I don't see a way around this. Using a library that does it doesn't solve the problem. There are good reasons we use static mut (please refer to the papers on Tock); we try to minimize them, but we have yet to come up with a good alternative approach with better tradeoffs.

adamreichold, 2019-12-01:
> However, for problem 1, the use of transmute, I don't see a way around this.

`MaybeUninit` as per #1446 does avoid it leaving the main issue that `static_init!` can be called twice without any interference from the compiler and this will indeed provide a way to get aliasing mutable references easily, e.g.
```
fn provide_ref() -> &'static mut bool {
   static_init!(bool, false)
}

let foo = provide_ref();
let bar = provide_ref();
```

Macros like `lazy_static!` avoid this by checking if the global is initialized before every access and running initialization exactly one (and not providing direct references at all). Since `static_init!` is used only for initialization and not to actually get a reference in the using code, I think protecting it against being executed twice should be the main issue, e.g.
```
static mut BUF: MaybeUninit<$T> = MaybeUninit::<$T>::uninit();
static FLAG: AtomicBool = AtomicBool::new(false);
if FLAG.compare_and_swap(false, true, Ordering::SeqCst) {
  panic!("Static initializer execute twice!");
}
BUF.as_mut_ptr().write($e);
&mut *BUF.as_mut_ptr() as &'static mut $T
```

adamreichold, 2019-12-01:
I think the general aliasing problems of `static mut` resp. `&'static mut` are not really an issue here as running `static_init!` once is the only way to get this reference and only passed on via reborrowing as @phil-levis describes. Or put different, the need to execute the `static_init!` block only once would also avoid any aliasing problems.

bradjc, 2020-03-18:
#1618 changes how static_init is implemented and structured.

bradjc, 2020-07-15:
We merged an update (#1924) to static_init that I believe addresses these issues nicely.

gendx, 2020-07-16:
To me, there is still a possible confusion between 2 sources of unsafety.
1. The call to `static_init` itself, which cannot be called twice (e.g. one cannot use `static_init` within a loop), otherwise objects are re-initialized without dropping. This macro cannot be made safe, because of the loop problem.
1. The value written inside of the statically-initialized object. This shouldn't call `unsafe` code, but unfortunately, because the whole `static_init!` macro must be wrapped within an `unsafe` block (due to the first point), we cannot ensure that the value written is indeed safe code. So there is still the risk of someone using something like `&mut kernel::debug::OUTPUT_BUF` within the initialized object, and thereby aliasing such an `OUTPUT_BUF` buffer.

In this case, the issue really is in obtaining a mutable reference to a static object (`OUTPUT_BUF`), which is an `unsafe` operation (because it can alias memory). I'm not sure how we can make sure that this problematic pattern doesn't occur again in the future (without using something like `lazy_static`, which works fine in loops but at the cost of keeping track of an "initialized" flag).

Yet, a first pragmatic step could be to forbid declaring `static mut` buffers anywhere (especially in capsules) - as one should use `static_init` instead (unless there really are good reasons where `static_init` doesn't apply). Unfortunately, the pattern of `static mut` buffers is [still very common](https://github.com/tock/tock/search?q="pub+static+mut").

Maybe this issue in its current form can be closed, but I think we should have an issue to track uses of `static mut` objects (which necessarily yields to unsafe code to write to these objects), given that there is the "safer" `static_init` macro.

adamreichold, 2020-07-17:
> This macro cannot be made safe, because of the loop problem.

I am not sure this is correct: We could make the macro emit a `static FLAG: AtomicBool;` that is set during execution and checked before. Then panicking if we attempt to execute the macro for a second time should avoid this unsafety?

hudson-ayers, 2020-07-17:
I agree that this seems like the easiest fix to both issues. If we do so then the ability to alias global `static mut` variables is outside the scope of issues with the `static_init()` macro.

gendx, 2020-07-20:
> > This macro cannot be made safe, because of the loop problem.
> 
> I am not sure this is correct: We could make the macro emit a `static FLAG: AtomicBool;` that is set during execution and checked before. Then panicking if we attempt to execute the macro for a second time should avoid this unsafety?

That's a good point! And something like `AtomicBool` is actually the primitive needed to implement a simple spinning mutex, so in a sense the `lazy_static` solution is similar to that. Which has the overhead of keeping this flag around.

We need to be careful of [memory ordering semantics](https://doc.rust-lang.org/1.0.0/core/atomic/enum.Ordering.html) though (even though Tock is single threaded, we don't want the compiler to reorder these instructions). That's why people usually rely on well-tested crates like `lazy_static`.

So would the following work inside of the `static_init` macro ([source code](https://github.com/tock/tock/blob/master/kernel/src/common/utils.rs#L16))? Note that the expression `$e` would be created outside of the `unsafe` block, so that users of `static_init` don't get a free pass to put unsafe code in there.

```rust
static INITIALIZED: AtomicBool = AtomicBool::new(false);
if INITIALIZED.swap(true, Ordering::SeqCst) {
    panic!("Already initialized");
}

let mut buf = $crate::static_buf!($T);
let expr: $T = $e;
// Safety: happens only once thanks to the INITIALIZED flag.
unsafe { buf.initialize(expr) }
```

hudson-ayers, 2020-07-20:
Would this be possible with a regular bool only read/modified using volatile reads / writes? Based on this the discussion here: https://github.com/tock/tock/issues/1654#issuecomment-660191771 , it might be nice if we could structure Tock to not rely on the presence of atomic instructions, as these are not available on some platforms (such as earlgray / cortex-m0) and access to LLVM intrinsics for atomics will always be unstable. This is especially true given that Tock's single threaded design really should not require them.

gendx, 2020-07-20:
> Would this be possible with a regular bool only read/modified using volatile reads / writes? Based on this the discussion here: [#1654 (comment)](https://github.com/tock/tock/issues/1654#issuecomment-660191771) , it might be nice if we could structure Tock to not rely on the presence of atomic instructions, as these are not available on some platforms (such as earlgray / cortex-m0) and access to LLVM intrinsics for atomics will always be unstable. This is especially true given that Tock's single threaded design really should not require them.

As far as I understand, atomics are the only primitive allowing to:
1. Provide strong memory ordering semantics, such as `Ordering::SeqCst`. Without them, the compiler is free to move unrelated instructions around. Because the guarded code doesn't touch the flag memory at all, the compiler can assume that a flag implemented as a regular boolean is useless and just optimize it away.
1. Safely mutate (non-mutable) `static` objects. A regular bool would have to be `static mut`, which adds a layer of unsafety.

Maybe the strong semantics offered by (1) also mean that they are hard to put into a compiler, and therefore the reason why earlgray/cortex-m0 don't support them? I'm not familiar with the specifics, but do these platform not support atomics for a lack of compiler support, or because the underlying CPUs can fundamentally not provide such semantics?

hudson-ayers, 2020-07-20:
> As far as I understand, atomics are the only primitive allowing to:
>
> 1. Provide strong memory ordering semantics, such as `Ordering::SeqCst`. Without them, the compiler is free to move unrelated [...]

I don't think we need strong memory ordering semantics here? All accesses to this regular boolean would be made by the same thread, so it is not possible for this code to be reentrant.

```rust
static mut INITIALIZED: bool = false;

if unsafe {
  let init_ptr: *mut bool = &mut INITIALIZED;
  core::ptr::read_volatile(init_ptr) } 
    {
    panic!("Already initialized");
} else {
    unsafe {
        core::ptr::write_volatile(init_ptr, true);
    }
    let mut buf = $crate::static_buf!($T);
    let expr: $T = $e;
    // Safety: happens only once thanks to the INITIALIZED flag.
    unsafe { buf.initialize(expr) }
}
```

Do you see a problem with this code? While the use of `static_mut` requires unsafe, I don't think this code can cause UB so long as `static_init!()` is never called from interrupt context, which it never will be given Tock's design (see https://github.com/rust-lang/rust/issues/53639 for description of why `static mut` is safe without reentrancy). And https://doc.rust-lang.org/beta/std/ptr/fn.read_volatile.html states "Volatile operations [...] are guaranteed to not be elided or reordered by the compiler across other volatile operations." , so I don't think ordering is an issue here.

> 2. Safely mutate (non-mutable) `static` objects. A regular bool would have to be `static mut`, which adds a layer of unsafety.

(see above)

> Maybe the strong semantics offered by (1) also mean that they are hard to put into a compiler, and therefore the reason why earlgray/cortex-m0 don't support them? I'm not familiar with the [...]

I just know that the thumbv6 and rv32imc instruction sets do not include atomic instructions. I assume this is because the CPUs using the instructions can't provide such semantics.

hudson-ayers, 2020-07-20:
I think this issue should remain open until we either

a) decide on some change to mitigate the possible unsoundness from calling `static_init()` in a loop or

b) explicitly decide that this potential unsoundness is okay because it can be prevented by correct use of the macro and we expect it to be prevented by code reviews.

I don't think this issue was resolved by the previous changes to `static_init()`, though they were helpful.

adamreichold, 2020-07-20:
> While the use of `static_mut` requires unsafe, I don't think this code can cause UB so long as `static_init!()` is never called from interrupt context

Please note that `static mut` might even be deprecated entirely and removed in future Rust editions. If atomic semantics are not required, then
```rust
static INITIALIZED: UnsafeCell<bool> = UnsafeCell::new(false);
```
is probably the way to go.

I am also not sure volatile access is really helpful here: No memory-mapped I/O is involved and any reordering of memory accesses by the compiler should be correct under single-threaded execution which seems to be assumed in any case if atomic access is to considered unnecessary.

hudson-ayers, 2020-07-20:
> Please note that `static mut` might even be deprecated entirely and removed in future Rust editions. If atomic semantics are not required, then
> 
> ```rust
> static INITIALIZED: UnsafeCell<bool> = UnsafeCell::new(false);
> ```
> 
> is probably the way to go.

Makes sense to me.

ppannuto, 2020-07-21:
> ... Provide strong memory ordering semantics, such as Ordering::SeqCst.

I _think_ we can also provide such things using barriers defined by each architecture that Tock supports. This is of course more work and error prone than being able to rely on intrinsics.

> but do these platform not support atomics for a lack of compiler support, or because the underlying CPUs can fundamentally not provide such semantics?

I can speak strongly the cortex-m0 as I know that µarch very well. There, it's mostly that the primitives are unneeded as the core itself is very simple. An M0 is a 3-stage in-order pipeline with no caches and no support for multiple cores. M0+ and M3/M4's have small icaches, optional LSQs, writeback buffers, etc, which is why they have support for various barriers. For the M0, however, effectively everything is always atomic*, so it's not meaningful to describe different memory consistency models are there is only one.

*Of course, there are complications to this. For example, if an exception occurs during the processing of an `LDM/STM{IA}`, it will be started over. But these are not consequential to memory _consistency_.

gendx, 2020-07-21:
The fundamental problem about memory ordering is not about *runtime* (multithreading/multiple CPU cores, interrupts/reentrancy), but about *compile-time* (see e.g. [*Who's afraid of a big bad optimizing compiler?*](https://lwn.net/Articles/793253/) article from LWN.net).

Without strong ordering semantics, the compiler is free to re-order or remove operations, and it does it all the time for optimization purposes (think about optimizing away algorithms on an abstraction like a Rust iterator when we in fact iterate over a simple range `0..n`).

Now about volatile, the Rust documentation indeed states that "Volatile operations [...] are guaranteed to not be elided or reordered by the compiler **across other volatile operations**". This isn't helpful for us because the volatile operations on the `INITIALIZED` flag will be independent of the non-volatile operations on the buffer being initialized.

Using volatile operations as an attempt to order operations is a common misconception, see also __*C++ and the Perils of Double-Checked Locking*__ (by Scott Meyers and Andrei Alexandrescu, well-known C++ experts), which mentions volatile in Section 5.

---

That being said, the compiler cannot reorder operations across side-effects. For example, I expect that a syscall in userspace libtock implemented as an `asm` block with a memory barrier will be considered such a side-effect (though we have to make sure the `asm` block is used properly).

In our case, I wonder whether the `panic!` statement is considered such a side-effect. And what exactly are Rust's semantics regarding accessing a `static UnsafeCell`.

For example, in the following code, the compiler must issue a `panic` at some point.

```rust
for _ in 0..2 {
    static INITIALIZED: UnsafeCell<bool> = UnsafeCell::new(false);
    if unsafe { INITIALIZED.get() } {
        panic!("Already initialized");
    }
    unsafe { INITIALIZED.set(true) }

    let mut buf = $crate::static_buf!($T);
    let expr: $T = $e;
    unsafe { buf.initialize(expr) }
}
```

But is it allowed to reorder it as:

```rust
for _ in 0..2 {
    let mut buf = $crate::static_buf!($T);
    let expr: $T = $e;
    unsafe { buf.initialize(expr) }

    static INITIALIZED: UnsafeCell<bool> = UnsafeCell::new(false);
    if unsafe { INITIALIZED.get() } {
        panic!("Already initialized");
    }
    unsafe { INITIALIZED.set(true) }
}
```

or as:

```rust
for _ in 0..2 {
    let mut buf = $crate::static_buf!($T);
    let expr: $T = $e;
    unsafe { buf.initialize(expr) }
}

for _ in 0..2 {
    static INITIALIZED: UnsafeCell<bool> = UnsafeCell::new(false);
    if unsafe { INITIALIZED.get() } {
        panic!("Already initialized");
    }
    unsafe { INITIALIZED.set(true) }
}
```

or even:

```rust
for _ in 0..2 {
    let mut buf = $crate::static_buf!($T);
    let expr: $T = $e;
    unsafe { buf.initialize(expr) }
}

panic!("Already initialized");
```

or simply:

```rust
panic!("Already initialized");
```

Ultimately, we actually *want* the compiler to optimize away the flag in the regular case where the initialization happens only once. But in the case where `static_init` is misused, I want to make sure that the semantics are clear and that we're not shooting ourselves in the foot.

hudson-ayers, 2020-07-21:
Thanks @gendx for the explanation, that is very helpful. I agree volatile is unnecessary here.

My line of thinking for the `UnsafeCell` version was that while reordering is possible, the `panic!()` is guaranteed to execute in the case of `static_mut()` being called twice, so it doesn't really matter if the second reference is taken right before the panic, because the panic will happen afterwards anyways. However I now realize that may not be true in the following scenario:

`static_init()` is called in a loop, thus there are multiple calls to `static_buf()`, and therefore two different references have been taken to the same `static mut BUF` object, which causes immediate UB -- even if at runtime, the second reference would never be used because of the panic. Because of the presence of UB, the compiler can do anything, include removing the panic entirely.  Am I understanding correctly?

If my understanding is correct, would using `static UnsafeCell<UninitializedBuffer> Buf` instead of `static mut BUF`, in combination with a `static UnsafeCell<bool> INITIALIZED` be sound?

gendx, 2020-07-22:
> Thanks @gendx for the explanation, that is very helpful. I agree volatile is unnecessary here.
>
> My line of thinking for the `UnsafeCell` version was that while reordering is possible, the `panic!()` [...]

That's the scenario I had in mind indeed.

> If my understanding is correct, would using `static UnsafeCell<UninitializedBuffer> Buf` instead of `static mut BUF`, in combination with a `static UnsafeCell<bool> INITIALIZED` be sound?

This is where things get tricky in rolling our own primitives IMO. My take is that I'd leave it to compiler experts to explain what is or isn't sound. That's why I'd advocate for using well-known, widely used, and preferably well audited libraries (or the stdlib) for these kind of primitives.

I think `lazy_static` is suitable (it's at least been widely used for a long time), although indeed as far as I know it relies one way or another on the presence of atomics, and while it's fair to say it's been widely used on desktop CPUs, it's probably been less tested on embedded systems. The overhead of `lazy_static` (due to the multi-threaded model it targets) might also not be suitable for us.

Maybe bringing up our use case to the Rust language or to a library like `lazy_static`, and advocating for more support for it, would be more effective than trying to play around ourselves (unless someone here has enough compiler expertise to build a robust solution).

adamreichold, 2020-07-22:
> The overhead of lazy_static (due to the multi-threaded model it targets) might also not be suitable for us.

Please note that `lazy_static` does not do the same thing as `static_init` does, i.e. supports blocking other threads until the thread that does the initialization is done with that. (Which is why it depends on `spin` in `no_std` environments.) This is not necessary for `static_init` why under correct usage is only called exactly once and here the atomics are necessary only to ensure safety when usage is incorrect, it does not need to actually coordinate multiple competing accesses.

Considering the question of compiler reorderings, when multiple execution contexts are expected (threads, interrupts, etc.), then atomics (or at least fences) are necessary AFAIU. When we can assume that execution is single-threaded, then compiler reorderings should not be a problem as they preserve observable program behavior and for example unwinding through a panic (or even aborting execution outright) is a control flow dependency the observable effects of which will not be reordered by the compiler meaning that regular accesses should suffice just as they suffice in e.g. bounds checking via
```
assert!(index < length);
buffer[index] = value;
```

(The role of `UnsafeCell` should just be that we need to tell the compiler that we will mutate a value via a shared reference `&'static T`. If this were not a global variable, then a regular `Cell` might even suffice but that is not `Sync` which is necessary for global data in the presence of multi-threaded execution.)

gendx, 2020-07-23:
> Please note that `lazy_static` does not do the same thing as `static_init` does, i.e. supports blocking other threads until the thread that does the initialization is done with that. (Which is why [...]

I agree that `lazy_static` does more than what we need, but what it does is sufficient for what we need.

I also agree that under correct usage `static_init` is fine, but we can also say the same thing about any `unsafe` Rust API. The hard part is to define what is "correct" usage for a given API, and how to ensure that usage doesn't mistakenly become incorrect.

Interestingly, looking for "singleton" pattern in Rust, I found this from the Rust embedded working group: https://rust-embedded.github.io/book/peripherals/singletons.html. So there is a `singleton!` macro in the `cortex_m` crate (https://docs.rs/cortex-m/0.6.3/cortex_m/macro.singleton.html) which basically does a similar thing as our `static_init!`. However, they take a different approach: the macro has a safe API, calling it more than once simply returns `Some(x)` the first time and `None` afterwards: https://docs.rs/cortex-m/0.6.3/src/cortex_m/macros.rs.html#52-76. I think it's a reasonable approach to remove a lot of `unsafe` initialization code in Tock (despite the potential overhead of the `Option`).

hudson-ayers, 2020-07-23:
> I agree that `lazy_static` does more than what we need, but what it does is sufficient for what we need.
>
> I also agree that under correct usage `static_init` is fine, but we can also say the same [...]

We have previously said that we do not want to include dependencies like `cortex-m` because that would dramatically increase the size of the TCB. We could not easily just import `singleton!()` as it has dependencies on lots of other code in that crate. I agree that the API of the singleton macro is nice, but I also think it ends up being not any different in practice from an example which panics on reuse, because we would pretty much always want to unwrap the returned item anyway.

I don't like the idea of using `lazy_static` because it is heavier weight than what we need and I still have a hunch that we will end up wanting to structure the kernel to not rely on atomics so that we can more quickly move to stable Rust (I could be wrong on this though, I plan to bring it up at the next core meeting). Also, I am not sure we could use lazy_static as-is if it uses atomics internally, we might need to modify it to use hand-rolled atomics via core-intrinsics.

I think that a PR with a straightforward modification to `static_init()` which enforced a panic on reuse would probably be welcomed, though I don't want to speak for others. We could even present the modified version to the compiler team for feedback on soundness. But given that we have yet to experience a bug in master related to incorrect use of `static_init!()`, I suspect that there is probably not a ton of interest in moving away from this tool that was designed with Tock's concurrency model in mind and which has generally served us well up to this point, especially when the alternatives come with downsides of their own (external dependencies, atomics requirements, spin, etc.)

gendx, 2020-07-24:
What we need is much closer to `singleton!` than `lazy_static!`. Even though we don't depend on either of those (for TCB's sake), it would IMO be preferable to have our implementation be as close as possible to either of them, and to provide a *safe* API. I don't think it's a good trade-off to reduce the number of dependencies but at the expense of dramatically increasing the `unsafe` surface internally (many lines of `unsafe` code in the initialization of boards).

So I think a reasonable approach would be to provide basically the same API as `singleton!`, with a similar implementation - but without the `$crate::interrupt::free` part, since the board initialization code shouldn't interfere with interrupts in Tock's model, right?

That way:
- We provide a safe API and can remove a lot of `unsafe` code. Ultimately boards could even be `forbid(unsafe)` like the capsules are.
- We are not an isolated island in the Rust world, as we have the same use case as the `cortex-m` crate. Ultimately this can help discussing with the Rust language how to better support our non-concurrent model, and maybe hope to have some standardized `singleton` API in Rust one day (so we don't have to handle and support the `unsafe` part ourselves).

What do you think?

bradjc, 2021-04-26:
We should translate this to a todo issue, or close this. Is the todo item to implement `static_init!()` with the same semantics as `singleton!()`?

hudson-ayers, 2021-04-27:
I agree with @gendx that we should provide a safe API, and that `singleton!()` is pretty much the interface we want. However, without the `interrupt::free` part, its not a completely safe interface, as it would not necessarily be safe to call after board initialization. I don't think it would be appropriate to provide a function with a safe interface which would be unsafe to call in code from which it could be accessed.

I *think* it might be safe if we put it all in an atomic block, but that requires a different implementation of `static_init!()` for each architecture to ensure the correct atomic block is called. Or maybe passing the architecture to the `static_init()` call.

tldr; `singleton!()` is what we want, and that should be the todo item. Just a note of caution that we can't just replace the `$crate::interrupt::free` part -- we need to replace it with an equivalent if we want the macro to be callable without `unsafe`

bradjc, 2021-04-27:
We could put `static_init()` in its own crate, and only include it from boards/.

hudson-ayers, 2021-04-27:
It would still be callable in functions defined in `boards/` that are passed to the core kernel to be called later. For example, someone could call `static_init!()` in `Platform::with_driver` or in a closure passed to a capsule (e.g. the `baud_rate_reset_bootloader_enter()` function passed as a function pointer to the USB CDC capsule) or in a closure passed to a `FaultPolicy`.

Of course, that is the kind of thing that would hopefully be caught in code review. But I think the intended usage of `unsafe` is that it should not be possible to write code that compiles but leads to memory unsafety without using the `unsafe` keyword at the point where code is written that leads to that memory unsafety.

bradjc, 2021-04-27:
> It would still be callable in functions defined in `boards/` that are passed to the core kernel to be called later. For example, someone could call `static_init!()` in `Platform::with_driver` or in [...]

Ok, that makes sense. I was thinking since `kernel_loop` returns `!` that anything would have to run before then but that isn't the case.

bradjc, 2021-04-27:
Tracking issue: #2564

closed 2021-04-27 by bradjc