tock/tock #1459: RFC: Flash HIL improvements

ia0, 2019-11-12:
## Motivation

Flash technology does not permit the usage of arbitrary data-structures due to restrictions like:
- It is not possible to switch a single bit back 0 to 1 without erasing all 0s to 1s in the page containing that bit. But it is possible to switch a single bit from 1 to 0.
- A page can only be erased a given amount of times before exhibiting undefined behavior. In other words, the flash has a short lifespan.

To implement flash-friendly data-structures, it is useful to take benefit of all features the flash provides like writing multiple times the same unit of flash (smaller than a full page) before having to erase it.

## Description

The current flash HIL interface is:
- The type of a page (must implement `AsMut<[u8]>`).
- Read a page asynchronously.
- Write a page asynchronously (page to read from must be mutable).
- Erase a page asynchronously.

This interface has the following issues:

### Issue 1: Writing a page takes a mutable page to read from.

This is not clear why the write function has the following signature:

```rust
fn write_page(&self, page_number: usize, buf: &'static mut Self::Page) -> ReturnCode;
```

It would seem more natural to take a shared slice to read from:

```rust
fn write_page(&self, page_number: usize, buf: &'static Self::Page) -> ReturnCode;
```

### Issue 2: The read and write buffer lifetimes are `'static`.

This probably requires the user of the HIL (e.g. a capsule) to have a static buffer allocated and probably copy data back and forth between this backend interface and its own frontend interface (like `AppSlice` in case of a capsule). This issue is due to the fact that the API is non-blocking and that the HIL implementation probably cannot allocate or free a buffer.

#### Proposal 1: Use blocking API.

### Issue 3: It is not possible to read a smaller slice than a page.

This is not clear why such restriction is in place.

Note however that exposing a read function is probably not useful in the first place if it is possible to read the flash directly, which is the case for nrf52840dk flash for example. So it is as well not useful to expose such function in a capsule since the application can already read its own flash.

See the proposal for the same issue for write operations if read operations should be provided.

### Issue 4: It is not possible to write a smaller slice than a page.

This is not clear why such restriction is in place. If flashs permit to write at a smaller granularity, this option should be exposed.

#### Proposal 1: Expose the smallest write granularity.

```rust
fn write_size(&self) -> usize;
/// Both `ptr` and `buf.len()` must be `write_size` aligned (otherwise `EINVAL`).
fn write_slice(&self, ptr: usize, buf: &[u8]) -> ReturnCode;
```

### Issue 5: It is not specified whether the flash may be used during an operation.

Flashs seem to have different behavior regarding what can be done during a mutable operation (like write or erase). Some flashs halt the CPU because instructions are read from the flash. Some flashs may have an instruction cache, which may permit cached instructions to be executed. Some flashs may require the user to not read from the flash (either all the flash or just the region under mutation). This is an issue with a non-blocking interface, since it is possible for code using this flash HIL interface to have undefined behavior without knowing.

#### Proposal 1: Always use blocking API.

#### Proposal 2: Simulate blocking API to keep the API safe.

This means that flashs that would exhibit undefined behavior if the flash is accessed during a mutable operation would do the mutable operation in the command phase instead of between the command phase and the callback phase.

#### Proposal 3: Expose when it is OK to use the flash concurrently of a mutable operation.

```rust
/// What is OK during a mutable operation.
fn concurrency(&self) -> Concurrency;
enum Concurrency {
    Nothing,  // should block until mutable operation is complete
    ExecuteCachedInstructions,
    ExecuteInstructions,
    ReadOtherFlashRegion,  // other = non-overlapping or maybe with some alignment
    MutateOtherFlashRegion,
}
```

### Issue 6: It is not specified how often the flash may be written.

This can become an issue once it's possible to write multiple times before erasing.

#### Proposal 1: Expose how many times the flash may be written.

```rust
/// Maximum number of times the flash may be written before erased.
/// This holds independently for each unit of `write_size`.
fn max_writes(&self) -> usize;
```

### Issue 7: It is not specified how often the flash may be erased.

The flash HIL may expose undefined behavior if a page is erased too often.

#### Proposal 1: Expose how many times the flash may be erased.

```rust
/// Maximum number of times a page may be erased before out of order.
fn max_num_erases(&self) -> usize;
```

## Proposals

Blocking API:

```rust
/// Size of a page in bytes.
fn page_size(&self) -> usize;  // 4096 for nrf52840dk

/// Size of the smallest unit of flash that can be read in bytes.
fn read_size(&self) -> usize;  // 1 for nrf52840dk

/// Whether the flash can be directly read.
fn direct_read(&self) -> bool;  // true for nrf52840dk (to avoid syscalls)

/// Size of the smallest unit of flash that can be written in bytes.
fn write_size(&self) -> usize;  // 4 for nrf52840dk

/// Whether the flash can be directly written.
fn direct_write(&self) -> bool;  // true for nrf52840dk (not sure if useful in practice).

/// Maximum number of times the smallest unit of flash can be written before erased.
fn max_writes(&self) -> usize;  // 2 for nrf52840dk

/// Maximum number of times a page can be erased before out of order.
fn max_erases(&self) -> usize;  // 10000 for nrf52840dk

/// Reads into `buf` the slice at `ptr` of length `buf.len()`.
/// - `ptr` and `buf.len()` must be `read_size` aligned.
/// - The slice starting at `ptr` of length `buf.len()` must be in flash.
fn read_slice(&self, ptr: usize, buf: &mut [u8]) -> ReturnCode;

/// Writes `buf` to the slice at `ptr` of length `buf.len()`.
/// -`ptr` and `buf.len()` must be `write_size` aligned.
/// - The slice starting at `ptr` of length `buf.len()` must be in flash.
/// - `buf` should not switch 0s into 1s. [This could actually be parametrized since on nrf52840dk it is OK to write a 0 to a 1, it will simply be ignored and stay a 0, as opposed to undefined behavior.]
/// - No units of `write_size` must have been written `max_writes` times.
fn write_slice(&self, ptr: usize, buf: &[u8]) -> ReturnCode;

/// Erases the page at `ptr` of length `page_size`.
/// - `ptr` must be `page_size` aligned.
/// - The page starting at `ptr` of length `page_size` must be in flash.
/// - The page must not have been erased `max_erases` times.
fn erase_page(&self, ptr: usize) -> ReturnCode
```

Non-blocking API (only the additional functions):

```rust
fn read_slice(&self, ptr: usize, buf: &'static mut [u8]) -> ReturnCode;
fn Client::read_done(&self, result: Result<(), ReadError>);
// If allocation is possible, it would be:
fn read_slice(&self, ptr: usize, len: usize) -> ReturnCode;
fn Client::read_done(&self, result: Result<Box<[u8]>, ReadError>);  // buf.len() == len

fn write_slice(&self, ptr: usize, buf: &'static [u8]) -> ReturnCode;
fn Client::write_done(&self, result: Result<(), WriteError>);
// If deallocation is possible, it would be:
fn write_slice(&self, ptr: usize, buf: Box<[u8]>) -> ReturnCode;
fn Client::write_done(&self, result: Result<(), WriteError>);

fn erase_page(&self, ptr: usize) -> ReturnCode;
fn Client::erase_done(&self, result: Result<(), EraseError>);
```

## Questions

- What are flashs that don't match the proposed APIs? Or what are flashs we are interested in so that I can look them up?
- Do we want blocking API? (only issue is the buffer copy if non-blocking since safety can be worked around by blocking during the command phase or documenting the safety issue in the interface)
- Should we distinguish embedded flash from external flash?

ia0, 2019-11-12:
### Flash-related drivers in Tock

There are 2 HILs:

- `hil::flash::Flash` provides read, write, and erase operations for pages only.

- `hil::nonvolatile_storage::NonvolatileStorage` provides read and write operations for arbitrary slices. However the write operation always erases before writing (since no erase operation is provided).

There are 4 related implementations:

- `sam4l::flashcalw::FLASHCALW` implements `hil::flash::Flash`.

- `capsules::mx25r6435f::MX25R6435F` implements `hil::flash::Flash`.

- `capsules::nonvolatile_to_pages::NonvolatileToPages<F: hil::flash::Flash>` implements `hil::nonvolatile_storage::NonvolatileStorage` and permits to use a `hil::flash::Flash` as a `hil::nonvolatile_storage::NonvolatileStorage`.

- `capsules::nonvolatile_storage_driver::NonvolatileStorage` implements `hil::nonvolatile_storage::NonvolatileStorage` and `kernel::Driver`. It takes a `hil::nonvolatile_storage::NonvolatileStorage` as argument and provides access to that storage to both the kernel and the applications.

Those components are chained (`capsules::nonvolatile_storage_driver::NonvolatileStorage` on top of `capsules::nonvolatile_to_pages::NonvolatileToPages` on top of a `hil::flash::Flash`) and used in 2 boards:

- `imix` uses `sam4l::flashcalw::FLASHCALW`.

- `nrf52dk_base` uses `capsules::mx25r6435f::MX25R6435F`.

### Conclusion

The proposed API would be some merge of both HILs expect that writing a slice would not erase what was there before. A convenient function could be provided for backward compatibility (this would be implemented in the trait directly or could be a helper function).

ia0, 2019-11-14:
## Internal flash survey for nRF52 series

The nRF52 series has an internal flash memory that can be controlled with the NVMC (Non-volatile memory controller). The specifications are (copy-pasted and reformatted from the [documentation](https://infocenter.nordicsemi.com/topic/struct_nrf52/struct/nrf52.html) with my comments in italic and TLDR in bold):
- The CPU must be halted before initiating a NVMC operation from the debug system. _Not sure if relevant for us._
- The flash is divided into 256 pages of 4 kB each. **num_pages=256 page_size=4096**
- The same 32-bit word in the flash can only be written 2 number of times before a page erase must be performed. **write_max_count=2** _The current proposal assumes that the maximum number of writes is for the minimum size requirement, i.e. write_min_size) which is satisfied in this case._
- The NVMC is only able to write 0 to bits in the flash that are erased (set to 1). It cannot rewrite a bit back to 1. _The current proposal is not parametric over this specification._
- Only full 32-bit words can be written. To write less than 32 bits, write the data as a full 32-bit word and set all the bits that should remain unchanged in the word to 1. Note that the restriction on the number of writes still applies in this case. **write_min_size=4**
- Only word-aligned writes are allowed. Byte or half-word-aligned writes will result in a hard fault. **write_min_align=4** _The current proposal assumes that write_min_align (minimum alignment requirement for a write) and write_min_size (minimum size requirement for a write) are equal which is satisfied in this case._
- The CPU is halted if the CPU executes code from the flash while the NVMC is writing to the flash.
- After erasing a flash page, all bits in the page are set to 1.
- **erase_max_count=10000** _This is only documented in the electrical specification as far as I can tell._

These specifications apply to the following boards:
- nRF52840
- nRF52833
- nRF52811
- nRF52810

nRF52832 is special and differs in the following ways:
- _The "CPU must be halted with debug system" is not present._
- The flash is divided into multiple pages. Each page is divided into 8 blocks. **num_pages=128 page_size=4096 block_size=512**
- The same block in the flash can only be written 181 number of times before an erase must be performed. **write_block_max_count=181** _This mechanism is different from the proposal, but fits the proposal by using write_max_count=write_block_max_count/(block_size/write_min_size)=1 although it is a waste._
- _Nothing specifies the endurance of the flash, i.e. erase_max_count._

## Conclusion

It would be nice to have equivalent survey for other boards. However, given the existing difference for nRF52832, it looks like trying to define a flash HIL now is too early. It seems preferable to firstly write and use board-specific capsules. Once we have enough stable such specific capsules (they would represent a minimum set of flashs we care about), we can try to come up with a generic flash HIL that satisfies those capsules.

phil-levis, 2019-11-15:
Issue 1: When you receive the buffer back, you don't want to lose mut. I.e., if I have a mutable reference to a buffer, I don't want it to become a regular reference if I read flash into it. This is a common design pattern on Tock APIs. 

Issue 2: All Tock kernel APIs have to be non-blocking because there is a single thread of control. Otherwise, the entire kernel (including interrupt bottom halves) will wait.

Issues 3, 4, 6: Many external flash chips have only page-level read interfaces. This API is intended to be a generic API for almost any flash chip. So it tries to include operations that are present in all flash chips. Having an additional API/trait for word or byte-level access would be reasonable. Some Flash chips can implement this trait. Same with multi-writes.

Issue 5: I don't quite understand this context: can you give an example of when this is needed/useful? It's my understanding that things like instruction caches are in the CPU, rather than the flash chip. The caches are small. Is what you're getting at whether someone can assume the operation seems blocking? It seems to be that we should just instead assume it isn't.

Issue 7: This is not something you know statically. It is a probabilistic guarantee based on the underlying flash properties. Also, the number of erases per page is only really useful if you also keep track of how many times each page *has* been erased. For NAND flashes with FTLs, this becomes mostly useless since it remaps everything anyways.

ia0, 2019-11-15:
> Issue 1: When you receive the buffer back, you don't want to lose mut. I.e., if I have a mutable reference to a buffer, I don't want it to become a regular reference if I read flash into it. This is a common design pattern on Tock APIs.

I see thanks! So it means this issue is actually a consequence of the non-blocking API. Since to simulate a borrow (not only mutable but also shared), one needs to give ownership and get it back. It makes sense, I missed that.

> Issue 2: All Tock kernel APIs have to be non-blocking because there is a single thread of control. Otherwise, the entire kernel (including interrupt bottom halves) will wait.

You probably wrote this before reading Issue 5. If the CPU is halted during flash operations, a non-blocking API won't help you execute instructions during the flash operation.

> Issues 3, 4, 6: Many external flash chips have only page-level read interfaces. This API is intended to be a generic API for almost any flash chip. So it tries to include operations that are present [...]

You probably miss the context of #1457 where a suggestion from @alevy was to extend or change the current flash HIL instead of adding a new one. The current and initial version of #1457 is adding a new HIL for flashs that provide word-level writes as you currently suggests. But given the current state of discussions, I believe it's better to not even try to create a new HIL and simply have a board-specific capsule exposing low-level access to the flash. Once we have enough stable board-specific flash drivers, we can try to unify them under one or more HILs.

> Issue 5: I don't quite understand this context: can you give an example of when this is needed/useful? It's my understanding that things like instruction caches are in the CPU, rather than the flash [...]

On the nRF52840 (and other nRF52 boards), the NVMC (non-volatile memory controller) [specification](https://infocenter.nordicsemi.com/topic/ps_nrf52840/nvmc.html?cp=4_0_0_3_2) states the following:
- The CPU is halted if the CPU executes code from the flash while the NVMC is writing to the flash.

I heard (but don't have references) that other boards may still permit executing instructions from flash during flash operations (as long as those instructions are not where the flash is writing). Similarly, some boards (no reference) may permit reading from flash as long as it's not in the region being written or erased. Similarly (I guess for nRF52 but I'm not sure I understand the specification correctly), if some instructions are cached, maybe it's fine to execute them while writing to the flash. Those are safety considerations to be considered during flash operations when instructions are executed from flash. I expect each board to have different specifications as described above. Using a blocking API would be on the safe side. But as suggested, we can as well block in the command phase and still expose a non-blocking API.

> Issue 7: This is not something you know statically. It is a probabilistic guarantee based on the underlying flash properties. Also, the number of erases per page is only really useful if you also [...]

Yes, the goal of #1457 (and thus this issue) is to be able to write flash-aware data-structures in user-land. For example, I have an implementation of a multi-set of slices with wear-leveling. Storing the number of page erasures also permits to notify the user that the device is dying and they should buy a new one and transfer their data when we are reaching the limit of page erasures. The maximum number of page erasures is also useful to know how many bits to use to store the page erasure count for each page.

Regarding FTL, this is mostly for when such feature is not present. Userland can implement data-structures that do wear leveling (and error correction).

I hope this clarifies the context.

rajivr, 2019-11-17:
I ran into a related issue yesterday and since I was aware of this issue upstream, it caused me to take a fresh look into flash controller of MCUs.

There are three scenarios that I discovered.

1. CPU can be stalled during flash operations such as program/erase.

This seems to be similar to what we are seeing with nRF52.

> The CPU is halted if the CPU executes code from the flash while the NVMC is writing to the flash.

In this case if we don't want to stall the CPU, the recommendation seems to be to run the application including the interrupt vector table from SRAM. 

This is not ideal for us and I am guessing that would be the same for other users of TockOS.

2. CPU has multiple flash memory banks (typically two) but _without_ a flash controller interrupt.

In this scenario, as long as we are using one bank for execute-in-place and another bank for flash operations, the CPU does not stall. However, because there is no flash controller interrupt, it would lead to a scenario, where we'll have to setup a timer and use that to check for the completion of the flash operations.

This would be workable for TockOS.

3. CPU has multiple flash memory banks and _with_ a flash controller interrupt.

This is similar to scenario 2, but we no longer need to attach a timer, as we'll get an interrupt indicating the status of the flash operation. The constraint of XiP using one flash bank and flash operations to the other still applies.

This would be the ideal case for TockOS.

I was unable to find an MCU that could do flash operations and XiP on the same bank without stalling.

phil-levis, 2019-11-19:
_You probably wrote this before reading Issue 5. If the CPU is halted during flash operations, a non-blocking API won't help you execute instructions during the flash operation._

Yes, of course. But this API should work for off-chip flash as well. If the API is blocking, then a write to an off-chip flash through an asynchronous SPI bus operation needs to be made blocking.

_Many external flash chips have only page-level read interfaces. This API is intended to be a generic API for almost any flash chip. So it tries to include operations that are present in all flash chips. Having an additional API/trait for word or byte-level access would be reasonable. Some Flash chips can implement this trait. Same with multi-writes._

This seems like a good path forward. It's fine to write chip-specific APIs that expose detailed features. The key thing is that their traits should explicitly indicate this. I.e., it should be clear that if my capsule is using chip-specific features of the STM25P family, it's calling a trait like Stm32PFlash. That way you can either write chip-independent code that works on lots of flash chips (e.g., using the current Flash HIL), but sacrifice features to do so, or you can use a specific chip and make the most of its capabilities.

ia0, 2019-11-20:
The last question of the original post is: Should we distinguish embedded flash from external flash? As @rajivr pointed out, the question is more subtle since we should not only take into account whether the flash is embedded but also if it has multiple banks. In which case the driver would need to know from which bank code is executed and which bank is being operated on.

Regarding chip-specific APIs, we now have 2 PRs at the opposite ends of the spectrum and your middle-ground proposal (also suggested by @bradjc):
- #1457 adds a HIL for embedded flashs. I think most of us agree this is too early to commit to an API. We first would need to wait until we have enough chip-specific drivers implemented and stable, then we could try to unify them under a single HIL.
- #1467 adds a chip-specific syscall driver for the embedded flash of the nRF52 boards. This has an orthogonal issue regarding whether chip-specific drivers should "understand" applications. Currently, `AppId` is part of `kernel::Driver` so this is the case by design, but this is not necessarily something desirable.
- The middle ground (not implemented, but I can do it if we believe this is how chip-specific drivers should be implemented; we don't have any implemented yet): A chip-specific HIL implemented by a chip and used by a capsule. That capsule would not fit the design in #1466 as I understand it.

To continue the survey I started with nRF52 series, I looked a bit at OpenTitan flash controller [documentation](https://docs.opentitan.org/hw/ip/flash_ctrl/doc/#features), and the following HIL should satisfy both nRF52 series (including nRF52832) and OpenTitan (although the documentation is not very precise):
- Number of banks
- Number of pages per bank
- Number of bytes per page
- Read granularity: I'm assuming it's a byte (true for nRF52, not stated for OpenTitan).
- Write granularity (it's 4 bytes for nRF52, not stated for OpenTitan)
- Maximum number of writes per line (it's 2 for nRF52 and 181 for nRF52832)
- Number of bytes per line (it's 4 for nRF52 and 512 for nRF52832)
- Maximum number of erases per page (it's 10000 for nRF52)
- Direct read: I'm assuming a flash controller is not needed to read from flash (true for nRF52). And since unaligned reads are permitted (read granularity is byte), then there is no safety concern to let users read the flash directly without going through the controller.
- Write slice (both address and length are aligned at write granularity). Users should not write the flash directly to avoid undefined behavior due to unaligned writes. This is enforced by Tock since the flash is read-only for the applications (including the writeable flash regions).
- Erase page (address aligned at page size)
- Erase bank