tock/tock #1464: Tracking: Kernel Documentation Round 2

bradjc, 2019-11-15:
## High-level advice documentation

- [ ] Document Tock design decisions.
  - [ ] Event-driven kernel
  - [ ] No dependencies
  - [ ] Ease of understanding code and getting started with Tock
- [ ] Generic overview on how to do various high-level goals.
  - Examples:
    - [ ] Write a chip peripheral.
    - [ ] Add an IC driver.
    - [ ] Pass an interface to userspace.
- [ ] When to use capabilities.

## More detailed developer help documentation

- [ ] Comment differences (`//` vs `///` vs `//!`) and when to use which
- [ ] How to write kernel tests.
- [ ] Expand the debugging help in the debugging folder

## Update existing docs

- [ ] Mutable references. Make sure doc is consistent with actual use.

## New documentation on Tock things

- [ ] Doc describing how resources are virtualized in capsules.

## Update Tock Book

- [ ] Add in notes about components for boards

## Other Ideas

- Merge all (most?) documentation into the book so there is one canonical place to look.

bradjc, 2020-03-18:
Can we volunteer people to write the remaining documents?

bradjc, 2020-03-18:
Also, will you upvote this comment if you support moving most of the documentation that is in tock/doc to the book (or downvote if not)? I would like to move all of the design-esque .md files to the book so that everything is in one place.

hudson-ayers, 2020-04-09:
I think that if we move all of the documentation in tock/doc to the book we need to reorganize the book so that documentation is at the start, and the hands-on guides are at the end.

bradjc, 2020-06-19:
We got fairly far on this. We can do a Round 3 as new documentation is requested/needed.

closed 2020-06-19 by bradjc