tock/tock #1470: macro static_init!

bandheight, 2019-11-18:
The current implementation of this macro is
```rust
macro_rules! static_init {
    ($T:ty, $e:expr) => {
        {
            use core::{mem, ptr};
            static mut BUF: Option<$T> = None;
            let tmp : &'static mut $T = mem::transmute(&mut BUF);
            ptr::write(tmp as *mut $T, $e);
            tmp
        };
    }
}
```
and why not is
```rust
macro_rules! static_init {
    ($T:ty, $e:expr) => {
        {
            use core::mem;
            mem::transmute::<&mut $T, &'static mut $T>(&mut $e)
        };
    }
}
```
?

gendx, 2019-11-18:
See also https://github.com/tock/tock/pull/1446 and https://github.com/tock/tock/issues/1447 for related discussions about `static_init`.

bandheight, 2019-11-19:
@gendx I look at this discussions. You are right! Use **MaybeUninit** to define a mutable static variable and then init later finally get mut ref is elegant way.

However, I test my way, I found this way is more memory efficient than you and origin way.
So, is there any other side effect of directly extending the lifetime by **mem::transmute** .

niklasad1, 2019-11-19:
@bandheight in your example it seems possible to pass for example an array on the stack instead of `static mut` and transmuting it as `&'static mut T` is undefined behavior.

bandheight, 2019-11-19:
The return value of `$e` is actually stored on the stack. You are right.

bandheight, 2019-11-19:
If we do this?
```rust
macro_rules! static_init {
    ($T:ty, $e:expr) => {
        {
            use core::mem;

            unsafe fn extend_lifetime(mut r: $T) -> &'static mut $T {
                mem::transmute::<&mut $T, &'static mut $T>(&mut r)
            }

            extend_lifetime($e)
        };
    }
}
```

alevy, 2019-11-19:
@bandheight I believe that still stores the value on the stack, and simply “lies” about the value’s lifetime.

gendx, 2019-11-19:
Taking a reference to a temporary expression and extending its lifetime is definitely undefined behavior that will break your program, and is precisely a memory safety issue that Rust's lifetimes aim to protect against. This is `unsafe` for a good reason.

bandheight, 2019-11-20:
There is a description of `core::mem::transmute` in the documentation.

> Extending a lifetime, or shortening an invariant lifetime. This is advanced, very unsafe Rust!

Maybe we really shouldn’t do it in such a `unsafe` way.

closed 2019-11-20 by bandheight