tock/tock #1472: HiFive1 timer interrupt not handled

alevy, 2019-11-19:
/cc @DieracDelta 

The HiFive1 port’s timer doesn’t completely work. I believe it’s because, while the timer is setup, the CLIC (? I think) interrupt is not actually handled in chips/e21/src/chip.rs’s `handle_interrupts`.

alistair23, 2019-11-20:
The HiFive1 uses the PLIC. Do you have a way to reproduce the issue?

DieracDelta, 2019-11-20:
So I think the functionality of a timer right now is essentially disabled. As far as I know, the Hifive1 board uses the e31 chip (chips/e310x/chip.rs), a PLIC, and a CLINT (CLINT is an alternative to the CLIC I think). In the e31 chip's service_pending_interrupts, we currently "handle" timer interrupts by reading them from the plic MMIO and then writing the interrupt ID back to that same spot in MMIO. Instead, we could add a machine timer interrupt line to the match statemenet (this should work because the memory layout of the CLINT and the arty timer are the same) and the boiler plate machine timer intialization code contained in the Arty board.

Adding the code should be easy but this will need to be tested.

@alistair23 does qemu produce timer interrupts?

alistair23, 2019-11-21:
QEMU should be producing timer interrupts, although I haven't checked in this case.

I can run the libtock-rs [hello example](https://github.com/tock/libtock-rs/blob/master/examples/hello.rs) on the `hifive1` and it correctly prints roughly every half a second with this change to handle QEMU's faster clock, so I'm not sure what isn't working.

```diff
diff --git a/kernel/src/hil/time.rs b/kernel/src/hil/time.rs
index 74fd3b6f..b0ebf2fc 100644
--- a/kernel/src/hil/time.rs
+++ b/kernel/src/hil/time.rs
@@ -44,7 +44,7 @@ impl Frequency for Freq16MHz {
 pub struct Freq32KHz;
 impl Frequency for Freq32KHz {
     fn frequency() -> u32 {
-        32768
+        10000000
     }
 }
```

AFAIK a timer interrupt will call into the `handle_trap()` function and then call the timer's `handle_interrupt()` function: https://github.com/tock/tock/blob/master/chips/e310x/src/chip.rs#L99

alevy, 2019-11-21:
OK, if @alistair23 can run it, I believe it works. I'll try it with the Rust userland (easier for me to compile anyway)

alistair23, 2020-05-29:
Can we close this?