tock/tock #1477: Possible Atomicity Violation in kernel deferred_call

BurtonQin, 2019-11-21:
https://github.com/tock/tock/blob/9b2a1d50e10a7303c205b4eaf2b00841ebef1801/kernel/src/common/deferred_call.rs#L43-L54

In function next_pending()
1. check atomic variable DEFERED_CALL
2. if it is 0, return None
3. otherwise, update DERFERED_CALL based on old value and return Some.

If two threads call next_pending() simultaneously, possible atomicity violation may happen.
1. Th1: load and check atomic variable 
2. Th2: load and check atomic variable and update the atomic to 0.
3. Th1: not knowing atomic val has become 0, still update atomic based on the old value and returns Some. It seems that this is not the original intention "return None on 0".

rajivr, 2019-11-21:
@BurtonQin AFAIK, TockOS kernel is single-threaded. I am not aware of TockOS running on a multi-core CPU. I expect this to be an issue that we'll need to deal with porting TockOS to something like a Cortex-M7.

phil-levis, 2019-11-22:
The kernel is not multithreaded. There are many pieces of code which would break if it were.

closed 2019-11-22 by phil-levis